ms-iot/ROSOnWindows #142: no dll files inside this folder

gptshubham595, 2019-08-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
* Catkin workspace setup instructions
  * Clone locations
  * build command line
  * Launch command line

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment information:**
 - OS version: `winver`
 - Chocolatey Versions: `choco list --local-only`

seanyen, 2019-08-21:
@gptshubham595 Hi, can you check if you have the same environment like this issue? https://github.com/ms-iot/ROSOnWindows/issues/134

gptshubham595, 2019-08-27:
No

gptshubham595, 2019-08-27:
Inside that folder only some cmake files exist but no dll without cmakelist.txt

gptshubham595, 2019-09-13:
I Got the solution it may work for others , I have installed Visual studio in drive D: and was installing it on C:
So it wasnt installing/ downloading all files

simply i have done:
Created shortcut inside drive d:
created shortcut
and on opening i changed as cd d:
then done same procedure it worked

gptshubham595, 2019-09-13:
Configuring done
CMake Error at CMakeLists.txt:13 (add_executable):
  No SOURCES given to target: MicoMoveIt

seanyen, 2019-09-13:
@gptshubham595 Did you use `catkin_make` to build? Have you followed through [Using ROS on Windows](https://wiki.ros.org/Installation/Windows) to setup a developer command prompt? ROS projects has its owned build tool in additions to pure CMake.

ooeygui, 2019-09-15:
Could you post your chocolatey log? it is located in c:\ProgramData\chocolatey\logs\chocolatey.log

You may also find that a dependency of MicroMoveIt isn't available yet on Windows. 
Could you post the repo for it and I'll take a peek.

gptshubham595, 2019-09-18:
Yes MicroMoveit was not installed so i did it . But now other problem is there
on running roscore after running setup.bat

[image]

logfile:

[Uploading roslaunch-SHUBHAM-10112.log…]()

Choco list:
Chocolatey v0.10.15
assimp 4.0.1.1809242322
boost 1.66.0.1904102339
bullet3 2.87.0.1809242326
bzip2 1.0.6.1901242108
chocolatey 0.10.15
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.4
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1903240636
git 2.22.0
git.install 2.22.0
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1904131648
hg 5.0.2
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1903221831
ros-cmake 3.11.4.1809242322
ros-melodic-desktop_full 20190729.1
ros-melodic-moveit 20190729.1
ros-python2 2.7.15.1904300710
ros-vcpkg 2018.11.23.1906112226
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.22.27821
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
75 packages installed.

gptshubham595, 2019-09-18:
check this I will remove this repo just to show what files I'm having:

https://github.com/gptshubham595/ROS-WINDOWS-FILES

gptshubham595, 2019-09-18:
[image]

seanyen, 2019-09-18:
@gptshubham595 I noticed that your installation is on `D` drive. Did you manually copy things over there? Currently the `ROS on Windows` build makes an assumption that it should be under `C` drive. If you need to make your ROS installation under other drives, you can go check this [page](https://ms-iot.github.io/ROSOnWindows/GettingStarted/UsingROSonWindows.html), where we have an example how to create a filesystem links across drives.

ooeygui, 2019-09-18:
Linking https://github.com/ms-iot/ROSOnWindows/issues/103

seanyen, 2020-09-22:
Closing it and use the meta ticket to track.

closed 2020-09-22 by seanyen