ms-iot/ROSOnWindows #146: Need to update the windows dependencies for azureiot

BaoxiJia, 2019-08-27:
**Describe the bug**
The **azureiot** dependency for Windows version is out-of-dated. The folder structure for Windows is different from the latest official version for Linux. 

**To Reproduce**

- On Windows, **macro_utils.h** is located in "C:\opt\rosdeps\x64\include\azureiot\**azure_c_shared_utility**" folder.

- On Linux, after install azure-iot-sdk-c-dev with the command "sudo apt-get install -y azure-iot-sdk-c-dev",  **macro_utils.h** is located in /user/include/azureiot/**azure_macro_utils** folder. 

We have refer to the different path in the source code  - 

#ifdef _WIN32
#include <azure_c_shared_utility/macro_utils.h>
#else
#include <azure_macro_utils/macro_utils.h>
#endif

**Environment information:**
 - OS version: 10.0.18362.295
 - Chocolatey Versions: ros-melodic-desktop_full 20190729.1

ooeygui, 2019-09-16:
Thanks for the post. We're working on vcpkg work for dependencies and will include this work

seanyen, 2020-04-24:
Now we switched to consume the vcpkg port, and https://github.com/microsoft/ros_azure_iothub has been switched to consume the port.

closed 2020-04-24 by seanyen