ms-iot/ROSOnWindows #147: Joint State Publisher node crashes without error messages.

giovent, 2019-08-27:
**Describe the bug**
joint_state_publisher node crashes. The process dies without any particular prompt.

**To Reproduce**
(Once you have run moveit)
`rosrun joint_state_publisher joint_state_publisher` 

**Expected behavior**
The node starts correctly.

**Environment information:**
 - OS version: Windows 10 1803
 - Chocolatey Versions: <details><summary>Chocolatey v0.10.11</summary><p>assimp 4.0.1.1809242322
boost 1.66.0.1809242326
bullet3 2.87.0.1809242326
bzip2 1.0.6.1809242322
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.3
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
eigen 3.3.4.1809242322
freeglut 3.0.0.1809242322
git 2.20.1
git.install 2.20.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1809242322
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.4
KB3035131 1.0.2
libccd 2.0.0.1809242322
libcurl 7.58.0.1809242322
libfcl 0.5.0.1809242322
libflann 1.9.1.1809090720
libglew 2.1.0.1809101757
libgraphviz 2.41.0.1810022039
libogg 1.3.3.1809090720
libompl 1.2.2.1809120057
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libqglviewer 2.7.1.1809090720
libqhull 2015.2.0.1809242322
libtheora 1.1.1.1809090720
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
octomap 1.9.0.1809090720
ogre 1.10.11.1809100020
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1810112239
ros-catkin-tools 0.0.1.1901130741
ros-cmake 3.11.4.1809242322
ros-melodic-desktop 20190121.1
ros-melodic-desktop_full 20190729.1
ros-melodic-moveit 20190127.1
ros-python2 2.7.15.1809242354
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.16.27024.1
yaml-cpp 0.6.2.1901170034</p></details>

ooeygui, 2019-09-16:
Sorry for the delay. Joint_state_publisher crashes when there is a bug in the urdf. Could you post logs or a stack trace?

giovent, 2019-10-23:
Apparently my PyQt module was broken, reinstalling it solved the problem, thank you!

closed 2019-10-23 by giovent