ms-iot/ROSOnWindows #150: [Setup] Include screenshot of shortcut for the setup.bat

ooeygui, 2019-09-19:
For this URL:
https://wiki.ros.org/Installation/Windows

Please include a shortcut screenshot to demonstrate what the setup.bat should look like.

gavanderhoorn, 2019-09-20:
May I suggest to not use screenshots to show the content of a file?

The [support guidelines](http://wiki.ros.org/Support) (admittedly, for ROS Answers and Discourse) explicitly request to not use screenshots for anything that is text.

If "screenshot" here is meant to mean an actual image, then it might be good to just show the context of `setup.batch` with the appropriate formatting (ie: code block).

ooeygui, 2019-09-20:
@gavanderhoorn Thank you for the feedback. We've had lots of support requests which boil down to not sourcing the setup.bat file.

Instead of a screen shot, perhaps we should change the document to delay creating the Windows shortcut until after ROS is installed?

gavanderhoorn, 2019-09-21:
I may have misunderstood this sentence:

> Please include a shortcut screenshot to demonstrate what the setup.bat should look like.

If the intent is to just show a screenshot of the windows desktop with a screenshot, that would be completely fine.

If the intent is to show the contents of a file with a screenshot, then just copy-pasting the file *contents* would be preferable.

ooeygui, 2019-09-21:
@gavanderhoorn Oh, thanks for the clarification. I edited the original comment. Since Windows is UI heavy, sometimes trying to describe what should go where is problematic. However, there are also issues with localization of screens, so I think some combination of reordering might be the best option anyway.

ooeygui, 2019-09-24:
I reordered the getting started guide. Hopefully this is clearer about the setup.bat need.

closed 2019-09-24 by ooeygui