ms-iot/ROSOnWindows #151: ros::master::check() hanging

songlan-hl, 2019-09-25:
**Describe the bug**
I tried to use ros::master::check() to see if roscore is ready. But it kept hanging in this function .

**To Reproduce**
I simply ran the program quoted in https://answers.ros.org/question/236678/check-master-ros-node-is-started-c/. If I didn't start roscore at the beginning and ran it, the program never output the "waiting..." line or anything. The program got hanging forever. If I started roscore before, it would output "master started!" as expected. 

**Environment information:**
Standard installation and catkin_make followed ROSonWindows page.

songlan-hl, 2019-09-25:
To add to the report, if I skipped this line, but used getTopics(), getNodes() when roscore is not ready, they will all hang. The right behavior should return empty.

seanyen, 2019-09-25:
@songlan-hl Thanks for reporting. I will test and see if this behavior only exhibits on Windows or is it platform-agnostic. And see what actions should we take after.

seanyen, 2019-09-26:
After some investigation, the hanging happens at the following location:
https://github.com/ros/ros_comm/blob/d487d5877049d61fe2ebcc6487e9a7a72d9f0346/utilities/xmlrpcpp/src/XmlRpcDispatch.cpp#L125

And in this situation, when a non-blocking socket is in use in [`WSAPoll`](https://docs.microsoft.com/en-us/windows/win32/api/winsock2/nf-winsock2-wsapoll), if it failed to connect to the peer host, [`WSAPoll`](https://docs.microsoft.com/en-us/windows/win32/api/winsock2/nf-winsock2-wsapoll) won't be signaled the error back to caller. Also, there are many [discussions](https://www.bing.com/search?q=%22Windows+8+Bugs+309411+%E2%80%93+WSAPoll+does+not+report+failed+connections%22) about this behavior.

I will be sending a pull request to [`ros_comm`](https://github.com/ros/ros_comm) iterating on the proper fix, and thank @songlan-hl for the reporting.

songlan-hl, 2019-09-26:
thanks to @seanyen for quick response. Any workaround to check if the master is running with current build?

seanyen, 2019-09-26:
The workaround is to checkout [`ros/ros_comm`](https://github.com/ros/ros_comm) in your `catkin` workspace and apply this [patch](https://patch-diff.githubusercontent.com/raw/ros/ros_comm/pull/1816.patch). After you source the `devel` workspace, you node should be able to consume the new built binaries.

seanyen, 2020-02-04:
The fix is merged by the upstream, so I am closing it.

closed 2020-02-04 by seanyen