ms-iot/ROSOnWindows #152: [meta ticket] ROS on Windows installation issues

seanyen, 2019-09-25:
This is a meta ticket to link all ROS install\upgrade related issues.
  - https://github.com/ms-iot/ROSOnWindows/issues/142
  - https://github.com/ms-iot/ROSOnWindows/issues/137
  - https://github.com/ms-iot/ROSOnWindows/issues/134
  - https://github.com/ms-iot/ROSOnWindows/issues/133
  - https://github.com/ms-iot/ROSOnWindows/issues/103
  - https://github.com/ms-iot/ROSOnWindows/issues/68

michaelajax, 2019-09-26:
@seanyen Just added https://github.com/ms-iot/ROSOnWindows/issues/153

GhengisGohan, 2020-05-20:
Hey ya'll, been working on the install for about 3 days and just cant get it! Here's the link to my issue.... 

https://answers.ros.org/question/352872/failure-install-ros-melodic-desktop_full-ros-catkin-tools-error-while-running-choclateyinstallps1/

ooeygui, 2020-05-20:
@GhengisGohan Thank you for the pointer. Since it wasn't tagged with Windows, we didn't see it.

GhengisGohan, 2020-05-20:
@ooeygui thanks! Now I know better.

ooeygui, 2020-05-20:
@GhengisGohan Looking at the log file, it looks like a file lock is preventing overwrite. I don't know if that is left over from a previous failed attempt. Perhaps a virus scanner is preventing the install? 

Could you attempt to exclude c:\opt from realtime scanning?

ooeygui, 2020-05-21:
@GhengisGohan Thanks for the update on the ROS answers. I see some interesting things in the log, but nothing that should cause this failure. I suspect your visual studio cmd environment is inheriting something from the host.

Could you post the output of `set`? (Please xxx out any private information)

GhengisGohan, 2020-05-21:
As requested... I don't know why I didn't think of it but I'm updating my VS 2019 community with 'choco update visualstudio2019community'. It must have been one of the early VS 2019 that I initial downloaded. 

```
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\XXX\AppData\Roaming
ChocolateyInstall=c:\opt\chocolatey
ChocolateyLastPathUpdate=132344855562799183
CommandPromptType=Native
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=XXX
ComSpec=C:\WINDOWS\system32\cmd.exe
DevEnvDir=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\
DriverData=C:\Windows\System32\Drivers\DriverData
ExtensionSdkDir=C:\Program Files (x86)\Microsoft SDKs\Windows Kits\10\ExtensionSDKs
Framework40Version=v4.0
FrameworkDir=C:\Windows\Microsoft.NET\Framework64\
FrameworkDir64=C:\Windows\Microsoft.NET\Framework64\
FrameworkVersion=v4.0.30319
FrameworkVersion64=v4.0.30319
FSHARPINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\FSharp\
HOMEDRIVE=C:
HOMEPATH=\Users\XXX
INCLUDE=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\ATLMFC\include;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\include;C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um;C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared;C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um;C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt
LIB=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\ATLMFC\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\lib\x64;C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\lib\um\x64;C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\ucrt\x64;C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\um\x64;
LIBPATH=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\ATLMFC\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\lib\x86\store\references;C:\Program Files (x86)\Windows Kits\10\UnionMetadata\10.0.17763.0;C:\Program Files (x86)\Windows Kits\10\References\10.0.17763.0;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;
LOCALAPPDATA=C:\Users\XXX\AppData\Local
LOGONSERVER=\\XXX
NETFXSDKDir=C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\
NUMBER_OF_PROCESSORS=8
OneDrive=C:\Users\XXX\OneDrive
OneDriveConsumer=C:\Users\XXX\OneDrive
OS=Windows_NT
Path=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.20.27508\bin\HostX64\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\x64\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\FSharp\;C:\Program Files (x86)\Windows Kits\10\bin\10.0.17763.0\x64;C:\Program Files (x86)\Windows Kits\10\bin\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\\MSBuild\Current\Bin;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;c:\opt\chocolatey\bin;C:\WINDOWS\Windows\System32\WindowsPowerShell\v1.0;c:\opt\chocolatey\lib\ros-catkin-tools\tools\;C:\Program Files\Git\cmd;c:\python37\;C:\Users\XXX\OneDrive\Documents\Anaconda 3;C:\Users\XXX\OneDrive\Documents\Anaconda 3\Scripts';c:\users\XXX\onedrive\documents\anaconda 3\lib\site-packages;C:\Users\XXX\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\XXX\AppData\Local\GitHubDesktop\bin;C:\System32\WindowsPowerShell\v1.0;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
Platform=x64
PROCESSOR_ARCHITECTURE=AMD64
[... cut ...]  Files\nodejs\;C:\ProgramData\chocolatey\bin;c:\opt\chocolatey\bin;C:\WINDOWS\Windows\System32\WindowsPowerShell\v1.0;c:\opt\chocolatey\lib\ros-catkin-tools\tools\;C:\Program Files\Git\cmd;c:\python37\;C:\Users\XXX\OneDrive\Documents\Anaconda 3;C:\Users\XXX\OneDrive\Documents\Anaconda 3\Scripts';c:\users\XXX\onedrive\documents\anaconda 3\lib\site-packages;C:\Users\XXX\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\XXX\AppData\Local\GitHubDesktop\bin;C:\System32\WindowsPowerShell\v1.0

```

ooeygui, 2020-05-21:
It looks like you have both python3 and Anaconda in the system path. I suspect that during a part of the install phase, python3 is getting executed instead of the python2 deployed by our ROS install scripts.

To test this hypothesis, can you remove `c:\python37\;C:\Users\XXX\OneDrive\Documents\Anaconda 3` from your path?

GhengisGohan, 2020-05-21:
@ooeygui took out the recommended paths. Didn't install the ros-catkin-tools. Took out another path (anaconda 3/lib/site-packages), ran the install again, still didn't go. Any more hypotheses?

I uploaded my latest logs here: 
https://github.com/GhengisGohan/chocolatey_logs

seanyen, 2020-05-21:
@GhengisGohan

Looking into the [`Chocolatey.log`](https://answers.ros.org/question/352872/failure-install-ros-melodic-desktop_full-ros-catkin-tools-error-while-running-choclateyinstallps1/), the last debug log from `ros-catkin-tools` is `Install released ROS catkin tools...`. And the next line of it in the install script is:

```
pip install -U -r $requirementsFile --disable-pip-version-check --no-cache-dir
```

I am suspecting it is using the wrong `pip` in your environment.

To help me understand your env, can you run the following commands & report the output?

```
:: show where the python in your system
where python

:: show where the pip in your system
where pip

:: print pip version
pip --version

:: print python paths
python -m site
```

GhengisGohan, 2020-05-21:
@seanyen as requested...
'''
C:\Windows\System32>where python
INFO: Could not find files for the given pattern(s).

C:\Windows\System32>where pip
INFO: Could not find files for the given pattern(s).

C:\Windows\System32>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>python -m site
'python' is not recognized as an internal or external command,
operable program or batch file.
'''
Starting to think that my python and pip installs are completely messed up. I know they are in my C:/user/xxx/onedrive because when installing I didn't want to take up a ton of local memory. 

I you check out my log update I ran an upgrade to python (successfully) then tried to install pip using chocolatey and it threw the error
'''
ERROR: Python is not installed. easy_install installation aborted!
The install of easy.install was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\easy.install\tools\chocolateyInstall.ps1'.
 See log for details.
'''
I'm going to look into it. In the meantime, your thoughts?

GhengisGohan, 2020-05-21:
@seanyen 
After running your previous tests in my regular cmd I got the following. 
```
C:\Users\Soren F Anderson>where python
C:\Python38\python.exe
C:\Users\Soren F Anderson\OneDrive\Documents\Anaconda 3\python.exe

C:\Users\Soren F Anderson>where pip
C:\Python38\Scripts\pip.exe

C:\Users\Soren F Anderson>pip --version
pip 19.2.3 from c:\python38\lib\site-packages\pip (python 3.8)

C:\Users\Soren F Anderson>python -m site
sys.path = [
    'C:\\Users\\Soren F Anderson',
    'C:\\Python38\\python38.zip',
    'C:\\Python38\\DLLs',
    'C:\\Python38\\lib',
    'C:\\Python38',
    'C:\\Python38\\lib\\site-packages',
]
USER_BASE: 'C:\\Users\\Soren F Anderson\\AppData\\Roaming\\Python' (exists)
USER_SITE: 'C:\\Users\\Soren F Anderson\\AppData\\Roaming\\Python\\Python38\\site-packages' (doesn't exist)
ENABLE_USER_SITE: True
```

seanyen, 2020-05-21:
@GhengisGohan Based on the `ros-catkin-tools` failure messages, I think you can find a Python installed under `c:\opt\python27amd64`. If it exists, can you try to run the following in a clean command prompt?

```Batchfile
set "PATH=c:\opt\python27amd64;c:\opt\python27amd64\Scripts;%PATH%"
set "PYTHONHOME=c:\opt\python27amd64"

pip install -U -r rosdep rosinstall_generator wstool rosinstall --disable-pip-version-check --no-cache-dir
```

I am trying to synthesize the same commands that the install script does, and try to see what's happened there.

GhengisGohan, 2020-05-21:
```
C:\Windows\System32>set "PATH=c:\opt\python27amd64;c:\opt\python27amd64\Scripts;%PATH%"

C:\Windows\System32>set "PYTHONHOME=c:\opt\python27amd64"

C:\Windows\System32>pip install -U -r rosdep rosinstall_generator wstool rosinstall --disable-pip-version-check --no-cache-dir
Traceback (most recent call last):
  File "c:\opt\python27amd64\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\opt\python27amd64\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "c:\opt\python27amd64\Scripts\pip.exe\__main__.py", line 5, in <module>
ImportError: cannot import name `main```
```
This is what came back.

seanyen, 2020-05-21:
@GhengisGohan That's some interesting information. Can you do `python -m site` on the same command prompt?

Also, I noticed there is a `USER_BASE` pointing to `C:\\Users\\xxx\\AppData\\Roaming\\Python`. What if you temporarily rename it and retry to `pip install ...` again?

GhengisGohan, 2020-05-21:
This is what I got from "python -m site" in the cmd. 
```
C:\Users\Soren F Anderson>python -m site
sys.path = [
    'C:\\Users\\Soren F Anderson',
    'C:\\Python38\\python38.zip',
    'C:\\Python38\\DLLs',
    'C:\\Python38\\lib',
    'C:\\Python38',
    'C:\\Python38\\lib\\site-packages',
]
USER_BASE: 'C:\\Users\\Soren F Anderson\\AppData\\Roaming\\Python' (exists)
USER_SITE: 'C:\\Users\\Soren F Anderson\\AppData\\Roaming\\Python\\Python38\\site-packages' (doesn't exist)
ENABLE_USER_SITE: True
```
in the VS Command Prompt:
```
C:\Windows\System32>python -m site
sys.path = [
    'C:\\Windows\\System32',
    'c:\\opt\\python27amd64\\python27.zip',
    'c:\\opt\\python27amd64\\DLLs',
    'c:\\opt\\python27amd64\\lib',
    'c:\\opt\\python27amd64\\lib\\plat-win',
    'c:\\opt\\python27amd64\\lib\\lib-tk',
    'c:\\opt\\python27amd64',
    'C:\\Users\\Soren F Anderson\\AppData\\Roaming\\Python\\Python27\\site-packages',
    'c:\\opt\\python27amd64\\lib\\site-packages',
]
USER_BASE: 'C:\\Users\\Soren F Anderson\\AppData\\Roaming\\Python' (exists)
USER_SITE: 'C:\\Users\\Soren F Anderson\\AppData\\Roaming\\Python\\Python27\\site-packages' (exists)
ENABLE_USER_SITE: True
```
At least the VS Command Prompt is recognizing python as opposed to before. 
Can you be more specific about renaming the USER_BASE and then running the pip install?

seanyen, 2020-05-21:
@GhengisGohan Sorry I was not clear about the instructions. Can you run the following in a clean command prompt:

```Batchfile
set "PATH=c:\opt\python27amd64;c:\opt\python27amd64\Scripts;%PATH%"
set "PYTHONHOME=c:\opt\python27amd64"

python -s -m pip install -U -r rosdep rosinstall_generator wstool rosinstall --disable-pip-version-check --no-cache-dir
```

This time I added `python -s` to explicitly disable the usage of [USER_SITE](https://docs.python.org/2/library/site.html#site.ENABLE_USER_SITE). I am interested in seeing if it is to do with the pip throwing the `ImportError`.

GhengisGohan, 2020-05-22:
morning at @seanyen looks like it didn't like rosdep. do we need to give it a rosdep location? I have the rosdep install and some rosdep_db but no actual .exe if that's what you're trying to call. If we look back in the logs, rosdepInstall.bat fails every time I call for choco upgrade ros-melodic-desktop_full. 

```
C:\Windows\System32>python -s -m pip install -U -r rosdep rosinstall_generator wstool rosinstall --disable-pip-version-check --no-cache-dir
Could not open requirements file: [Errno 2] No such file or directory: 'rosdep'
```

seanyen, 2020-05-22:
@GhengisGohan Sorry for the typo there. Can you remove `-r` from the command? Like running `python -s -m pip install -U rosdep rosinstall_generator wstool rosinstall --disable-pip-version-check --no-cache-dir`.

GhengisGohan, 2020-05-22:
@seanyen That worked. Tremendous. Correct me I'm wrong but I should be able to go through with the choco ros install now? Looks like ros-catkin-tools and rosdep showed up in my python scripts directory.

GhengisGohan, 2020-05-28:
@seanyen @ooeygui Morning gents, any chance we can pick up where we left off?
I was thinking about installing Linux on a SD card and just running ROS on that when I need it.

seanyen, 2020-05-28:
@GhengisGohan since it looked like that your USER_BASE might have something to interfere with the installation, can you try to remove the `c:\opt` and do the following in an opened Visual Studio Command Prompt as Administrator?

```Batchfile
set PYTHONNOUSERSITE=1
mkdir c:\opt\chocolatey
set ChocolateyInstall=c:\opt\chocolatey
choco source add -n=ros-win -s="https://roswin.azurewebsites.net/api/v2" --priority=1
choco upgrade ros-melodic-desktop_full -y --execution-timeout=0
```

GhengisGohan, 2020-06-05:
Some how - one day... It just happened. I included a -n and -v just for some more debugging and it just forced it's way through. 

```
C:\Windows\System32>choco upgrade ros-melodic-desktop_full -y -v -n --execution-timout=0
Chocolatey v0.10.15
Upgrading the following packages:
ros-melodic-desktop_full
By upgrading you accept licenses for the packages.
ros-melodic-desktop_full is not installed. Installing...
[NuGet] Attempting to resolve dependency 'ros-catkin-tools (≥ 0.0.1.1903221831)'.
[NuGet] Attempting to resolve dependency 'ros-python2 (≥ 2.7.15)'.
[NuGet] Attempting to resolve dependency 'vcredist140 (≥ 14.16.27024.1)'.
[NuGet] Attempting to resolve dependency 'chocolatey-core.extension (≥ 1.3.3)'.
[NuGet] Attempting to resolve dependency 'KB3033929 (≥ 1.0.3)'.
[NuGet] Attempting to resolve dependency 'chocolatey-windowsupdate.extension (≥ 1.0.4)'.
[NuGet] Attempting to resolve dependency 'KB3035131 (≥ 1.0.3)'.
[NuGet] Attempting to resolve dependency 'KB2919355 (≥ 1.0.20160915)'.
[NuGet] Attempting to resolve dependency 'KB2919442 (≥ 1.0.20160915)'.
[NuGet] Attempting to resolve dependency 'kb2999226 (≥ 1.0.20170509)'.
Progress: Downloading ros-melodic-desktop_full 20200501.1... 100%
Progress: Downloading ros-melodic-desktop_full 20200501.1... 100%
[NuGet] Attempting to resolve dependency 'ros-vcpkg (≥ 2018.11.23.1906112226)'.
[NuGet] Attempting to resolve dependency 'vcredist2010 (≥ 10.0.40219.2)'.
[NuGet] Installing 'ros-catkin-tools 0.0.1.1903221831'.
[NuGet] Successfully installed 'ros-catkin-tools 0.0.1.1903221831'.

ros-catkin-tools v0.0.1.1903221831
ros-catkin-tools package files upgrade completed. Performing other installation steps.
 The upgrade of ros-catkin-tools was successful.
  Software install location not explicitly set, could be in package or
  default install location if installer.
[NuGet] Installing 'ros-melodic-desktop_full 20200501.1'.
[NuGet] Successfully installed 'ros-melodic-desktop_full 20200501.1'.

ros-melodic-desktop_full v20200501.1
ros-melodic-desktop_full package files upgrade completed. Performing other installation steps.
 The upgrade of ros-melodic-desktop_full was successful.
  Software install location not explicitly set, could be in package or
  default install location if installer.

Chocolatey upgraded 2/2 packages.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).
```

Aman-Makkar, 2020-11-22:
Hello, I am facing issues with installing vcpkg, can you please help me through it? Please find the log attached.

[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/5579683/chocolatey.log)

ooeygui, 2020-11-30:
@Aman-Makkar it looks like you are using the older distribution point. Could you modify it using the following:

```
choco source remove -n=ros-win
choco source add -n=ros-win -s="https://aka.ms/ros/public" --priority=1
```

IoTDan, 2022-10-11:
this meta-ticket has lasted it's course. If you have any install/setup issues of any version of ROS on Windows distributed through Chocolatey, please open up a new issue.

closed 2022-10-11 by IoTDan