ms-iot/ROSOnWindows #153: ROS Install can "silently" fail if Desktop C++ package isn't installed in Visual Studio

michaelajax, 2019-09-26:
**Describe the bug**
A requirement to install ROS, as documented [here](https://wiki.ros.org/Installation/Windows), is to `Include "Desktop development with C++" workload` when installing VS2019.

Hypothetically - definitely not if you're me - someone could think that they had the C++ toolchain installed, but not actually have it installed. When this happens, the install "silently" fails - as in, `ros-vcpkg` will throw errors, but then will show `The upgrade of ros-vckpg was successful`:

```
Upgrading the following packages:
ros-vcpkg
By upgrading you accept licenses for the packages.
ros-vcpkg is not installed. Installing...
Progress: Downloading ros-vcpkg 2018.11.23.1906112226... 100%
Progress: Downloading ros-vcpkg 2018.11.23.1906112226... 100%
 
ros-vcpkg v2018.11.23.1906112226
ros-vcpkg package files upgrade completed. Performing other installation steps.
A subdirectory or file c:\opt already exists.
Cloning into 'vcpkg'...
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 64642 (delta 8), reused 0 (delta 0), pack-reused 64624
Receiving objects: 100% (64642/64642), 16.29 MiB | 25.09 MiB/s, done.
Resolving deltas: 100% (40563/40563), done.
Updating files: 100% (3969/3969), done.
 
Building vcpkg.exe ...
 
C:\opt\vcpkg\toolsrc\vcpkglib\vcpkglib.vcxproj(27,3): error MSB4019: The imported project "C:\Program Files (x86)\
Microsoft Visual Studio\2019\Preview\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.Default.props" was not found. Confirm
 that the path in the <Import> declaration is correct, and that the file exists on disk.
C:\opt\vcpkg\toolsrc\vcpkg\vcpkg.vcxproj(27,3): error MSB4019: The imported project "C:\Program Files (x86)\Micros
oft Visual Studio\2019\Preview\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.Default.props" was not found. Confirm that
the path in the <Import> declaration is correct, and that the file exists on disk.
C:\opt\vcpkg\toolsrc\vcpkgmetricsuploader\vcpkgmetricsuploader.vcxproj(27,3): error MSB4019: The imported project
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Preview\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.Default.props
" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
C:\opt\vcpkg\scripts\bootstrap.ps1 : Building vcpkg.exe failed. Please ensure you have installed Visual Studio
with the Desktop C++ workload and the Windows SDK for Desktop C++.
At line:1 char:4
+ & {& 'c:\opt\vcpkg\scripts\bootstrap.ps1' }
+    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Write-Error], WriteErrorException
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorException,bootstrap.ps1
 
 The upgrade of ros-vcpkg was successful.
  Software install location not explicitly set, could be in package or
  default install location if installer.
```

Someone not looking closely will think their install was successful, and then be confused when things don't work later on.

To make matters slightly worse, if you install the C++ toolchain and then try to re-build `ros-vcpkg` - even if you fully uninstall the package and then reinstall - it will silently fail to build the EXE, since the build script simply checks for `c:\opt\vcpkg` being installed:

```
WARNING: Vcpkg may already have be installed under c:\opt\vcpkg. Considered it is installed
```

At a minimum, I think we should update the `ros-vcpkg` script to explicitly check for `vcpkg.exe` when deciding if it needs to rebuild. Also, we should consider just failing the entire `ros-vcpkg` installation if the C++ toolchain - assuming we give clear direction to users on how to resolve and finish the ROS installation process when this happens. Consider that the installation process of ROS as a whole (`choco upgrade ros-melodic-desktop_full` blah blah) would be interrupted if this module fails. Luckily today it's the last thing that installs, but I don't think we should always rely on that.

**To Reproduce**
See the above description, but essentially install ROS on a machine that has VS19 installed but not the C++ toolchain. Presumably it would fail if VS19 is totally uninstalled as well - another failure mode we should consider

**Expected behavior**
See bug description - user is clearly informed of the issue and told exactly how to address and resume installation.

**Environment information:**
 - OS version: 1903 build 18362.356
 - Chocolatey Versions:
```
C:\Users\miajax>choco list --local-only
Chocolatey v0.10.15
assimp 4.0.1.1809242322
boost 1.66.0.1904102339
bullet3 2.87.0.1809242326
bzip2 1.0.6.1901242108
chocolatey 0.10.15
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.4
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1903240636
git 2.23.0
git.install 2.23.0
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1904131648
hg 5.1
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1903221831
ros-cmake 3.11.4.1809242322
ros-melodic-desktop_full 20190828.1-pre
ros-python2 2.7.15.1904300710
ros-vcpkg 2018.11.23.1906112226
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.22.27821
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
74 packages installed.
```

dweekly, 2020-04-12:
I ran into this issue when installing ROS for the first time. Had to manually delete c:\opt\vcpkg to successfully reinstall vcpkg.exe

dsato9, 2020-04-25:
I am currently having a similar problem. 
```
ERROR: rosdepInstall.bat failed.
The upgrade of ros-melodic-desktop_full was NOT successful.
Error while running 'c:\opt\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.

Chocolatey upgraded 0/1 packages. 1 packages failed.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

Failures
[chocolatey_log1.txt](https://github.com/ms-iot/ROSOnWindows/files/4532266/chocolatey_log1.txt)

 - ros-melodic-desktop_full (exited -1) - Error while running 'c:\opt\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.
```
Attached are my logs

ooeygui, 2020-04-27:
Hi @dsato9 Sorry for the delay, 
Thank you for posting the log files. There is an error emanating from rosdepInstall.bat, but chocolatey didn't capture the output in the chocolatey log. 

Can you run:
`c:\opt\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat`

dsato9, 2020-04-27:
I tired to run 
`c:\opt\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat`

however i received the following error code, and have tried manually checking the file path and i do not have "ros-melodic-desktop-full" in the file path
[image]
[image]

ooeygui, 2020-04-27:
Chocolatey sometimes will move a package that failed into a lib-bad path, if you have:
`c:\opt\chocolatey\lib-bad\ros-melodic-desktop_full\tools\rosdepInstall.bat`

Can you run it and check the output?

dsato9, 2020-04-27:
it would run, then get a point where it would display something about depreciation, python 2.7.... even though i do have python 3.7 installed. then close my command prompt window with out any additional prompts

ooeygui, 2020-04-27:
When you run `python --version` in that command window, what version is output?

dsato9, 2020-04-27:
> 
> 
> When you run `python --version` in that command window, what version is output?

it does show 2.7.15. but my python is 3.7.5
[image]

ooeygui, 2020-04-27:
ROS Melodic isn't supported on Python 3.7 which is why the ROS installer injects 2.7 into c:\opt. Are you attempting to run the commands in a python 3.7 console?

ROS Noetic will support Python 3.7+

dsato9, 2020-04-28:
I am trying to run ros melodic because other parts of the project that i am working on are built on melodic. are there compatibility issues if i use noetic?

ooeygui, 2020-04-28:
There are compatibility problems going across distros, so is not generally supported. 

I think what has happened is that Python 3 is being picked up instead of python 2.7, which would affect rosdep install. 

Can you remove python 3 from your system path and relaunch the command window?

KamaljeetSahoo, 2020-05-09:
> I am trying to run ros melodic because other parts of the project that i am working on are built on melodic. are there compatibility issues if i use noetic?
> 
> no, i am using the windows command prompt

I am having the same error; Please, fix it.

ooeygui, 2020-05-10:
@KamaljeetSahoo Please remove Python3 from your host environment, and relaunch the windows command prompt. Melodic does not support Python3, noetic will.

ooeygui, 2020-05-11:
@KamaljeetSahoo Which error are you seeing? Can you attach your chocolatey log?

Aman-Makkar, 2020-11-21:
Hello I am still having an issue with this, can you please advise? Please find a copy of the log attached 
[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/5577942/chocolatey.log)

IoTDan, 2022-10-11:
we no longer require a build during the install process. So closing this bug.

closed 2022-10-11 by IoTDan