ms-iot/ROSOnWindows #158: Catkin bug fix in upstream for INTERFACE libraries

johnwason, 2019-10-08:
A bug in Catkin was fixed in https://github.com/ros/catkin/pull/1034. Can the Windows choco repository be updated to include this fix?

**Describe the bug**
When using add_library with an INTERFACE library, it will fail on Windows.

**To Reproduce**
Add an INTERFACE library and attempt to build using catkin.

**Expected behavior**
add_library silently succeeds

seanyen, 2019-10-08:
@johnwason In general, when you install `ROS on Windows`, you can do `choco upgrade ros-melodic-desktop_full -y --pre` to get the daily builds from the `pre-release` channel, which are built from the upstream sources (with some exceptions, but we are using the upstream `catkin`) and you should be able to get the fix there.

And for this fix, since it just gets merged today, you may need to wait and get the fix from the next daily build.

seanyen, 2019-10-10:
@johnwason The fix should be in the latest `prerelease` build. I will close it for now, and feel free to re-open if there is any problems.

closed 2019-10-10 by seanyen