ms-iot/ROSOnWindows #159: CMake Error on catkin_make

rachelhson, 2019-10-10:
**Describe the bug**

I tried to catkin_make, but CMake Error happen on following directory

`CMake Error at C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find BLAS (missing: BLAS_LIBRARIES)

-- Configuring incomplete, errors occurred!
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeError.log".
NMAKE : fatal error U1077: 'C:\opt\python27amd64\Lib\site-packages\cmake\data\bin\cmake.exe' : return code '0x1'
Stop.
Invoking "nmake cmake_check_build_system" failed`

**To Reproduce**
Steps to reproduce the behavior:
* Catkin workspace setup instructions
  * Clone locations C:\opt\ros\melodic\catkin_ws\src
  * build command line 
  * Launch command line 

**Expected behavior**
successfully catkin_make 

**Environment information:**
 - OS version: `melodic`
 - Chocolatey Versions: Chocolatey v0.10.15

seanyen, 2019-10-10:
Can you post the full `catkin_make` logs? Do you have `openblas` listed after you run `choco list -lo`?

rachelhson, 2019-10-10:
```
C:\opt\ros\melodic\catkin_ws>catkin_make
Base path: C:\opt\ros\melodic\catkin_ws
Source space: C:\opt\ros\melodic\catkin_ws\src
Build space: C:\opt\ros\melodic\catkin_ws\build
Devel space: C:\opt\ros\melodic\catkin_ws\devel
Install space: C:\opt\ros\melodic\catkin_ws\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\opt\ros\melodic\catkin_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.23.28105.4
Copyright (C) Microsoft Corporation.  All rights reserved.

CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using CATKIN_DEVEL_PREFIX: C:/opt/ros/melodic/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64;C:/opt/vcpkg/installed/x64-windows
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Found PythonInterp: C:/opt/python27amd64/python.exe (found suitable version "2.7.15", minimum required is "2")
-- Using PYTHON_EXECUTABLE: C:/opt/python27amd64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/python27amd64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/opt/ros/melodic/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/python27amd64/Scripts/nosetests-2.7.exe
-- catkin 0.7.19
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 16 packages in topological order:
-- ~~  - kinova_bringup
-- ~~  - kinova_control
-- ~~  - kinova_gazebo
-- ~~  - kinova_msgs
-- ~~  - kinova_driver
-- ~~  - kinova_demo
-- ~~  - j2n6s300_ikfast
-- ~~  - j2s6s300_ikfast
-- ~~  - j2s7s300_ikfast
-- ~~  - m1n6s300_ikfast
-- ~~  - kinova_arm_moveit_demo
-- ~~  - kinova_description
-- ~~  - j2n6s300_moveit_config
-- ~~  - j2s6s300_moveit_config
-- ~~  - j2s7s300_moveit_config
-- ~~  - m1n6s300_moveit_config
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'kinova_bringup'
-- ==> add_subdirectory(kinova-ros/kinova_bringup)
-- +++ processing catkin package: 'kinova_control'
-- ==> add_subdirectory(kinova-ros/kinova_control)
-- +++ processing catkin package: 'kinova_gazebo'
-- ==> add_subdirectory(kinova-ros/kinova_gazebo)
-- +++ processing catkin package: 'kinova_msgs'
-- ==> add_subdirectory(kinova-ros/kinova_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action kinova_msgs/ArmJointAngles C:/opt/ros/melodic/catkin_ws/src/kinova-ros/kinova_msgs/action/ArmJointAngles.action
-- Generating .msg files for action kinova_msgs/ArmPose C:/opt/ros/melodic/catkin_ws/src/kinova-ros/kinova_msgs/action/ArmPose.action
-- Generating .msg files for action kinova_msgs/Arm_KinovaPose C:/opt/ros/melodic/catkin_ws/src/kinova-ros/kinova_msgs/action/Arm_KinovaPose.action
-- Generating .msg files for action kinova_msgs/SetFingersPosition C:/opt/ros/melodic/catkin_ws/src/kinova-ros/kinova_msgs/action/SetFingersPosition.action
-- kinova_msgs: 35 messages, 12 services
-- +++ processing catkin package: 'kinova_driver'
-- ==> add_subdirectory(kinova-ros/kinova_driver)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PythonInterp: C:/opt/python27amd64/python.exe (found version "2.7.15")
-- +++ processing catkin package: 'kinova_demo'
-- ==> add_subdirectory(kinova-ros/kinova_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'j2n6s300_ikfast'
-- ==> add_subdirectory(kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find BLAS (missing: BLAS_LIBRARIES)
Call Stack (most recent call first):
  C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindBLAS.cmake:806 (find_package_handle_standard_args)
  C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindLAPACK.cmake:193 (find_package)
  kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeLists.txt:28 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeError.log".
NMAKE : fatal error U1077: 'C:\opt\python27amd64\Lib\site-packages\cmake\data\bin\cmake.exe' : return code '0x1'
Stop.
Invoking "nmake cmake_check_build_system" failed
```

This is full of `catkin_make` logs. and I do not have `openblas` in `choco list -lo`.

seanyen, 2019-10-10:
@rachelhson can you do `choco install openblas -y` and rebuild it again?

rachelhson, 2019-10-10:
I resolve the issue with  `choco install openblas -y` but i have another CMake Error with LAPACK API.
```
CMake Error at C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindLAPACK.cmake:417 (message):
  A required library with LAPACK API not found.  Please specify library
  location.
Call Stack (most recent call first):
  kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeLists.txt:28 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeError.log".
NMAKE : fatal error U1077: 'C:\opt\python27amd64\Lib\site-packages\cmake\data\bin\cmake.exe' : return code '0x1'
Stop.
Invoking "nmake cmake_check_build_system" failed
```

seanyen, 2019-10-10:
@rachelhson Hi, can you try to do `choco install lapack`? We do have some dependencies hosted on `https://roswin.azurewebsites.net/`, and you can do a quick search to shop for what you need.

If it doesn't exist, you can always go with [`vcpkg`](https://github.com/microsoft/vcpkg) to install the missing packages from your ROS command prompt (by default we install a copy of vcpkg under `c:\opt\vcpkg` for your ROS installation side-by-side.

rachelhson, 2019-10-11:
I resolved for Lapack using `choco install clapack` 

I did not know i could find dependencies in azurewebsites. 

Now my ros in window is ready to run !!  

Thank You!!!

seanyen, 2019-10-14:
@rachelhson Just want to check back, does installing `lapack` solve the problem?

rachelhson, 2019-10-15:
@seanyen  Install `lapack` using `choco install clapack` works for me. 
currently I am facing different issue in catkin_make 
```
[ 85%] Building CXX object kinova-ros/kinova_driver/CMakeFiles/kinova_driver.dir/src/kinova_api.cpp.obj
kinova_api.cpp
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C2374: '__attribute__': redefinition; multiple initialization
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C2374: '__attribute__': redefinition; multiple initialization
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C2374: '__attribute__': redefinition; multiple initialization
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C2374: '__attribute__': redefinition; multiple initialization
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C2146: syntax error: missing ';' before identifier 'Packet'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C2374: '__attribute__': redefinition; multiple initialization
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(327): error C3861: 'visibility': identifier not found
[... cut ...]
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed
```
I posted the issue in manufacturer's github, but I want to make sure whether there is anything I can try to fix this issue. Please let me know if you have an idea. 

Thank you for your help.

ooeygui, 2019-10-17:
I'm going to rename this issue to constrain it to blas.

ooeygui, 2019-10-17:
@rachelhson thanks for confirming on lapack. The issue you are hitting now is that Kinova has code which is specific to the GCC compiler -  `__attribute__ ((visibility ("default")))`. To port the kinova code to Windows. The ROS2 model for this is to use a `visibility_control.h` header, which handles the platform switch. Check out `https://github.com/ros2/demos/blob/master/image_tools/include/image_tools/visibility_control.h`

(it looks like kinova started to do this work, there's a comment here: `https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/include/kinova/Kinova.API.USBCommLayerUbuntu.h`)

seanyen, 2020-02-25:
Closing it since the original `BLAS` issue was solved.

closed 2020-02-25 by seanyen