ms-iot/ROSOnWindows #161: Moveit Tutorial links broken and build does not work

nikhiljgeorge, 2019-11-02:
I installed ROS using instructions here - http://wiki.ros.org/Installation/Windows. Installation seemed to go fine.

I then tried to execute the instructions given here https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html
The instructions failed at step git clone https://github.com/ms-iot/moveit_tutorials -b init_windows - apparently the link is invalid. Could you please fix the link.

Here I went off the beaten path and instead cloned git clone https://github.com/ros-planning/moveit_tutorials because this just seemed to be documentation.

Then I went ahead and ran the following commands from the instructions:
> git clone https://github.com/ms-iot/panda_moveit_config -b init_windows
> git clone https://github.com/frankaemika/franka_ros-release -b release/kinetic/franka_description
> cd c:\moveit_ws
> catkin_make

Now catkin_make fails with the following error (possibly linked to me going off the beaten path above and cloning the wrong repo?) in CMakeError.txt:
Determining if the include file pthread.h exists failed with the following output:
Change Dir: C:/moveit_ws/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_6f82a\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe" -f CMakeFiles\cmTC_6f82a.dir\build.make /nologo -L                  CMakeFiles\cmTC_6f82a.dir\build

Building C object CMakeFiles/cmTC_6f82a.dir/CheckIncludeFile.c.obj

	C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1423~1.281\bin\Hostx64\x64\cl.exe @C:\Users\nikhi\AppData\Local\Temp\nmC2E2.tmp

CheckIncludeFile.c

C:\moveit_ws\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory

NMAKE : fatal error U1077: 'C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1423~1.281\bin\Hostx64\x64\cl.exe' : return code '0x2'

Stop.

NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'

Stop. 

**Environment information:**
 - OS version: Ver 1903 OS Build 18362.418
 - Chocolatey Versions: Chocolatey v0.10.15

ooeygui, 2019-11-04:
Hi @nikhiljgeorge 
Thanks for the report. We will the upstream documentation updated and links corrected.

pthread isn't available on Windows, so we have been migrating instances of it to boost threads, which offer equivalent cross platform capabilities. 

I do see a Kinetic link in there - Sorry for the confusion - Windows is only supported on Melodic forward.

SlyLi, 2020-03-03:
Hi @nikhiljgeorge @ooeygui
 I have met the same error, so how could I make it work?

ooeygui, 2020-03-03:
Reopening - fix in flight.

SlyLi, 2020-03-03:
Actually I have tried to solve it by installing pthread on my window. It didn't work and generated cmakeerror.log and cmapkeoutput.log in catkin_win\build\CMakeFiles. Then, I add a piece of code to pthread.h .

```
#if !defined(PTHREADLIB)
#define PTHREADLIB
#define pthread_atfork(F1, F2, F3) 0
#pragma comment(lib, "pthreadVC2.lib")
#endif
```
I tried to run it again. It still didn't work, but it only generated camkeoutput.log

lilustga, 2020-03-03:
Try building only the moveit tutorial package and it's dependencies:
catkin_make_isolated --only-pkg-with-deps moveit_tutorials --install

seanyen, 2020-09-22:
Closing it. The MoveIt tutorials have been switched to use the upstream repositories.

closed 2020-09-22 by seanyen