ms-iot/ROSOnWindows #164: vcpkg dll directory is not added to the path

traversaro, 2019-11-24:
**Describe the bug**
I installed the ros-vcpkg chocolatey package, and used the vcpkg installed at `c:\opt\vcpkg` to install some vcpkg libraries using the `x64-windows` triplet. However, I can't use them without adding to the path the `c:\opt\vcpkg\installed\x64-windows\bin` directory that contains the installed Release dll. 

**To Reproduce**
* Install `ros-vcpkg` and some libraries such as `ace` 
* Call `c:\opt\ros\melodic\x64\setup.bat`
* Inspect the `PATH` env variable

**Expected behavior**
I would expect the `c:\opt\vcpkg\installed\x64-windows\bin` directory to be part of the path, as the `c:\opt\vcpkg` is, to permit to use the libraries installed by vcpkg. 

**Environment information:**
 - OS version: Version 1709 (OS Build 16299.192)
 - Chocolatey Versions:
~~~
c:\opt\vcpkg\installed\x64-windows\bin>choco list --local-only
Chocolatey v0.10.15
assimp 4.0.1.1809242322
boost 1.66.0.1904102339
chocolatey 0.10.15
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.4
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.4.1.1909060324
git 2.24.0
git.install 2.24.0
hg 5.2
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jq 1.6
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libwebp 0.6.1.1811210731
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
pkg-config 0.29.2.1809242322
protobuf 3.6.1.1811130545
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1903221831
ros-melodic-desktop_full 20190729.1
ros-python2 2.7.15.1904300710
ros-vcpkg 2018.11.23.1906112226
sbcl 0.0.0.1809242326
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml2 6.1.0.1809242322
unzip 6.0
urdfdom 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.22.27821
Wget 1.20.3.20190531
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
56 packages installed.
~~~

ooeygui, 2019-11-25:
Could you share which github repository you were using which consumed ace?

ooeygui, 2019-11-25:
We have seen some vcpkg ports not respecting the RelWithDebInfo flag - generating a debug version which is dropped into the install bin directory. I'm working on this, but I suspect this will have to be fixed on a per port basis :-(

traversaro, 2019-11-25:
> Could you share which github repository you were using which consumed ace?

I was trying to make sure that 'YARP ( https://github.com/robotology/yarp ) could be compiled easily in a ROSonWindows workspace.

traversaro, 2019-11-25:
> Could you share which github repository you were using which consumed ace?

I was trying to make sure that 'YARP ( https://github.com/robotology/yarp ) could be compiled easily in a ROSonWindows workspace. 

> We have seen some vcpkg ports not respecting the RelWithDebInfo flag - generating a debug version which is dropped into the install bin directory. I'm working on this, but I suspect this will have to be fixed on a per port basis :-(

seanyen, 2020-01-23:
This one is fixed in the latest prerelease build so I am closing it.

closed 2020-01-23 by seanyen