ms-iot/ROSOnWindows #165: Random nodelets "finished cleanly" at launch

songlan-hl, 2019-11-25:
**Describe the bug**
I wonder whether anyone has seen similar behaviors. I am using ROS nodelets as the basic operation units of my program. Usually there are about 10 nodelets loaded to one manager. I have been using this way for a while and have not had problems. 

Now I am running into strange problems when I test my ROS program on other computers. Every time I ran roslaunch, All nodelets were initialized successfully (according to log file), but after 10-15 seconds, two of the nodelets would die with "$$$ has finished cleanly" printed to the log. Every time it could be two different nodelets, and it seemed random.

**To Reproduce**
Steps to reproduce the behavior:
I don't know how to make this into a simple case for others to reproduce. We've tested the same program on different computers, and not all of them suffer from it. The first time we saw this was a couple of months ago on one PC. We couldn't resolve it but decided to move on to another computer since it was only one case and we just assumed something wrong in the settings. In the past week, several of our computers started this problem - some of them have been used to run this program for months. And once it started on one computer, it would not go away, even after we reverse the code to previous versions. It just kept happening every launch. 

**Expected behavior**
The expected behavior is obvious. I understand I am not providing much as I don't know where to look further. I would appreciate any suggestion to try out or get more information to help us diagnose. 

**Environment information:**
 - OS version: `winver`
 - Chocolatey Versions: `choco list --local-only`

songlan-hl, 2019-11-25:
This is the launch file if it helps.

<launch>

  
  <arg name="config" />
  <rosparam command="load" file="$(arg config)" />
  <arg name="workdir" default="" />
  <param name="job/log_directory" value="$(arg workdir)" unless="$(eval workdir == '')" />
  <arg name="datadir" default="" />
  <param name="output/result_folder" value="$(arg datadir)" unless="$(eval datadir == '')" />
  <arg name="modeldir" default="" />
  <param name="templates/detectors_yml" value="$(arg modeldir)/detectors.yaml" unless="$(eval modeldir == '')" />
  <param name="templates/process_yml" value="$(arg modeldir)/conf_inspect.yaml" unless="$(eval modeldir == '')" />

  
  <node pkg="nodelet" type="nodelet" name="MYPROGRAM_job_manager" args="manager">
    <param name="num_worker_threads" value="16" />
  </node>
  <node pkg="nodelet" type="nodelet" name="instructor" args="load MYPROGRAM_package/MYPROGRAMInstructionNodelet MYPROGRAM_job_manager" required="true" />
  <node pkg="nodelet" type="nodelet" name="imageReader" args="load MYPROGRAM_package/MYPROGRAMImageReaderNodelet MYPROGRAM_job_manager">
    <remap from="/imageReader/original_image" to="/MYPROGRAM/original_image" />
  </node>
  <node pkg="nodelet" type="nodelet" name="cameraController" args="load MYPROGRAM_package/MYPROGRAMCameraControllerNodelet MYPROGRAM_job_manager">
    <remap from="/cameraController/original_image" to="/MYPROGRAM/original_image" />
    <remap from="/cameraController/io_controller" to="/MYPROGRAM/io_controller" />
  </node>
  <node pkg="nodelet" type="nodelet" name="IOController" args="load MYPROGRAM_package/MYPROGRAMIOControllerNodelet MYPROGRAM_job_manager">
    <remap from="/IOController/io_controller" to="/MYPROGRAM/io_controller" />				 
  </node>
  <node pkg="nodelet" type="nodelet" name="dispatcher" args="load MYPROGRAM_package/MYPROGRAMDispatcherNodelet MYPROGRAM_job_manager">
    <remap from="/dispatcher/image_input" to="/MYPROGRAM/original_image" />
    <remap from="/dispatcher/output_0" to="/MYPROGRAM/to_inspector0" />
    <remap from="/dispatcher/output_1" to="/MYPROGRAM/to_inspector1" />
    <remap from="/dispatcher/output_2" to="/MYPROGRAM/to_inspector2" />
    <remap from="/dispatcher/output_3" to="/MYPROGRAM/to_inspector3" />
  </node>
  <node pkg="nodelet" type="nodelet" name="inspector0" args="load MYPROGRAM_package/MYPROGRAMInspectionNodelet MYPROGRAM_job_manager">
    <remap from="/inspector0/original_image" to="/MYPROGRAM/to_inspector0" />
    <remap from="/inspector0/inspection_result" to="/MYPROGRAM/defect" />
    <remap from="/inspector0/defect_image" to="/MYPROGRAM/defect_image" />
  </node>
  <node pkg="nodelet" type="nodelet" name="inspector1" args="load MYPROGRAM_package/MYPROGRAMInspectionNodelet MYPROGRAM_job_manager">
    <remap from="/inspector1/original_image" to="/MYPROGRAM/to_inspector1" />
    <remap from="/inspector1/inspection_result" to="/MYPROGRAM/defect" />
    <remap from="/inspector1/defect_image" to="/MYPROGRAM/defect_image" />
</node>
  <node pkg="nodelet" type="nodelet" name="inspector2" args="load MYPROGRAM_package/MYPROGRAMInspectionNodelet MYPROGRAM_job_manager">
    <remap from="/inspector2/original_image" to="/MYPROGRAM/to_inspector2" />
    <remap from="/inspector2/inspection_result" to="/MYPROGRAM/defect" />
    <remap from="/inspector2/defect_image" to="/MYPROGRAM/defect_image" />
</node>
  <node pkg="nodelet" type="nodelet" name="inspector3" args="load MYPROGRAM_package/MYPROGRAMInspectionNodelet MYPROGRAM_job_manager">
    <remap from="/inspector3/original_image" to="/MYPROGRAM/to_inspector3" />
    <remap from="/inspector3/inspection_result" to="/MYPROGRAM/defect" />
    <remap from="/inspector3/defect_image" to="/MYPROGRAM/defect_image" />
  </node>
  <node pkg="nodelet" type="nodelet" name="defectHandler" args="load MYPROGRAM_package/MYPROGRAMDefectHandlerNodelet MYPROGRAM_job_manager">
    <remap from="/defectHandler/original_image" to="/MYPROGRAM/original_image" />
    <remap from="/defectHandler/inspection_result" to="/MYPROGRAM/defect" />
    <remap from="/defectHandler/io_controller" to="/MYPROGRAM/io_controller" />
    <remap from="/defectHandler/defect_image" to="/MYPROGRAM/defect_image" />
    <remap from="/MYPROGRAM/defect_report" to="/MYPROGRAM/UI_GetResult" />
    <remap from="/MYPROGRAM/image_display" to="/MYPROGRAM/UI_ShowImage" />
  </node>
</launch>

seanyen, 2019-11-25:
@songlan-hl, Hi, thanks for the reporting. I would recommend to start from enabling more logging to see any obvious problems there. Can you try to do the following?

1. Create and use [`rosconsole.config`](http://wiki.ros.org/rosconsole#Configuration) to enable `nodelet` related `DEBUG` logging.

```
# Override nodelet package to output everything
log4j.logger.ros.nodelet=DEBUG
```

2. In the launch file, add `output="screen"` to the `nodelet` and force everything to `stdout` to be printed.

3. You may also try to use [ROS extension](https://aka.ms/ros/vscode) to do [debug from a launch file](https://github.com/ms-iot/vscode-ros/blob/master/doc/debug-support.md#launch) and see if any issues seen in the debugger.

Hope that helps the debugging.

songlan-hl, 2019-11-25:
Not much can be found here:

[ INFO] [1574716365.333913900]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1574716365.726890200]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.726612600]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.726756500]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.728351800]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.728490300]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.728919700]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.733331900]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.733579400]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716365.734704100]: waitForService: Service [/MYPROGRAM_job_manager/load_nodelet] is now available.
[ INFO] [1574716379.627049200]: Bond broken, exiting
[ INFO] [1574716379.637663900]: Bond broken, exiting
[IOController-6] process has finished cleanly
log file: C:\Users\DR\.ros\log\4a4ce291-0fc8-11ea-86d7-982cbc970c9c\IOController-6*.log
[inspector1-9] process has finished cleanly
log file: C:\Users\DR\.ros\log\4a4ce291-0fc8-11ea-86d7-982cbc970c9c\inspector1-9*.log

songlan-hl, 2019-11-25:
actually there is more in master.log

[rosmaster.master][INFO] 2019-11-26 05:33:21,351: publisherUpdate[/HLZN_job_manager/bond] -> http://DESKTOP-I78E87T:55221/ ['http://DESKTOP-I78E87T:55236/', 'http://DESKTOP-I78E87T:55199/', 'http://DESKTOP-I78E87T:55226/', 'http://DESKTOP-I78E87T:55201/', 'http://DESKTOP-I78E87T:55209/', 'http://DESKTOP-I78E87T:55185/', 'http://DESKTOP-I78E87T:55231/', 'http://DESKTOP-I78E87T:55195/', 'http://DESKTOP-I78E87T:55217/']: sec=8.01, exception=[Errno 10061] 
[rosmaster.threadpool][ERROR] 2019-11-26 05:33:21,354: Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosmaster\threadpool.py", line 218, in run
    result = cmd(*args)
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosmaster\master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "C:\opt\python27amd64\lib\xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "C:\opt\python27amd64\lib\xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "C:\opt\python27amd64\lib\xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "C:\opt\python27amd64\lib\xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "C:\opt\python27amd64\lib\xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "C:\opt\python27amd64\lib\httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "C:\opt\python27amd64\lib\httplib.py", line 882, in _send_output
    self.send(msg)
  File "C:\opt\python27amd64\lib\httplib.py", line 844, in send
    self.connect()
  File "C:\opt\python27amd64\lib\httplib.py", line 821, in connect
    self.timeout, self.source_address)
  File "C:\opt\python27amd64\lib\socket.py", line 575, in create_connection
    raise err
error: [Errno 10061]

songlan-hl, 2019-11-25:
could the problem be the connection timeout?

ooeygui, 2019-11-25:
@songlan-hl This log looks promising. It looks like the Windows firewall is blocking the connection. Could you try (in a secure environment) disabling the firewall and re-running to see if this reproduces?

songlan-hl, 2019-11-25:
I turned off windows defender firewall. And it didn't show any difference in the behaviors. 

Is there a way to extend the timeout limit?

songlan-hl, 2019-11-25:
I tried adding 11311 to the inbound rule of the firework as well and still no change. It is puzzling as other 8 nodelets apparently were successfully launched. And only 2...

songlan-hl, 2019-11-25:
Another thing about my setup: I used more than one publishers to publish to the same topic. It was working fine even though there were messages "ERROR: failed to advertise a service that is already advertised in this node". According to some general ROS forum this message can be ignored. And I didn't have problems on most of the computers before.  On the computer I am debugging now, its ros system was installed recently. Up to this week it was running ok although didn't show these ERROR messages. Now it's having problems....

ooeygui, 2019-11-25:
The Firewall has caused issues in the past, but that isn't it for this bug.
(Debugging nodelet failures is quite difficult, we'll look to see what we can do to help improve this).

A Nodelet is failing to load, which can be caused by a number of reasons:

- The nodelet is missing a dependency.
- What the nodelet is connecting to is failing.

**Debugging nodelets**
To check these, it is best to capture the node under a debugger. Sean mentioned vscode debugging. You can isolate the failing nodes into a single roslaunch file, then start the rest of the ros graph minus those nodes. You can then use the VSCode ROS extension to launch the nodelet manager under a debugger. 

**Dependencies**
If a nodelet is missing a dependency it will fail to load at all. This can be difficult to diagnose. Fortunately, the Windows Debugger tools has a feature called `gflags.exe`. Install the [Windows Debugger Tools](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/), then load gflags.exe. In the `image file execution options`, type `nodelet.exe`, then press *tab*. Select `Snow Loader Snaps`, which will dump every import and export. You can see at runtime what it was attempting to load - and what failed.

**Failing after startup**
If the nodelet is failing after startup, without good error messages, you'll need to step through the startup of the node itself. 

Would you be able to provide repro steps?

songlan-hl, 2019-11-26:
Thanks for taking the time. Looking through the possible reasons, I wonder why these would trigger "random" nodelets to stop working? Everytime the two nodeletes (now I have seen some cases failed with one nodelet in the past couple of hours) could be different.

ooeygui, 2019-11-26:
At this point, I don't think the issue is systemic, so I'd like to understand what's happening within the nodelets themselves. Could you provide a repro? (Are they in a public github we can investigate?)

songlan-hl, 2019-11-26:
I will try to simply our application into something easier to reproduce. 

In the meanwhile, I am tracing backward in ros system logs. The previous connection failure actually happened after the two nodes aborted, so it's not the cause. Around the time of abortion, master.log showed this:

> [rosmaster.master][INFO] 2019-11-26 08:39:34,213: publisherUpdate[/HLZN/original_cam_image] -> http://DESKTOP-I78E87T:57101/ []
> [rosmaster.master][INFO] 2019-11-26 08:39:34,214: publisherUpdate[/HLZN/original_cam_image] -> http://DESKTOP-I78E87T:57101/ []: sec=0.00, result=[1, '', 0]
> [rosmaster.master][INFO] 2019-11-26 08:39:35,740: -PUB [/rosout] /cameraController http://DESKTOP-I78E87T:57060/
> [rosmaster.master][INFO] 2019-11-26 08:39:35,740: -PUB [/HLZN_job_manager/bond] /cameraController http://DESKTOP-I78E87T:57060/
> [rosmaster.master][INFO] 2019-11-26 08:39:35,742: -SUB [/HLZN_job_manager/bond] /cameraController http://DESKTOP-I78E87T:57060/
> [rosmaster.master][INFO] 2019-11-26 08:39:35,743: -SERVICE [/cameraController/get_loggers] /cameraController rosrpc://DESKTOP-I78E87T:57059
> [rosmaster.master][INFO] 2019-11-26 08:39:35,744: -SERVICE [/cameraController/set_logger_level] /cameraController rosrpc://DESKTOP-I78E87T:57059

In another run where the cameraController node did not abort, the corresponding message is 

> [rosmaster.master][INFO] 2019-11-26 08:40:27,532: publisherUpdate[/HLZN/original_cam_image] -> http://DESKTOP-I78E87T:57534/ ['http://DESKTOP-I78E87T:57534/']
> [rosmaster.master][INFO] 2019-11-26 08:40:27,532: publisherUpdate[/HLZN/original_cam_image] -> http://DESKTOP-I78E87T:57534/ ['http://DESKTOP-I78E87T:57534/']: sec=0.00, result=[1, '', 0]
> 

where 57534 is related to job manager. I wonder if you have any quick thoughts about this.

ooeygui, 2020-07-02:
Nodelets need to be dropped into the correct path in order to be found. If you are still hitting this problem, please reopen with details about which nodelets have issues

closed 2020-07-02 by ooeygui