ms-iot/ROSOnWindows #167: [Choco upgrade ros-melodic-desktop_full] Gazebo9 installation failed

JoelTari, 2019-12-01:
**Bug description**
gazebo9 installation failed when doing the 'choco upgrade' command instruction on http://wiki.ros.org/Installation/Windows
> The upgrade of gazebo9 was NOT successful.
> gazebo9 not installed. An error occurred during installation:
>  Unable to resolve dependency 'hg (= 0.0.0)'.  

Several connection warnings occured prior:
> Error retrieving packages from source 'https://chocolatey.org/api/v2/':
>  Could not connect to the feed specified at 'https://chocolatey.org/api/v2/'. Please verify that the package source (located in the Package Manager Settings) is valid and ensure your network connectivity.

**Environment information:**
 - OS version: 1903
 - Chocolatey Versions: v0.10.15
 - No proxy

ooeygui, 2019-12-02:
I attempted to reproduce this error, but was unable to. It looks like a transient error on the main chocolatey feed, not related to the ROS feed. Can you attempt to reproduce? If you can, could you post the chocolatey log located in your install at `c:\ProgramData\chocolatey\logs\chocolatey.log`?

JoelTari, 2019-12-02:
>  It looks like a transient error on the main chocolatey feed, not related to the ROS feed.

It might be indeed. I reproduced the command, but this time it worked without any change. 

Thank you anyway, here is the log just in case:

> 2019-12-01 17:32:39,052 14184 [DEBUG] - _ Chocolatey:ChocolateyUpgradeCommand - Normal Run Mode _
2019-12-01 17:32:39,052 14184 [INFO ] - Upgrading the following packages:
2019-12-01 17:32:39,052 14184 [INFO ] - gazebo9
2019-12-01 17:32:39,052 14184 [INFO ] - By upgrading you accept licenses for the packages.
2019-12-01 17:32:39,114 14184 [WARN ] - gazebo9 is not installed. Installing...
2019-12-01 17:32:39,146 14184 [DEBUG] - Using 'https://roswin.azurewebsites.net/api/v2'.
2019-12-01 17:32:41,114 14184 [DEBUG] - - Supports prereleases? 'True'.
2019-12-01 17:32:41,130 14184 [DEBUG] - - Is ServiceBased? 'True'.
2019-12-01 17:32:42,005 14184 [DEBUG] - Package 'gazebo9' found on source 'https://roswin.azurewebsites.net/api/v2'
2019-12-01 17:32:42,005 14184 [DEBUG] - Using 'https://chocolatey.org/api/v2/'.
2019-12-01 17:32:42,395 14184 [DEBUG] - - Supports prereleases? 'True'.
2019-12-01 17:32:42,395 14184 [DEBUG] - - Is ServiceBased? 'True'.
2019-12-01 17:34:22,426 14184 [WARN ] - Error retrieving packages from source 'https://chocolatey.org/api/v2/':
 Could not connect to the feed specified at 'https://chocolatey.org/api/v2/'. Please verify that the package source (located in the Package Manager Settings) is valid and ensure your network connectivity.
2019-12-01 17:34:22,504 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'libpng (= 0.0.0)'.
2019-12-01 17:34:25,066 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'tiff (= 0.0.0)'.
2019-12-01 17:34:25,316 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'zlib (= 0.0.0)'.
2019-12-01 17:34:25,394 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'jxrlib (= 0.0.0)'.
2019-12-01 17:34:25,644 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'jasper (= 0.0.0)'.
2019-12-01 17:34:25,910 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'ilmbase (= 0.0.0)'.
2019-12-01 17:34:26,175 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'liblzma (= 0.0.0)'.
2019-12-01 17:34:26,441 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'libraw (= 0.0.0)'.
2019-12-01 17:34:26,706 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'libwebp (= 0.0.0)'.
2019-12-01 17:34:26,972 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'openexr (= 0.0.0)'.
2019-12-01 17:34:27,253 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'openjpeg (= 0.0.0)'.
2019-12-01 17:34:28,347 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'libjpeg-turbo (= 0.0.0)'.
2019-12-01 17:34:29,409 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'lcms (= 0.0.0)'.
2019-12-01 17:34:30,316 14184 [INFO ] - [NuGet] Attempting to resolve dependency 'hg (= 0.0.0)'.
2019-12-01 17:36:05,966 14184 [WARN ] - [NuGet] Could not connect to the feed specified at 'https://chocolatey.org/api/v2/'. Please verify that the package source (located in the Package Manager Settings) is valid and ensure your network connectivity.
2019-12-01 17:36:06,201 14184 [WARN ] - [NuGet] Could not connect to the feed specified at 'https://chocolatey.org/api/v2/'. Please verify that the package source (located in the Package Manager Settings) is valid and ensure your network connectivity.
2019-12-01 17:36:06,497 14184 [WARN ] - [NuGet] Could not connect to the feed specified at 'https://chocolatey.org/api/v2/'. Please verify that the package source (located in the Package Manager Settings) is valid and ensure your network connectivity.
2019-12-01 17:36:06,747 14184 [WARN ] - [NuGet] Could not connect to the feed specified at 'https://chocolatey.org/api/v2/'. Please verify that the package source (located in the Package Manager Settings) is valid and ensure your network connectivity.
2019-12-01 17:36:09,154 14184 [ERROR] - gazebo9 not installed. An error occurred during installation:
 Le serveur distant a retourné une erreur : (503) Serveur non disponible. Service Unavailable
2019-12-01 17:36:09,185 14184 [INFO ] - gazebo9 package files upgrade completed. Performing other installation steps.
2019-12-01 17:36:09,200 14184 [DEBUG] - No package information as package is null.
2019-12-01 17:36:09,232 14184 [DEBUG] - No package information to save as package is null.
2019-12-01 17:36:09,247 14184 [DEBUG] - Sending message 'HandlePackageResultCompletedMessage' out if there are subscribers...
2019-12-01 17:36:09,247 14184 [ERROR] - The upgrade of gazebo9 was NOT successful.
2019-12-01 17:36:09,263 14184 [ERROR] - gazebo9 not installed. An error occurred during installation:
 Le serveur distant a retourné une erreur : (503) Serveur non disponible. Service Unavailable
2019-12-01 17:36:09,279 14184 [WARN ] - 
Chocolatey upgraded 0/1 packages. 1 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
2019-12-01 17:36:09,279 14184 [INFO ] - 
2019-12-01 17:36:09,279 14184 [ERROR] - Failures
2019-12-01 17:36:09,294 14184 [ERROR] -  - gazebo9 (exited 1) - gazebo9 not installed. An error occurred during installation:
 Le serveur distant a retourné une erreur : (503) Serveur non disponible. Service Unavailable
2019-12-01 17:36:09,294 14184 [WARN ] - 
Enjoy using Chocolatey? Explore more amazing features to take your
experience to the next level at
 https://chocolatey.org/compare

ooeygui, 2019-12-04:
Thank you for letting us know that it was a transient error. Closing.

closed 2019-12-04 by ooeygui