ms-iot/ROSOnWindows #169: Running python files

OliHas, 2019-12-16:
Thank you for the 'Getting started/UsingROSOnWindows' Introduction.

I'm unable to run the following command:
```
c:\catkin_ws>rosrun beginner_tutorials talker.py
[rosrun] Couldn't find executable named talker.py
```
But this works:
```
c:\catkin_ws>rosrun beginner_tutorials scripts/talker.py
```

Regarding the [Python](https://github.com/ms-iot/ROSOnWindows/blob/master/GettingStarted/UsingROSonWindows.md#python) section, it is unclear to me, how I should rename the [python file](https://github.com/ms-iot/ROSOnWindows/blob/master/GettingStarted/UsingROSonWindows.md#python).

Is this related to this question [here](https://answers.ros.org/question/326072/windows10-rosrun-couldnt-find-python-scripts-in-path-scripts/)?

Could someone give any clarification?

ooeygui, 2019-12-16:
Thank you for the report. I think there was a regression in the rosrun.bat file - it should do a deep find on the script file. investigating.

ooeygui, 2020-07-02:
This has been corrected in later builds

ooeygui, 2020-11-09:
Reopening, as this appears to be broken on Noetic.

IoTDan, 2022-10-11:
Please re-open if this still repros

closed 2022-10-11 by IoTDan