ms-iot/ROSOnWindows #171: access denied when installing

CesMak, 2020-01-07:
Hey there,

when installing moveit in Windows10 I got access denied:

`C:\WINDOWS\system32>choco upgrade ros-melodic-moveit -y                                                                 Chocolatey v0.10.15                                                                                                     Upgrading the following packages:                                                                                       ros-melodic-moveit                                                                                                      By upgrading you accept licenses for the packages.                                                                      ros-melodic-moveit is not installed. Installing...                                                                      Progress: Downloading ros-melodic-moveit 20190729.1... 100%                                                             Progress: Downloading ros-melodic-moveit 20190729.1... 100%                                                                                                                                                                                     ros-melodic-moveit v20190729.1                                                                                          ros-melodic-moveit package files upgrade completed. Performing other installation steps.                                Extracting 64-bit C:\ProgramData\chocolatey\lib\ros-melodic-moveit\tools\drop.zip to c:\opt\ros...                      ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\cpp_common.dll                      ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\dynamic_reconfigure_config_init_mutex.dll                                                                                                                   ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\rosconsole.dll                      ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\rosconsole_backend_interface.dll    ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\rosconsole_log4cxx.dll              ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\roscpp.dll                          ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\roscpp_serialization.dll            ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\roslib.dll                          ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\rospack.dll                         ERROR: Can not delete output file : Zugriff verweigert : c:\opt\ros\melodic\x64\bin\rostime.dll    `  

any ideas how to solve these issues?

CesMak, 2020-01-07:
Could solve this issue. Just had to restart PC access was denied cause of gzserver still running in background. You can close this issue.

ooeygui, 2020-01-07:
@CesMak 
Thank you for the report. gzserver should shut down when the rest of the ROS graph does. We're going to be doing a Gazebo robustness spike in February, so we'll look into this.

seanyen, 2020-09-22:
Closing it. The latest installation (inno setup) will kill the running processes first to avoid the files locked issues like this.

closed 2020-09-22 by seanyen