ms-iot/ROSOnWindows #180: The code execution cannot proceed because ".dll" file was not found. Reinstalling the program may fix the problem.

qwe0254, 2020-01-24:
**Describe the bug**
The code execution cannot proceed because ".dll" file was not found. Reinstalling the program may fix the problem.

**To Reproduce**
# catkin_make
Base path: C:\temp_ws
Source space: C:\temp_ws\src
Build space: C:\temp_ws\build
Devel space: C:\temp_ws\devel
Install space: C:\temp_ws\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\temp_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.23.28107.0
Copyright (C) Microsoft Corporation.  All rights reserved.

####
#### Running command: "nmake" in "C:\temp_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.23.28107.0
Copyright (C) Microsoft Corporation.  All rights reserved.

[ 18%] Built target _setup_util.py_executable_install_python
[ 36%] Built target item
[ 36%] Built target std_msgs_generate_messages_cpp
[ 36%] Built target my_trailer_generate_messages_check_deps_Num
[ 45%] Built target my_trailer_generate_messages_cpp
[ 45%] Built target std_msgs_generate_messages_nodejs
[ 54%] Built target my_trailer_generate_messages_nodejs
[ 54%] Built target std_msgs_generate_messages_lisp
[ 63%] Built target my_trailer_generate_messages_lisp
[ 63%] Built target std_msgs_generate_messages_eus
[ 81%] Built target my_trailer_generate_messages_eus
[ 81%] Built target std_msgs_generate_messages_py
[100%] Built target my_trailer_generate_messages_py
[100%] Built target my_trailer_generate_messages

#roscore (one terminal)
#rosrun my_trailer item
A System Error window pops out
"The code execution cannot proceed because ".dll" file was not found. Reinstalling the program may fix the problem."

**Expected behavior**
I am expecting it to run without any problems.
It works on some other computers, but not all.

**Attempted Solution**
1) Ran choco upgrade ros-melodic-desktop_full -y --force to trigger rosdep again.
2) Installed all versions of Microsoft Visual C++ Restributable.
3) Reinstalled all dependencies.

**Environment information:**
 - OS version: Window 10
 - Chocolatey Versions: Chocolatey v0.10.15
 - ROS version: ROS Melodic (x64)

seanyen, 2020-01-24:
@qwe0254 Thanks for the reporting. Did it show what `DLL` causes the problem? And for your project, what does the dependency look like?

ooeygui, 2020-01-24:
Please check our the [Porting Cookbook](https://ms-iot.github.io/ROSOnWindows/Porting/Cookbook.html). Under the Symbols Visibility section, is a node about Gflags.exe - a Windows tool which helps identify export resolution issues.

If you can attach a screen shot of the dialog causing the problem, that would be helpful as well.

qwe0254, 2020-01-25:
Sorry for the late reply. It took me a while to figure out the debugging tools.

After getting the debugging tools, I ran catkin_make again. Surprisingly, I got an error this time (with the same code).

**CATKIN_MAKE**
Base path: C:\temp_ws
Source space: C:\temp_ws\src
Build space: C:\temp_ws\build
Devel space: C:\temp_ws\devel
Install space: C:\temp_ws\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\temp_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.23.28107.0
Copyright (C) Microsoft Corporation.  All rights reserved.

####
#### Running command: "nmake" in "C:\temp_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.23.28107.0
Copyright (C) Microsoft Corporation.  All rights reserved.

[  9%] Built target _setup_util.py_executable_install_python
[ 14%] Building CXX object my_trailer/CMakeFiles/item.dir/src/item.cpp.obj
item.cpp
C:\temp_ws\src\my_trailer\src\item.cpp(30): fatal error C1083: Cannot open include file: 'vector.h': No such file or directory
NMAKE : fatal error U1077: 'C:\PROGRA~2\MICROS~1\2019\COMMUN~1\VC\Tools\MSVC\1423~1.281\bin\Hostx64\x64\cl.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed

**Here is my CMake file:**
find_path(VECTOR_INCLUDE_DIR vector.h
          PATHS include/Vector/)
find_library(VECTOR_LIBRARY64 vector64.dll
             PATHS bin/)

if (VECTOR_INCLUDE_DIR AND VECTOR_LIBRARY64)
    set(VECTOR_FOUND TRUE)

    include_directories(${VECTOR_INCLUDE_DIR})
    set(YOUR_LIBRARIES ${YOUR_LIBRARIES} ${VECTOR_LIBRARY64})
endif()

add_executable(item src/item.cpp)
target_link_libraries(item ${catkin_LIBRARIES})
target_link_libraries(item ${YOUR_LIBRARIES})

**To specify the problem that I encountered:**
Apparently, the build path is messed up on Windows. How could I fix this?

qwe0254, 2020-01-26:
I suspect that the first error occurred because ROS couldn't traverse to a correct path. That was why it built, yet ran with a missing library error.

closed 2020-02-01 by qwe0254