ms-iot/ROSOnWindows #189: Security-related error at build

dobbory, 2020-02-10:
Can a security program cause a build to fail?

seanyen, 2020-02-10:
@dobbory Could you be more specific? What error did you see? Any steps to reproduce the error? And what security program do you run? Did turning it off help?

ooeygui, 2020-02-10:
In general, yes, a Security program can cause a build to fail. While we have not documented any cases with ROS on Windows after install, we have seen Chocolatey installs blocked which can cause a build failure.

- Can you post the output of `catkin_make`?
- Can you post the chocolaty log file: `c:\ProgramData\Chocolaty\logs\chocolaty.log`

seanyen, 2020-09-22:
@dobbory Let us know if it still an issue to you. Due to inactivity, I am closing it now.

closed 2020-09-22 by seanyen