ms-iot/ROSOnWindows #194: [Feature Request] Support Process Priority in launch files

ooeygui, 2020-02-19:
Some ROS nodes need to have higher priority execution - either generating data or maintaining a heartbeat on hardware. While it is possible to do this today from 'within' the node using OS features, it becomes problematic when tuning the whole system.

Wouldn't it be cool if we can declare a relative priority from within a launch file? This way the implementer of the solution can determine the overall priority of the solution.

(Adding to ROSOnWindows github in order to link to a doc where this is mentioned).

Upvote if you find this interesting.

gavanderhoorn, 2020-02-19:
Are you thinking of doing something with the [launch-prefix](https://wiki.ros.org/roslaunch/XML/node#Attributes) attribute?

ooeygui, 2020-02-19:
Oooh! - that's a good solution. It doesn't require changing infrastructure.

Sean mentioned that there's a process priority conversation in the ROS2 TSC.

gavanderhoorn, 2020-02-20:
Doing it in a platform agnostic way -- without needing to change anything else -- could probably be done by using an intermediate executable or script which checks which OS it is being executed on and then delegates to whatever tool could be used to run processes with required priorities (and other configuration).

If you don't care about being OS agnostic you could just use `start` with `/AboveNormal` or something.

IoTDan, 2022-10-11:
this is a ROS feature

closed 2022-10-11 by IoTDan