ms-iot/ROSOnWindows #196: catkin_make fails to find proper python env

mgdelmonte, 2020-02-24:
The setup script `c:\opt\ros\melodic\x64\setup.bat` properly sets the `PYTHONHOME` env var, but it does not set `PYTHON_EXECUTABLE`, which `catkin_make` references independently.  On a system with many python interpreters, this means `catkin_make` uses the wrong environment entirely (it uses the environment of the first-found interpreter, and not the one in PYTHONHOME).

I'd call this a bug in `catkin_make`, unless they have a good reason (unlikely) for using a random interpreter.  But nevertheless this appears to happen only on Windows.

To reproduce, install two python 2 environments and then ROS using the second environment.  Then install any ROS repo and run `catkin_make`.  Unless both python environments have `empy` It will fail with "python can't find library em..." (meaning `empy`).

Workaround:  manually add that PYTHON_EXECUTABLE env var to the ROS setup.  It must have the full path to the python executable.  (But you can't add it to `setup.bat` because it's autogenerated.)

OS: Windows 10
Chocolatey:  v0.10.15

ooeygui, 2020-02-24:
Thank you for the writeup and investigation. We've been trying to map the ROS environment to a locally installed python path in c:\opt, so that it is isolated from potentially tainted environments outside of the ROS environment.

IoTDan, 2022-10-11:
we will consider checking PYTHONHOME in the setup and offer a warning during setup that a previous python environment was detected. the user can still click to continue setup.