ms-iot/ROSOnWindows #197: [ROS2][Dashing] local_setup.bat can't find other batch files

ooeygui, 2020-02-24:
When running local_setup.bat, the following 4 batch files are listed as missing.
```
not found: "c:\opt\ros\dashing\x64\share/rttest/local_setup.bat"
not found: "c:\opt\ros\dashing\x64\share/tlsf/local_setup.bat"
not found: "c:\opt\ros\dashing\x64\share/tlsf_cpp/local_setup.bat"
not found: "c:\opt\ros\dashing\x64\share/pendulum_control/local_setup.bat"
```

seanyen, 2020-02-24:
They seem to share something in common. Their projects are all skipped on Windows build.

https://github.com/ros2/realtime_support/blob/583ffcdf68af70a7e5f4f1abb0e6ed7a7c123247/rttest/CMakeLists.txt#L6

https://github.com/ros2/demos/blob/fab4fc38d5ea589c019659d6a49da4ba9c20594b/pendulum_control/CMakeLists.txt#L6

ooeygui, 2020-02-26:
Seems like it should be nuked from the bat file since they aren't on the win build.

seanyen, 2020-05-08:
This is addressed by this PR: https://github.com/ms-iot/ros-windows-build/pull/89

Closing it now.

closed 2020-05-08 by seanyen