ms-iot/ROSOnWindows #199: Missing environment variables for Windows on ROS 2

seanyen, 2020-03-06:
After `choco install ros-elquent-desktop -y` and running `local_setup.bat`, all the environment should be all set and ready to use ROS 2. However, currently there are some environment variables missing.

## Workaround:
Before we figure out the fix, one can run the following before `local_setup.bat` is invoked:
```Batchfile
set "PATH=c:\opt\python37amd64;c:\opt\python37amd64\scripts;c:\opt\rosdeps\x64\bin;c:\opt\rosdeps\x64\lib;%PATH%"
set "PYTHONHOME=c:\opt\python37amd64"
set "PYTHONPATH=c:\opt\python37amd64\lib\site-packages;c:\opt\ros\eloquent\x64\lib\site-packages"
call "c:\opt\ros\eloquent\x64\local_setup.bat"
```

ooeygui, 2020-03-06:
Verified workaround

seanyen, 2020-09-22:
Closing it for now. This is no longer required.

closed 2020-09-22 by seanyen