ms-iot/ROSOnWindows #202: sensor_msgs.msg

void32, 2020-03-30:
**Describe the bug**
Can't import from sensor_msgs.msg 

**To Reproduce**
cd c:\opt\ros\melodic\x64\
c:\opt\ros\melodic\x64>
python
>>> from sensor_msgs.msg import Image
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named msg

**Expected behavior**
Import Image from sensor_msgs.msg with no errors.

**Environment information:**
 - OS version: Windos 10 (OS Build 19541. 1000)
 - Chocolatey Versions: Chocolatey v0.10.15
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
git 2.26.0
git.install 2.26.0
4 packages installed.

ooeygui, 2020-03-30:
@void32 From what's presented, It appears that you aren't sourcing the ros environment. Your chocolatey versions are missing - all of the ROS chocolatey packages.

Can you do a `set ChocolateyInstall`?
You should see:
```
ChocolateyInstall=c:\opt\chocolatey
```

And if you do `set ROS`, you should see:
```
ROS_DISTRO=melodic
ROS_ETC_DIR=C:/opt/ros/melodic/x64/etc/ros
ROS_IP=127.0.0.1
ROS_MASTER_URI=http://127.0.0.1:11311
ROS_PACKAGE_PATH=C:\opt\ros\melodic\x64\share
ROS_PYTHON_VERSION=2
ROS_ROOT=C:/opt/ros/melodic/x64/share/ros
ROS_VERSION=1
```

And of you do `where python` you should see:
```
C:\opt\python27amd64\python.exe
```

Let me know what doesn't match your environment.
Thank you.

gavanderhoorn, 2020-03-31:
I believe a similar case on ROS Answers: [Error importing sensor_msgs/JointState](https://answers.ros.org/question/347925).

aSorathia, 2020-03-31:
@ooeygui  I have the similar issue for me all the environment variable matches as yours

void32, 2020-03-31:
> @void32 From what's presented, It appears that you aren't sourcing the ros environment. Your chocolatey versions are missing - all of the ROS chocolatey packages.
>
> Can you do a `set [...]

Hi ooeygui,
Thanks for the quick reply, and bringing ROS to Windows :-) 
Just to be clear, by sourcing you mean run the setup.bat in C:\opt\ros\melodic\x64?

My out seams to fit yours
```
C:\opt\ros\melodic\x64>set ChocolateyInstall
ChocolateyInstall=C:\ProgramData\chocolatey

C:\opt\ros\melodic\x64>set ROS
ROS_DISTRO=melodic
ROS_ETC_DIR=C:/opt/ros/melodic/x64/etc/ros
ROS_IP=127.0.0.1
ROS_MASTER_URI=http://127.0.0.1:11311
ROS_PACKAGE_PATH=C:\opt\ros\melodic\x64\share
ROS_PYTHON_VERSION=2
ROS_ROOT=C:/opt/ros/melodic/x64/share/ros
ROS_VERSION=1

C:\opt\ros\melodic\x64>where python
C:\opt\python27amd64\python.exe
C:\Users\Morten\AppData\Local\Microsoft\WindowsApps\python.exe
C:\Python27\python.exe
```
Btw I've found a workaround by replacing sensor_msgs-1.12.7-py2.7.egg with a zip of the sensor_msgs folder. This works for me because all I'm currently using from ROS, is extracting data from rosbags in a python script.
However, I noticed that the current content of sensor_msgs-1.12.7-py2.7.egg does not match the content of sensor_msgs. How are the .egg files build?

asasine, 2020-04-02:
Hi @void32, I tried your mitigation and it didn't work for me. Are these the correct steps?
1. Delete the _C:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg_ file (or move it to another directory as backup)
1. Zip _C:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs\_ as _sensor_msgs.zip_
1. Place _sensor_msgs.zip_ at path _C:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs.zip_

ooeygui, 2020-04-02:
We are preparing an updated release.

aSorathia, 2020-04-02:
@asasine To make successful egg file Create a new folder somewhere in the system
Copy sensor_msgs folder into the new folder and create a blank file inside sensor_msgs folder and rename it as `__init__.py`
inside the new folder create `setup.py` file and add this script

```
from setuptools import setup, find_packages

setup(
    name = "sensor_msgs",
    version = "1.12.7",
    packages = find_packages()
    )
```
`find_packages()` does not work if any folder does not have `__init__.py` file	
save it.
Your dir structure should look like this

```
-New_folder
--setup.py
--sensor_msgs
---__init__.py
---msg
---srv
```

from inside the new_folder run setup.py
it will create 3 folders 
your egg file will be inside dist folder.

seanyen, 2020-04-02:
Hi folks, I pushed a version `20200402.1.0` and the latest should have this `sensor_msgs` issue (and all Python egg problems) fixed.

asasine, 2020-04-03:
Version `20200402.1.0` fixes the issue for me! Thank you!

seanyen, 2020-04-07:
This is fixed by the latest release, so I am closing it.

closed 2020-04-07 by seanyen

void32, 2020-04-10:
I've [tested it](https://wiki.ros.org/Installation/Windows) and can confirm that it works.