ms-iot/ROSOnWindows #203: vcpkg: Vcpkg is not installed

seanyen, 2020-04-08:
This is a cross-post for the issues reported here:
*  https://answers.ros.org/question/348689/error-installing-ros-on-windows10/
*  https://answers.ros.org/question/348731/roswin10-vcpkg-vcpkg-is-not-installed/

DanielSepeda, 2020-04-09:
@seanyen @ooeygui 
Attached is my chocolatey log. I created the post on https://answers.ros.org/question/348689/error-installing-ros-on-windows10/

[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/4453327/chocolatey.log)

boomboompsh, 2020-04-09:
Attached is my chocolatey log, I had the same problem.
[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/4454017/chocolatey.log)

seanyen, 2020-04-09:
Hi all, a version of ` 20200409.2.0` is released and this should be addressed. Let me know if you still run into this issue. Thanks!

dweekly, 2020-04-11:
> Hi all, a version of ` 20200409.2.0` is released and this should be addressed. Let me know if you still run into this issue. Thanks!

Hi - new ROS user installing for the first time ever on Windows 10; I'm running into this issue even after specifically running a "choco upgrade ros-vcpkg" first.

Still seeing "ERROR: the following rosdeps failed to install" vcpkg: Vcpkg is not installed". This is with ros-melodic-desktop_full V20200409.2. This is on Windows 10 1909 (OS Build 18363.752) and Visual Studio Community Edition 2019; fresh install.

seanyen, 2020-04-11:
@dweekly Thanks for reporting, can you check if `c:\opt\vcpkg\vcpkg.exe` is created after the broken install?

dweekly, 2020-04-12:
Ah, it was not. My issue was caused by https://github.com/ms-iot/ROSOnWindows/issues/153

seanyen, 2020-04-24:
Closing it since the fixed release is out.

andyworkingholiday, 2020-04-25:
> Closing it since the fixed release is out.

Can you know me where is fixed release??
I'm having exactly the same issue.

ooeygui, 2020-04-25:
Hi @andyworkingholiday ,
Sorry the resolution was not clear. To fix your install, please rerun the upgrade command:
`choco upgrade ros-melodic-desktop_full -y --execution-timeout=0`

lwoffice, 2020-04-27:
Hi, I also have the same trouble.

I have installed VC2019 community (because I do not want to mess with my 2017 work license) just today and added the desktop C++ package, so that should be there.

I followed this "http://wiki.ros.org/Installation/Windows" until 5.1 and ran into the problem.
I found the "solution" from the first post, which did not change anything, I still got the same error messages.

Then, I upgraded "ros-vcpkg":

C:\Windows\System32>choco upgrade ros-vcpkg -y --execution-timeout=0
Chocolatey v0.10.15
Upgrading the following packages:
ros-vcpkg
By upgrading you accept licenses for the packages.
ros-vcpkg v2018.11.23.1906112226 is the latest version available based on your source(s).
Chocolatey upgraded 0/1 packages.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

I guess there is nothing to update. I also have a "C:\opt\vcpkg" folder with lots of stuff, so there is something.

Then I again called "choco upgrade ros-melodic-desktop_full -y --execution-timeout=0"

and now my error messages for the ros installation changed to this:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_core: No definition of [rosbag_migration_rule] for OS [windows]
rospack: No definition of [python-catkin-pkg-modules] for OS [windows]
diagnostic_common_diagnostics: No definition of [hddtemp] for OS [windows]
trajectory_msgs: No definition of [rosbag_migration_rule] for OS [windows]
desktop: No definition of [common_tutorials] for OS [windows]
simulators: No definition of [stage_ros] for OS [windows]
joint_trajectory_controller: No definition of [code_coverage] for OS [windows]
joystick_drivers: No definition of [wiimote] for OS [windows]
Continuing to install resolvable dependencies...
ERROR: the following rosdeps failed to install
  vcpkg: Vcpkg is not installed

I don't understand that. I upgraded vcpkg which says I have the latest version and then the next call now says I do not have any version ?

What should I do? Delete some folders and start over?

regards

ooeygui, 2020-04-27:
@lwoffice, I have a hypothesis.

To confirm, is there `c:\opt\vcpkg\vcpkg.exe`?

If not, can you:
```
cd c:\opt\vcpkg
bootstrap-vcpkg.bat
```

and post the output?

lwoffice, 2020-04-27:
Well, I am currently figuring it out and it has to do with messed up Path variables. This is what I have right now:

I partly solved the issue by starting from scratch.  I found that my Windows settings missed an entry in the PATH variable for Powershell (%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\)

As a result, the installation of the vcpkg outputs a message which can be quickly overlooked and says that the command "Powershell" can not be found. Nevertheless there will be a green message that the installation of vcpkg was successful !

I looked up thread #153 where I found that their is a build script after the download procedure which outputs message "Building vcpkg.exe ...". Without powershell, that build script does not work obviously.

I looked up the windows features, but Powershell was installed. Whatever, I removed and reinstalled it, but it still did not work. I manually added the path from above and I was able now to call Powershell from the x64 Dev Console.

Additionally, I also had to completely delete the c:\opt folder, otherwise the build process was also not started correctly. After that, it worked fine and a "vcpkg.exe" was created in C:\opt\vcpkg.

Nevertheless, the installation of ros-melodic afterwards still fails: 

C:\Windows\System32> choco upgrade ros-melodic-desktop_full -y --execution-timeout=0
Chocolatey v0.10.15
[Pending] Removing incomplete install for 'ros-melodic-desktop_full'
Upgrading the following packages:
ros-melodic-desktop_full
By upgrading you accept licenses for the packages.
ros-melodic-desktop_full is not installed. Installing...
Progress: Downloading ros-melodic-desktop_full 20200409.2... 100%
Progress: Downloading ros-melodic-desktop_full 20200409.2... 100%

ros-melodic-desktop_full v20200409.2
ros-melodic-desktop_full package files upgrade completed. Performing other installation steps.
Extracting 64-bit C:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\drop.zip to c:\opt\ros...
c:\opt\ros
running rosdep...
Der Befehl "rosdep" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
Der Befehl "rosdep" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
Der Befehl "rosdep" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
ERROR: rosdepInstall.bat failed.
The upgrade of ros-melodic-desktop_full was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.

Chocolatey upgraded 0/1 packages. 1 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Failures
 - ros-melodic-desktop_full (exited -1) - Error while running 'C:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.

I looked up "chocolateyInstall.ps1" and "rosdepInstall.bat" and I guess it now has something to do with a messed Python variable too. Trying to figure that out now ...

ooeygui, 2020-04-27:
Thank you for investigating.

The vcpkg bootstrap script does not use a full path to powershell:
`powershell.exe -NoProfile -ExecutionPolicy Bypass "& {& '%~dp0scripts\bootstrap.ps1' %*}"`

(the ROS install is explicit about calling powershell by full path to install chocolatey)

Can you run `c:\opt\chocolatey\lib\ros-melodic-desktop_full\tools\rosdepInstall.bat` and paste the output?

lwoffice, 2020-04-28:
That was not successful, but I recognized some (probably) strange folder issues. 

In the c:\opt\chocolatey\lib" I only have the "ros-vcpkg" folder. There are no further subfolders. Whenever I use the install procedure for ros (choco upgrade ros-melodic-desktop_full ... ), this "ros_melodic..." folder is created in "C:\ProgramData\chocolatey\lib" and after it fails it is automatically moved into the "lib-bad" folder.

So ... I can find that rosdepInstall.bat here "C:\ProgramData\chocolatey\lib-bad\ros-melodic-desktop_full\tools". Running this manually immediately crashes the whole x64 console, so I can not see anything.

I am really wondering, why I have this chocolatey folder in "C:\ProgrammData", although I installed chocolatey according to "http://wiki.ros.org/Installation/Windows" section 4. Looking at that installation string it might have something to do with the "PATH" variable as well?

I made a screenshot with all the folders and you I found a "c:\ProgramData\chocolatey\bin" in the System Path variable. I do not know if that is supposed to be there and why it is there.
[image]

ooeygui, 2020-04-28:
Thank you for working with us on this, I really do appreciate it.

The console window crashing is quite ominous. What version of Windows are you using? (I can't tell from the screen shot).

I noticed that you have mingw32 in your path, I'm a bit worried that the inherited environment is causing compatibility problems. ROS is quite sensitive to the host environment, which is why we've tried to isolate it as much as possible. However, environment from the host can 'leak' into the ROS environment in strange ways.

When you create the shortcut to the console window, we recommended adding a chocolatey bypass, so that it too is isolated:
```C:\Windows\System32\cmd.exe /k "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\VsDevCmd.bat" -arch=amd64 -host_arch=amd64&& set ChocolateyInstall=c:\opt\chocolatey&& c:\opt\ros\melodic\x64\setup.bat
````
When the console is launched with that command line, chocolatey will reference the packages from the `c:\opt\chocolatey` directory.

lwoffice, 2020-04-29:
Version is Win 10 Pro 1809 (Build 17763.1098). 
shortcut: well ... I was stuck at section 5.1 in the installation manual, and that hint to the cmd is in section 6, I just did not continue reading the manual :D 

After all, I figured it out. I completely removed the c:\opt and the c:\ProgramData\chocolately folders to get rid of the crappy installations before. I restarted all steps from section 4, but I used a cmd with the "bypass" option.
Might be a good idea to adapt the manual http://wiki.ros.org/Installation/Windows :

1)
Move section 6 inbetween 3 and 4 and modify it slightly: I removed the last statement (&& c:\opt\ros\melodic\x64\setup.bat), because I did not want the setup.bat to be executed in the steps 4 and 5 already, I just wanted the console with the bypass which worked fine (maybe it would not harm to have setup.bat here already, I do no know). I used that modified cmd for steps in section 4 and 5 and when I reached section 6, I added the "...\setup.bat" part to the link.

Honestly, I do not know why there is a difference between setting "set ChocolateyInstall=..." in the link instead of calling it in the console itself, but on my system that made the difference. Maybe that is a Windows secret :)

2)
You could add the workaround for vcpkg to section 5.1 (and probably 5.2), so people do no longer need to search this thread. At least it does not harm I think, although it might not be necessary in all cases:
  choco upgrade ros-vcpkg -y --execution-timeout=0

It might also be good to add the hint with regard to Powershell path issues that I had. Users should check their User Path variables and add "%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\" if it does not exist. That solved my build issues with regard to vcpkg.

That's it for me ...

Chocolatey upgraded 10/10 packages.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

Done  :D

ooeygui, 2020-04-29:
> After all, I figured it out. I completely removed the c:\opt and the c:\ProgramData\chocolately folders to get rid of the crappy installations before. I restarted all steps from section 4, but I used a cmd with the "bypass" option.
I'm glad you figured it out.

> Move section 6 inbetween 3 and 4 and modify it slightly: I removed the last statement (&& c:\opt\ros\melodic\x64\setup.bat), because I did not want the setup.bat to be executed in the steps 4 and 5 [...]
We had put it in this order because setup.bat won't exist the first time around which causes an error launching the script. 

> Honestly, I do not know why there is a difference between setting "set ChocolateyInstall=..." in the link instead of calling it in the console itself, but on my system that made the difference. Maybe that is a Windows secret :)
I'll add some clarity to the doc. We decided to move the chocolatey install into the c:\opt directory so that it was self-contained. you can reinstall ROS by removing c:\opt or move the install to another machine by copying c:\opt. That said, we are looking at alternative environment management solutions for future deployments. (Including CondaForge)

> You could add the workaround for vcpkg to section 5.1 (and probably 5.2), so people do no longer need to search this thread. At least it does not harm I think, although it might not be necessary in all cases:
> choco upgrade ros-vcpkg -y --execution-timeout=0

> 
> It might also be good to add the hint with regard to Powershell path issues that I had. Users should check their User Path variables and add "%SYSTEMROOT%\System32\WindowsPowerShell\v1.0" if it does not exist. That solved my build issues with regard to vcpkg.
I added a troubleshooting section.

> That's it for me ...
I really appreciate you working with us on this!

David-milne, 2020-05-08:
Hi, I'm having issues. ```ros-melodic-desktop_full``` fails to install and ```c:\opt\vcpkg\vcpkg.exe``` doesn't exist.
I tried multiple time to install ROS,
I had ensured no versions of python were on the system,
Powershell is installed and is in the Path variables,
I've used the ```choco upgrade ros-vcpkg -y --execution-timeout=0``` workaround and the 
```
cd c:\opt\vcpkg
bootstrap-vcpkg.bat
```
which created ```vcpkg.exe``` but after ```choco upgrade ros-melodic-desktop_full -y --execution-timeout=0``` it always fails.
here is my log from the last attempt:
[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/4597610/chocolatey.log)

Also, there is no path to the python 2.7, that the install downloads, in the path variable. I don't know if there should be.

seanyen, 2020-05-08:
@David-milne Is it possible to provide a screenshot of what error you saw? And the `%PATH%` environment variable from the console you run the `choco` commands?

From the logs it said it failed at `[ERROR] - ERROR: rosdepInstall.bat failed.` but unfortunately I didn't see more details other than that, so I am requesting more information.

David-milne, 2020-05-09:
@seanyen After I posted last I rebuilt ```vcpkg.exe``` and used ```c:\opt\chocolatey\lib-bad\ros-melodic-desktop_full\tools\rosdepInstall.bat``` which completed without errors. But then I ran ```choco upgrade ros-melodic-desktop_full -y --execution-timeout=0``` and got the same error as before.

Here is my path environment variables:
[image]

I built the ```vcpkg.exe``` again and ran ```choco upgrade ros-melodic-desktop_full -y --execution-timeout=0```

Here is the failed install output:
[image]

KamaljeetSahoo, 2020-05-10:
> @seanyen After I posted last I rebuilt `vcpkg.exe` and used `c:\opt\chocolatey\lib-bad\ros-melodic-desktop_full\tools\rosdepInstall.bat` which completed without errors. But then I ran `choco upgrade [...]

Yes that's it 
I am getting the exact same error.
Please someone help us.

lwoffice, 2020-05-11:
Hi,

I noticed that for both of you console outputs it still says that vcpkg.exe is not installed. I guess you have called installation part, but it still did not successfully build that executable. Please have a look into C:\opt\vcpkg and if there is really an executable.

regards

David-milne, 2020-05-11:
Thanks for the suggestion, but I have started with a clean ```C:\opt```, built ```vcpkg.exe``` successfully before starting ```choco upgrade ros-melodic-desktop_full -y --execution-timeout=0```, the cmd prompt is set to run as Administrator, and I have watched ```vcpkg.exe``` in file manager get removed during the install and then fail because ```vcpkg.exe``` does not exist.
Also this is a new computer. ROS 2 installed fine, but ROS 1 won't and I need ROS 1 to control Baxter robot.
p.s. I had already removed ROS 2 before my latest fresh install.

ooeygui, 2020-05-11:
@David-milne Thank you for working with us on this issue. We're actively working to understand the environment where this fails. 
Are you using the ROS2 install instructions from http://wiki.ros.org/Installation/Windows or from https://index.ros.org/doc/ros2/Installation/Eloquent/?
In the command window where the error occurs, can you paste into this issue the output of 

```
set
```
Please remove any personally identifying information, or other secret data. We're specifically interested in the Visual Studio, Path, Python and ROS related environment variables.

David-milne, 2020-05-11:
@ooeygui Sorry if I was confusing before, I installed ROS2 from [http://wiki.ros.org/Installation/Windows]( http://wiki.ros.org/Installation/Windows) and it worked fine. My issue is installing ROS1.

Hereis the output from ```set```
```
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\david\AppData\Roaming
ChocolateyInstall=c:\opt\chocolatey
ChocolateyLastPathUpdate=132333921702576266
CommandPromptType=Native
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=LAPTOP-RO5HSBBB
ComSpec=C:\windows\system32\cmd.exe
DevEnvDir=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\
DriverData=C:\Windows\System32\Drivers\DriverData
ExtensionSdkDir=C:\Program Files (x86)\Microsoft SDKs\Windows Kits\10\ExtensionSDKs
Framework40Version=v4.0
FrameworkDir=C:\Windows\Microsoft.NET\Framework64\
FrameworkDir64=C:\Windows\Microsoft.NET\Framework64\
FrameworkVersion=v4.0.30319
FrameworkVersion64=v4.0.30319
HOMEDRIVE=C:
HOMEPATH=\Users\david
INCLUDE=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\ATLMFC\include;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\include;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt;C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt
JAVA_HOME=C:\Program Files\Java\jdk-14.0.1
KINECTSDK10_DIR=C:\Program Files\Microsoft SDKs\Kinect\v1.7\
LIB=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\ATLMFC\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\lib\x64;C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\ucrt\x64;C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\um\x64;
LIBPATH=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\ATLMFC\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\lib\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\lib\x86\store\references;C:\Program Files (x86)\Windows Kits\10\UnionMetadata\10.0.18362.0;C:\Program Files (x86)\Windows Kits\10\References\10.0.18362.0;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;
LOCALAPPDATA=C:\Users\david\AppData\Local
LOGONSERVER=\\LAPTOP-RO5HSBBB
NITE2_INCLUDE64=C:\Program Files\PrimeSense\NiTE2\Include\
NITE2_LIB64=C:\Program Files\PrimeSense\NiTE2\Lib\
NITE2_REDIST64=C:\Program Files\PrimeSense\NiTE2\Redist\
NUMBER_OF_PROCESSORS=12
OneDrive=C:\Users\david\OneDrive
OneDriveConsumer=C:\Users\david\OneDrive
OnlineServices=Online Services
OPENNI2_INCLUDE64=C:\Program Files\OpenNI2\Include\
OPENNI2_LIB64=C:\Program Files\OpenNI2\Lib\
OPENNI2_REDIST64=C:\Program Files\OpenNI2\Redist\
OPENSSL_CONF=C:\Program Files\OpenSSL-Win64\bin\openssl.cfg
OS=Windows_NT
Path=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\bin\HostX64\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\x64;C:\Program Files (x86)\Windows Kits\10\bin\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\\MSBuild\Current\Bin;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS\;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\MATLAB\R2020a\bin;C:\Program Files\dotnet\;C:\ProgramData\chocolatey\bin;c:\opt\chocolatey\bin;C:\Program Files\Git\cmd;C:\Program Files\Cppcheck;C:\Users\david\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Mercurial\;C:\Program Files\OpenSSL-Win64\bin;C:\libjpeg-turbo64\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja;C [... cut ...] am Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\MATLAB\R2020a\bin;C:\Program Files\dotnet\;C:\ProgramData\chocolatey\bin;c:\opt\chocolatey\bin;C:\Program Files\Git\cmd;C:\Program Files\Cppcheck;C:\Users\david\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Mercurial\;C:\Program Files\OpenSSL-Win64\bin;C:\libjpeg-turbo64\bin
```

seanyen, 2020-05-12:
@David-milne Thanks for your time working with us on this issue. At this moment I am still trying to figure out what causes the Vcpkg missing problem. But, I did some updates for Vcpkg installation checking and error handling. And hope that can catch any error earlier and give more useful error messages. Currently it is in pre-release channel, can you try `choco upgrade ros-melodic-desktop_full -y --execution-timeout=0 --pre` and let us know how it goes?

David-milne, 2020-05-12:
@seanyen I tried a fresh install using ```choco upgrade ros-melodic-desktop_full -y --execution-timeout=0 --pre```, it failed immediately after building ```vcpkg.exe``` instead of after downloading all the other packages (like gazebo, ect)
output:
[image]
Log:
[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/4619013/chocolatey.log)

seanyen, 2020-05-13:
@David-milne Thanks. I intentionally failed the rest of the installation and made it fail fast. And it seems to me that it did build `vcpkg.exe`, however, it didn't pass the Vcpkg existence check. Can you try do `dir /s /b c:\opt\vcpkg\*.exe` and check the file existence of `C:\opt\vcpkg\.vcpkg-root`?

I am trying to figure out why the Vcpkg.exe is not [copied](https://github.com/microsoft/vcpkg/blob/master/scripts/bootstrap.ps1#L426-L430) to `$vcpkgRootDir` defined [here](https://github.com/microsoft/vcpkg/blob/master/scripts/bootstrap.ps1#L52-L58).

David-milne, 2020-05-13:
@seanyen here is the output for ```dir /s /b c:\opt\vcpkg\*.exe```:
[image]

and ```C:\opt\vcpkg\.vcpkg-root``` does exist:
[image]

I built ```vcpkg.exe``` using
```
cd c:\opt\vcpkg
bootstrap-vcpkg.bat
```
Checking the time marks there was 11 minuets between outputting the "Telemetry..." and "The upgrade of ros-vcpkg was successful" (I don't know why that takes so long)
I then used ```dir /s /b c:\opt\vcpkg\*.exe``` again:
[image]
[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/4620099/chocolatey.log)

seanyen, 2020-05-13:
@David-milne Thanks for new information. This is indeed strange to me in some cases that `Vcpkg.exe` is not generated. Just wondering do you have any anti-virus software installed? Can you temporarily turn off them and retry again? I am checking because I read Vcpkg threads talking about that could be one reason to render this `Vcpkg.exe` missing issue.

https://github.com/microsoft/vcpkg/issues/9704
https://github.com/Microsoft/vcpkg/issues/5187

jiaodaxiaoli, 2020-05-13:
ros-colcon-tools and ros-eloquent-desktop 
ERROR: rosdepInstall.bat failed.
The upgrade of ros-eloquent-desktop was NOT successful.
[image]
[image]

jiaodaxiaoli, 2020-05-13:
i  

> ros-colcon-tools和ros-eloquent-desktop
> 错误：rosdepInstall.bat失败。
> ros-eloquent-desktop的升级失败。
> [image]
> [image]
I hope that someone can help me ,thank you!

David-milne, 2020-05-13:
@seanyen Well now I feel extremely dumb for not realising that earlier.

The laptop came with McAfee preinstalled, btw. I turned off both McAfee and Window's anti-virus real-time scanning and tried a fresh install and it completed. Yay!
I guess ```vcpkg.exe``` was being flagged by McAfee and automatically deleted, idk
Thanks for your patience and hard work :)

seanyen, 2020-05-13:
@jiaodaxiaoli Can you share your `chocolatey.log` and the environment variables (for example, type `set > env.log` in the console and share the `env.log`)? And please remove any personal information before sharing. Thanks!

seanyen, 2020-05-13:
@David-milne Glad to hear now the problem is solved!

ooeygui, 2020-05-13:
I updated the wiki.ros.org install instructions to cover this case. Thank you for working through this with us!

seanyen, 2020-05-13:
@jiaodaxiaoli Sorry I didn't notice an error `ERROR: The remote name could not be resolved` in your screenshot. During the ROS installation, it requires to pull some files from Internet and some are from GitHub. It seems that in your environment, it has a problem to access this location: `https://raw.githubusercontent.com/ms-iot/rosdistro-db/init_windows/rosdistro_cache/colcon-requirements.txt`

Can you double check if your network configuration?

jiaodaxiaoli, 2020-05-15:
[image]
[image]

Chocolatey upgraded 10/12 packages. 2 packages failed.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

Upgraded:
 - kb2919355 v1.0.20160915
 - kb3033929 v1.0.5
 - chocolatey-core.extension v1.3.5.1
 - kb2999226 v1.0.20181019
 - ros-python37 v3.7.2.1903082244
 - vcredist2010 v10.0.40219.2
 - kb2919442 v1.0.20160915
 - vcredist140 v14.25.28508.3
 - kb3035131 v1.0.3
 - chocolatey-windowsupdate.extension v1.0.4

Failures
 - ros-colcon-tools (exited -1) - Error while running 'c:\opt\chocolatey\lib\ros-colcon-tools\tools\chocolateyinstall.ps1'.
 See log for details.
 - ros-eloquent-desktop (exited -1) - Error while running 'c:\opt\chocolatey\lib\ros-eloquent-desktop\tools\chocolateyInstall.ps1'.
 See log for details.

Attached is my chocolatey log and env.log.Thank you fou helping me with the problem.
[env.log](https://github.com/ms-iot/ROSOnWindows/files/4632074/env.log)

[chocolatey.log](https://github.com/ms-iot/ROSOnWindows/files/4632048/chocolatey.log)

jiaodaxiaoli, 2020-05-15:
sorry，I don't quite understand your requirements.What network configuration is required？ @seanyen

jiaodaxiaoli, 2020-05-15:
ERROR: The remote name could not be resolved: 'raw.githubusercontent.com'
The upgrade of ros-colcon-tools was NOT successful.
Error while running 'c:\opt\chocolatey\lib\ros-colcon-tools\tools\chocolateyinstall.ps1'.
 See log for details. @seanyen

gavanderhoorn, 2020-05-15:
@jiaodaxiaoli: are you behind a (corporate) firewall?

What is the output of `ping raw.githubusercontent.com`?

If you are behind a firewall, additional configuration may be needed to allow `vcpkg` and/or Chocolately access to the internet. I would not know what configuration that would be though.

jiaodaxiaoli, 2020-05-15:
[image]
[image]I shut down the firewall.

jiaodaxiaoli, 2020-05-15:
[image]
Ping request could not find host raw.githubusercontent.com. @gavanderhoorn

jiaodaxiaoli, 2020-05-18:
Sorry,I don't understand
I've downloaded it, but running this statement (ros2 run demo_nodes_cpp talker) shows the above error.
c:\opt\ros\melodic\x64\setup.bat does not exist. 
C:\opt\ros\eloquent\x64\setup.bat   exists.
[image]

[image]
[image]
[image]
[Uploading    @seanyen chocolatey.log…]()

ooeygui, 2020-05-18:
@jiaodaxiaoli You are experiencing multiple separate issues:

- It looks like you have installed the eloquent chocolatey packages but not the melodic ones. Which instructions did you follow to install?

-  `raw.githubusercontent.com` is the static resource server used by `github.com`. If you cannot access github, then something is very wrong on your system. As @gavanderhoorn suggested, this could be a proxy issue. Please check out https://chocolatey.org/docs/proxy-settings-for-chocolatey to see if that resolves your problem. If this doesn't help, you may want to consult your IT department.

- You need to be online to set up ROS the first time. If you need to work offline, connect your system to set up, then disconnect in order to run offline.

jiaodaxiaoli, 2020-05-18:
@ooeygui  hi, I installed it according to the instructions on this website--http://roswiki.autolabor.com.cn/Installation(2f)Windows.html
but I didn't install step 5.1  .

[image]

jiaodaxiaoli, 2020-05-18:
I don't know how to set up ros, because I am a beginner. I hope you can explain it carefully.

ooeygui, 2020-05-18:
@jiaodaxiaoli Thank you for the pointer. That appears to be a copy of the official documentation, and is missing steps. The official documentation is located: https://wiki.ros.org/Installation/Windows.

Please follow the steps from top to bottom. The distribution of ROS2 in that document is designed to work within ROS1.

If you are interested in ROS2 only, the official documentation is here: https://index.ros.org/doc/ros2/.

At this point, I recommend removing `c:\opt` and starting over.

jiaodaxiaoli, 2020-05-19:
@ooeygui Thank  you very much for your help, I have installed ROS.I have installed ROS, but now I have a new problem. My RVIZ starts normally, but gazebo doesn't. It shows that 'gazebo' is not an internal or external command, nor is it a runnable program or batch file.Here are my system variables.
[image]
[image]
[image]

ooeygui, 2020-06-02:
@jiaodaxiaoli It is best to file a new issue as this is not vcpkg related. Gazebo is provided in the gazebo_ros package, and is from roslaunch.

940222JESUS, 2020-06-02:
Hi
I have a similar problem, i follow the instructions of the wiki http://wiki.ros.org/Installation/Windows but when i put de command "choco upgrade ros-melodic-desktop_full -y --execution-timeout=0" the installation is interrupt with the next error:

[image]

`You are using pip version 18.1, however version 20.2b1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
ERROR: the following rosdeps failed to install
  vcpkg: command [vcpkg install sdl2 --triplet x64-windows] failed
  vcpkg: Failed to detect successful installation of [sdl2]
ERROR: rosdepInstall.bat failed.
  ros-melodic-desktop_full can be automatically uninstalled.
Environment Vars (like PATH) have changed. Close/reopen your shell to
 see the changes (or in powershell/cmd.exe just type `refreshenv`).
The upgrade of ros-melodic-desktop_full was NOT successful.
Error while running 'c:\opt\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.

Chocolatey upgraded 11/12 packages. 1 packages failed.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

Upgraded:
 - kb2919355 v1.0.20160915
 - kb3033929 v1.0.5
 - kb2999226 v1.0.20181019
 - ros-catkin-tools v0.0.1.1903221831
 - ros-python2 v2.7.15.1904300710
 - vcredist2010 v10.0.40219.2
 - kb2919442 v1.0.20160915
 - ros-vcpkg v2018.11.23.1906112226
 - vcredist140 v14.26.28720.3
 - kb3035131 v1.0.3
 - chocolatey-windowsupdate.extension v1.0.4

Failures
 - ros-melodic-desktop_full (exited -1) - Error while running 'c:\opt\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.`

I have try to install in two computers but the error is the same, i have trying with the solve 20200409.2.0 but it not change nothing, the file vcpkg.exe is created in the folder "C:\opt\vcpkg" , I have "%SYSTEMROOT%\System32\WindowsPowerShell\v1.0" in path variables, i follow the recomendation of lwoffice commented on 29 Apr, but the result is the same. What i can do it?

ooeygui, 2020-06-02:
vcpkg is the [canary in the coalmine](https://en.wiktionary.org/wiki/canary_in_a_coal_mine). It is the first time that ROS attempts to build in a prompt. If it fails, it means that the environment has not been set up correctly.

Can you do the following:
Open the ROS command prompt.
```
vcpkg install sdl2 --triplet x64-windows
```

Paste the output?

940222JESUS, 2020-06-02:
[image]

The result is " "vcpkg" is not recognized as an internal or external command, program, or executable batch file."

ooeygui, 2020-06-02:
@940222JESUS in your shortcut, did you include the setup.bat for the Window? If not, can you run `c;\opt\ros\melodic\x64\setup.bat` and retry?

940222JESUS, 2020-06-02:
This is the output

[image]

ooeygui, 2020-06-02:
Thank you for digging in with us. The English language pack is requirement within vcpkg itself: https://github.com/Microsoft/vcpkg/issues/3842. 

https://github.com/microsoft/vcpkg/issues/10401.

I'll update the document on the ros install page.

940222JESUS, 2020-06-02:
[image]

I installed the English language pack and this is the result, Right now I executed roscore and i think that it's working fine, there are another step that i need to follow or it's over?

If not thank you so much for help me.

ooeygui, 2020-06-02:
@940222JESUS I'm glad we could get you past that. The vxpkg install happens early in the ROS install process, so ROS isn't complete installed.

to complete the install upgrade
```
choco upgrade ros-melodic-desktop_full -y --execution-timeout=0
```

This will resume where it left off. Afterward, reopen the ROS prompt. (Make sure that setup.bat is called), then you can check out ROS repos.

Thanks again!
Lou

Wal1082, 2020-06-09:
[image]
 Hi! Please help me get through with this issue. I am installing ROS melodic in windows 10.
Thank you.

seanyen, 2020-06-09:
@Wal1082 It seems that it cannot reach out to the powershell URL. Are you behind a proxy? Are you able to change the network or disable proxy to retry again?

[4 more replies]

closed 2020-06-28 by ooeygui