ms-iot/ROSOnWindows #207: [Workaround] Failing to upgrade due to  Can not open output file

ooeygui, 2020-04-17:
You may encounter a problem upgrading. This is due to an external change which converts a binary release do a directory. 
```
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\dependency_links.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\PKG-INFO
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\SOURCES.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\top_level.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\dependency_links.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\PKG-INFO
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\SOURCES.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\top_level.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\dependency_links.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\PKG-INFO
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\SOURCES.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\top_level.txt
```

Please delete the affected files and upgrade.

This will be fixed in a future release with:
https://github.com/ms-iot/ros-windows-build/pull/81

ooeygui, 2020-04-27:
This has been fixed in the latest build

closed 2020-04-27 by ooeygui

johnwason, 2020-05-04:
I am still seeing this issue trying to upgrade `ros-melodic-desktop_full v20200409.2`

```
C:\>choco install ros-melodic-desktop_full -y
Chocolatey v0.10.15
Installing the following packages:
ros-melodic-desktop_full
By installing you accept licenses for the packages.
Progress: Downloading ros-melodic-desktop_full 20200409.2... 100%

ros-melodic-desktop_full v20200409.2
ros-melodic-desktop_full package files install completed. Performing other installation steps.
Extracting 64-bit C:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\drop.zip to c:\opt\ros...
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\dependency_links.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\PKG-INFO
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\SOURCES.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\gennodejs-2.0.1-py2.7.egg-info\top_level.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\dependency_links.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\PKG-INFO
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\SOURCES.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\rqt_rviz-0.6.0-py2.7.egg-info\top_level.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\dependency_links.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\PKG-INFO
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\SOURCES.txt
ERROR: Can not open output file : The system cannot find the path specified. : c:\opt\ros\melodic\x64\lib\site-packages\sensor_msgs-1.12.7-py2.7.egg-info\top_level.txt
ERROR: 7-Zip encountered a fatal error while extracting the files. This is most likely an issue with the 'ros-melodic-desktop_full' package and not with Chocolatey itself. Please follow up with the package maintainer(s) directly.
The install of ros-melodic-desktop_full was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.

Chocolatey installed 0/1 packages. 1 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Failures
 - ros-melodic-desktop_full (exited 2) - Error while running 'C:\ProgramData\chocolatey\lib\ros-melodic-desktop_full\tools\chocolateyInstall.ps1'.
 See log for details.

Did you know the proceeds of Pro (and some proceeds from other
 licensed editions) go into bettering the community infrastructure?
 Your support ensures an active community, keeps Chocolatey tip top,
 plus it nets you some awesome features!
 https://chocolatey.org/compare
```