ms-iot/ROSOnWindows #208: python3 support

missxa, 2020-04-22:
Your docs clearly state  

> ROS for Windows expects the 64-bit Python2.7 

but is it possible to build ROS on Windows with python>=3.6 support?

seanyen, 2020-04-22:
@missxa For ROS, there are `ROS 1` and `ROS 2`. For `ROS 2`, the Python 3 is already the target runtime and you can find more information [here](https://index.ros.org/doc/ros2/Installation/Eloquent/Windows-Install-Binary/). For the `ROS 1`, [`Noetic`](https://github.com/ros-infrastructure/rep/blob/master/rep-0003.rst#noetic-ninjemys-may-2020---may-2025) will be the release focused on Python 3.7/3.8 migration. And that would be the first ROS 1 release which you can expect to work with Python 3.7/3.8.

missxa, 2020-04-22:
I meant ROS 1, sorry I wasn't precise about that. 
So noetic releases in May as I can see. Are there any build instructions  from source already available?

ooeygui, 2020-05-22:
You can install noetic prerelease builds by using `choco install ros-noetic-ALL --pre`. Please understand that this is a pre-release package, and most 3rd packages have not been enabled on noetic.

closed 2020-05-22 by ooeygui