ms-iot/ROSOnWindows #213: [ROS2] "RuntimeError: VisualStudioVersion is not set, please run within a Visual Studio Command Prompt." When Running "colcon build --merge-install"

songyuc, 2020-04-30:
**Describe the bug**
I am learning about ROS2 on Windows 10 these days,
and I try building the workspace with `colcon`, with the command `colcon build --merge-install`.
An error appears as,

> D:\dev_ws\src\ros_tutorials>colcon build --merge-install
> Starting >>> turtlesim
> --- stderr: turtlesim
> Traceback (most recent call last):
>   File "c:\python37\lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__
>     rc = await self.task(*args, **kwargs)
>   File "c:\python37\lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__
>     return await task_method(*args, **kwargs)
>   File "c:\python37\lib\site-packages\colcon_ros\task\ament_cmake\build.py", line 72, in build
>     additional_hooks=additional_hooks)
>   File "c:\python37\lib\site-packages\colcon_cmake\task\cmake\build.py", line 86, in build
>     rc = await self._reconfigure(args, env)
>   File "c:\python37\lib\site-packages\colcon_cmake\task\cmake\build.py", line 153, in _reconfigure
>     'VisualStudioVersion is not set, '
> RuntimeError: VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.
> ---
> Failed   <<< turtlesim  [ Exited with code 1 ]
> 
> Summary: 0 packages finished [1.17s]
>   1 package failed: turtlesim
>   1 package had stderr output: turtlesim
> VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.

**To Reproduce**
Steps to reproduce the behavior:
* Follow the tutorial [Creating a workspace](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#creating-a-workspace)
  * follow the guide and commands in the tutorial above
  * in the step 5 -- [5 Build the workspace with colcon](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#build-the-workspace-with-colcon), run the command `colcon build --merge-install`

Then the error appears.

**Expected behavior**
According to the tutorial, it says,
[image]
 

**Environment information:**
 - OS version: `winver`-- 1909
 - Chocolatey Versions: `choco list --local-only`
```bash
Chocolatey v0.10.15
asio 1.12.1
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
cmake 3.17.1
cmake.install 3.17.1
cunit 2.1.3
eigen 3.3.4
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
log4cxx 0.10.0
python 3.7.5
python3 3.7.5
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
vcredist140 14.25.28508.3
vcredist2013 12.0.40660.20180427
20 packages installed.
```

ooeygui, 2020-04-30:
Hi there,
Based on the error, it looks like you aren't running the command from a visual studio command prompt. Can you confirm what the shortcut looks like that launched the command window?

songyuc, 2020-05-04:
Thanks @ooeygui , and I have resolved this issue by the information from [Ros2 windows10 colcon build fails on ros2_example_ws tutorial [closed]](https://answers.ros.org/question/317475/ros2-windows10-colcon-build-fails-on-ros2_example_ws-tutorial/?answer=320193#post-id-320193).

closed 2020-05-04 by songyuc