ms-iot/ROSOnWindows #214: [ROS2] "colon" building turtlesim package failed

songyuc, 2020-05-04:
**Describe the bug**
I am learning about ROS2 on Windows 10 today, 
and the "colon" building turtlesim package failed as the tutorial, [5 Build the workspace with colcon](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#build-the-workspace-with-colcon)

**To Reproduce**
Steps to reproduce the behavior:
* Follow the tutorial [Creating a workspace](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#creating-a-workspace)
  * follow the guide and commands in the tutorial above
  * before the step 5, run the command <br> `call "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat" x86_amd64`
  * in the step 5 -- [5 Build the workspace with colcon](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#build-the-workspace-with-colcon), <br> run the command `colcon build --merge-install`

Then the terminal outputs:

> Starting >>> turtlesim
> [Processing: turtlesim]
> Failed   <<< turtlesim  [ Exited with code 1 ]
Summary: 0 packages finished [35.5s]
  1 package failed: turtlesim

**Expected behavior**
According to the tutorial, it says,
[image]

**Environment information:**
 - OS version: `winver`-- 1909
 - Chocolatey Versions: `choco list --local-only`
```bash
Chocolatey v0.10.15
asio 1.12.1
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
cmake 3.17.1
cmake.install 3.17.1
cunit 2.1.3
eigen 3.3.4
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
log4cxx 0.10.0
python 3.7.5
python3 3.7.5
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
vcredist140 14.25.28508.3
vcredist2013 12.0.40660.20180427
20 packages installed.
```

seanyen, 2020-05-05:
@songyuc  Can you run `colcon build --event-handlers console_cohesion+` for more diagnostic messages? That will help to tell what's error in the workspace.

songyuc, 2020-05-06:
Hi, @seanyen , thanks for your response.
I run the command, and got the information as

> Starting >>> turtlesim
> [31.534s] colcon.colcon_core.event_reactor ERROR Exception in event handler extension 'console_cohesion': 'utf-8' codec can't decode byte 0xd3 in position 2253: invalid continuation byte
> Traceback (most recent call last):
>   File "c:\python37\lib\site-packages\colcon_core\event_reactor.py", line 78, in _notify_observers
>     retval = observer(event)
>   File "c:\python37\lib\site-packages\colcon_output\event_handler\console_cohesion.py", line 53, in __call__
>     self._lines[job]).decode() + \
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd3 in position 2253: invalid continuation byte
> 
> Failed   <<< turtlesim  [ Exited with code 1 ]
> 
> Summary: 0 packages finished [29.4s]
>   1 package failed: turtlesim

Looking forward to your reply!

seanyen, 2020-05-09:
> [31.534s] colcon.colcon_core.event_reactor ERROR Exception in event handler extension 'console_cohesion': 'utf-8' codec can't decode byte 0xd3 in position 2253: invalid continuation byte

@songyuc This is something new to me. By searching the same error messages, it is most likely to do with the system [code page](https://docs.microsoft.com/en-us/windows/win32/intl/code-pages).

Can you try to run `chcp` and report your code page? And also you can try `chcp 65001` to switch your console to use `UTF-8` and retry the colcon command again.

ooeygui, 2020-06-28:
Closing due to inactivity. Please reopen if the codepage change does not work for you.

closed 2020-06-28 by ooeygui