ms-iot/ROSOnWindows #215: [ROS2] Build from source // Exception in shell extension 'bat'

missxa, 2020-05-04:
I'm trying to build ROS2 eloquent from source by following steps described [here](https://index.ros.org/doc/ros2/Installation/Eloquent/Windows-Development-Setup/).
 
When I run (after completing the steps required before that) 
```
cd c:\dev\ros2_eloquent
colcon build --merge-install
``` 
it fails with the following output:
```
colcon.colcon_core.shell ERROR Exception in shell extension 'bat': Expected ['c:\\dev\\ros2_eloquent\\src\\build\\ament_flake8\\colcon_command_prefix_setup_py.bat', '&&', 'set'] to pass
Traceback (most recent call last):
  File "c:\python37\lib\site-packages\colcon_core\shell\__init__.py", line 289, in get_command_environment
    task_name, Path(build_base), dependencies)
  File "c:\python37\lib\site-packages\colcon_core\shell\bat.py", line 128, in generate_command_environment
    env = await get_environment_variables(cmd, cwd=str(build_base))
  File "c:\python37\lib\site-packages\colcon_core\shell\__init__.py", line 325, in get_environment_variables
    output = await check_output(cmd, cwd=cwd, shell=shell)
  File "c:\python37\lib\site-packages\colcon_core\subprocess.py", line 107, in check_output
    assert not rc, 'Expected {args} to pass'.format_map(locals())
AssertionError: Expected ['c:\\dev\\ros2_eloquent\\src\\build\\ament_flake8\\colcon_command_prefix_setup_py.bat', '&&', 'set'] to pass
```

When I call `colcon_command_prefix_setup_py.bat`, it results in `The syntax of the command is incorrect.`

I installed ROS2 eloquent from binaries alongside and it works as expected, so the prerequisites are satisfied. 

**Environment information:**
 - OS version: `Windows 10`
<details>
<summary>Chocolatey Versions</summary>
<p>

```
Chocolatey v0.10.15
asio 1.12.1
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
cmake 3.17.1
cmake.install 3.17.1
cppcheck 1.90
cunit 2.1.3
curl 7.69.1
eigen 3.3.4
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
log4cxx 0.10.0
patch 2.5.9
pip 1.2.0
python3 3.8.2
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
vcredist140 14.25.28508.3
vcredist2013 12.0.40660.20180427
vcredist2015 14.0.24215.20170201
```
</p>
</details>

ooeygui, 2020-05-04:
@lilustga can you take a look at this?

missxa, 2020-05-04:
I thought it might be a problem with my Python (I had 3.7 installed with the official installer), so I installed python 3.7.5 from choco (and removed all other installations), but the issue persists.

ooeygui, 2020-05-04:
Could you elaborate on why you building ROS2 from source?

(Knowing this will help us with documentation and scripts. We'll be working to improve the workflow - both on customer setup as well as build)

missxa, 2020-05-04:
I want to make my node a standalone executable, so I need to compile static libs for rclcpp, etc. 
The idea is to omit the full ROS2 installation on the end-user PC. Does that make sense?

ooeygui, 2020-05-04:
Thank you for the explanation. The installation is quite complex, and we too are investigating how to improve it.

lilustga, 2020-05-04:
I noticed it does say python3 version 3.8.2 in your listed choco packages. Is that outdated from before you switched to 3.7.5?

Is python added to the path? If not, try adding it before building e.g:
`set "PATH=C:\opt\python37amd64\DLLs;C:\opt\python37amd64\;C:\opt\python37amd64\Scripts`

missxa, 2020-05-04:
yea, 3.8.2 was outdated (also then I was using the official installation with 3.7, not choco). 

Thanks for the pointer to look at the `PATH`. I did have python in the `PATH` (only `DLLs` folder was missing), but my pythonpath was `PYTHONPATH=c:\Python37"` and likely that quotation was causing path issue in batch files... 🤦

closed 2020-05-04 by missxa