ms-iot/ROSOnWindows #216: wstool fails on ros_tutorials

yrzhanov, 2020-05-05:
Running
wstool init src melodic-ros_comm.rosinstall
proceeds fine until:

Exception caught during install: Error processing 'ros_tutorials' : [ros_tutorials] Checkout of https://github.com/ms-iot/ros_tutorials.git version init_windows into C:\ros_catkin_ws\src\ros_tutorials failed.

ERROR in config: Error processing 'ros_tutorials' : [ros_tutorials] Checkout of https://github.com/ms-iot/ros_tutorials.git version init_windows into C:\ros_catkin_ws\src\ros_tutorials failed.

The next step:
C:\ros_catkin_ws>rosdep install --from-paths src --ignore-src --rosdistro melodic -r -y

fails for every entry:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
test_rosbag: No definition of [bzip2] for OS [windows]
python_qt_binding: No definition of [qtbase5-dev] for OS [windows]
nodelet: No definition of [uuid] for OS [windows]
....

ooeygui, 2020-05-05:
Hi there,
We don't have a fork of ros_tutorials. We have upstreamed all of the work there, so it should use the mainline rosintall instructins.

We need to fix the documentation - where did you find instructions where it says to use ms-iot?

yrzhanov, 2020-05-05:
Hi Lou,

It’s on:

https://ms-iot.github.io/ROSOnWindows/Build/source.html

 

Should I use other instructions?

 

The following failure I mentioned is not related to tutorials, is it? But I thought that there might be something AFTER the tutorials that lead to failure in the next step.

 

Thanks,

    Yuri

 

===============================================================================

Yuri Rzhanov

Center for Coastal and Ocean Mapping/Joint Hydrographic Center

University of New Hampshire

Tel: 1-603-8620075

 

Let go all your heartaches, it's as easy as being free 
You don't have to wait a while

lilustga, 2020-06-01:
Hi @yrzhanov,

I haven't been able to repro this issue while using those instructions. I think the solution here is to find here is to find where you have a dependency on the no longer existent ms-iot fork of ROS tutorials.

Can you provide more information on how you set up your environment? What steps did you run before the ones described on https://ms-iot.github.io/ROSOnWindows/Build/source.html?

Thanks,
Lior

yrzhanov, 2020-06-01:
Hi Lior,
I repeated all the steps, and now running:
wstool init src melodic-ros_comm.rosinstall

returns:
Error: There already is a workspace config file .rosinstall at "src". Use wstool install/modify.

This file indeed exists and contains entries like:
- git:
    local-name: actionlib
    uri: https://github.com/ros/actionlib.git
    version: melodic-devel

However, the next step:
rosdep install --from-paths src --ignore-src --rosdistro melodic -r -y

gives the same:
C:\ros_catkin_ws>rosdep install --from-paths src --ignore-src --rosdistro melodic -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nodelet: No definition of [uuid] for OS [windows]
tf2_bullet: No definition of [bullet] for OS [windows]
desktop: No definition of [ros_tutorials] for OS [windows]
....

Which definitions rodsep is looking for?

Thanks,
    Yuri

ooeygui, 2020-06-01:
@yrzhanov Thank you for the prompt reply.
There are a couple of things to be aware of.

- Windows release for ROS includes many sub-packages contained within the metapackage. Because of this, rosdep can't see them. Some errors are actually not missing (like bullet).
- the ROS ecosystem contains thousands of packages, not all have been enabled or released on Windows.
- If a package hasn't been released as a chocolatey package (either by Microsoft or the package owner), then you can check the sources out and build them in your environment. For uuid, you can clone https://github.com/ros-geographic-info/unique_identifier into your workspace.

We are working on expanding the number of binary deployments by using github actions to build packages, and a rosdep extension which will extract the binaries off github.

ooeygui, 2020-07-02:
Closing the issue. This should be a warning, not an error.
rosdep is not informed of how ROS is packaged on Windows, so issues errors. We are not planning on repackaing to eliminate these errors.

closed 2020-07-02 by ooeygui