ms-iot/ROSOnWindows #228: [ROS2] rqt does not launch

ooeygui, 2020-06-03:
RQT does not currently launch due to missing Qt binaries. (however, rViz does).

HorvathDawson, 2020-06-16:
any update on this issue or fix? I am encountering it after installing the chocolately packages for eloquent. I also experienced a very similar issue with foxy binaries on the ROS website.

ooeygui, 2020-06-16:
Not yet, We're working on this and related environment issues.

HorvathDawson, 2020-06-16:
I will be following closely hopefully the fix will extend to foxy! 

Side note, thank you for all the work your team has done for this project, it has been insanely helpful :-)

DawsonAbCellera, 2020-06-19:
https://answers.ros.org/question/354707/qtqpaplugin-could-not-find-the-qt-platform-plugin-windows-in/ this might be applicable

HorvathDawson, 2020-06-19:
might be applicable to the solution, 
https://answers.ros.org/question/354707/qtqpaplugin-could-not-find-the-qt-platform-plugin-windows-in/

ooeygui, 2020-06-19:
Thank you for investigating. 
We had done some work in the python_qt_plugin for ROS1 for this exact scenario; so I was tracing through that path. I think the environment variable is a good solution - I'm just worried about environment load on Windows - there's a maximum for the entire environment size.

ooeygui, 2020-06-19:
This works - we will add it to the install script:
`set QT_QPA_PLATFORM_PLUGIN_PATH=c:\opt\ros\eloquent\x64\bin\platforms`

HorvathDawson, 2020-06-19:
Awesome!

closed 2020-06-19 by ooeygui

seanyen, 2020-06-19:
The fix should be in the next daily build. :)

HorvathDawson, 2020-06-20:
Did your rqt load in with icons? Because with this fix mine has no icons

ooeygui, 2020-06-20:
Unfortunately, Icons not loading in has been a general problem in ROS1 as well, https://github.com/ms-iot/ROSOnWindows/issues/162

I'm currently exercising various parts of Rqt, and will likely find other problems. We will create them as individual issues.

Please note that our fix is in the Microsoft chocolatey package - documented here: https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html

We're working with Open Robotics to improve the main docs.