ms-iot/ROSOnWindows #229: 'rviz_plugin_tutorials/Imu' could not be loaded.

jcursiolf, 2020-06-04:
**Describe the bug**
Trying to see IMU data from a rosbag file in rviz returns failure:

> **Imu**
The class required for this display, 'rviz_plugin_tutorials/Imu', could not be loaded.
**Error:**
Could not find library corresponding to plugin rviz_plugin_tutorials/Imu. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

**To Reproduce**
Steps to reproduce the behavior:

- New ros-melodic-desktop_full installation following ROS documentation steps.
- cmd shortcut target: `%windir%\system32\cmd.exe /k "C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\Tools\VsDevCmd.bat" -arch=amd64 -host_arch=amd64 && set ChocolateyInstall=c:\opt\chocolatey && c:\opt\ros\melodic\x64\setup.bat`
- Steps:
  - Open a first cmd window through the shortcut, run `roscore`
  - Open a second cmd window through the shortcut, navigate to rosbag folder and run `rosbag play imu.bag -l`
  - Open a third cmd window through the shortcut, run `rviz`, which outputs:
    ```
    [ INFO] [1591303133.597871600]: rviz version 1.13.11
    [ INFO] [1591303133.598111000]: compiled against Qt version 5.10.1
    [ INFO] [1591303133.598217700]: compiled against OGRE version 1.10.11 (Xalafu)
    [ INFO] [1591303133.618324100]: Forcing OpenGl version 0.
    [rospack] Warning: cannot create rospack cache directory : boost::filesystem::create_directory: The system cannot find the path specified
    [rospack] Warning: cannot create rospack cache directory : boost::filesystem::create_directory: The system cannot find the path specified
    [ INFO] [1591303138.407403200]: Stereo is NOT SUPPORTED
    [ INFO] [1591303138.407992100]: OpenGl version: 4.5 (GLSL 4.5).
    ```
  - Open the rivz window, click on "Add" to add a new display.
  - In the "by display type" tab, select "Imu" in the folder "rviz_plugin_tutorials". Otherwise, select the topic in the "by topic" tab. The output in the command prompt is:
    ```
    [ERROR] [1591303152.175136900]: PluginlibFactory: The plugin for class 'rviz_plugin_tutorials/Imu' failed to load.  Error: Could not find library corresponding to plugin rviz_plugin_tutorials/Imu. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
    ```

    [image]

**Expected behavior**
See image bellow with the same rosbag file in a Linux environment.

[image]

**Environment information:**
 - OS version: `winver`
    Microsoft Windows 10 Enterprise
    Version 1909 (OS Build 18363.836)
 - Chocolatey Versions: `choco list --local-only`
    ```
    Chocolatey v0.10.15
    assimp 4.0.1.1809242322
    boost 1.66.0.1904102339
    bullet3 2.87.0.1809242326
    bzip2 1.0.6.1901242108
    chocolatey 0.10.15
    chocolatey-core.extension 1.3.5.1
    chocolatey-windowsupdate.extension 1.0.4
    chocolateygui 0.17.0
    console_bridge 0.4.0.1809242322
    cppunit 1.12.1.1809242322
    cppzmq 4.2.2.1811162207
    dlfcn-win32 1.1.1.1811162226
    DotNet4.5.2 4.5.2.20140902
    eigen 3.3.4.1809242322
    freeimage 3.17.0.1811160037
    gazebo9 9.4.1.1909060324
    git 2.26.2
    git.install 2.26.2
    google-mock 1.8.0.1809242326
    google-test 1.8.0.1809242322
    gtk2 2.22.1.1904131648
    ilmbase 2.2.1.1811210800
    jasper 2.0.14.1811210919
    jxrlib 1.1.0.1811210809
    KB2919355 1.0.20160915
    KB2919442 1.0.20160915
    KB2999226 1.0.20181019
    KB3033929 1.0.5
    KB3035131 1.0.3
    lcms 2.8.0.1811210903
    libcurl 7.58.0.1809242322
    libflann 1.9.1.1809090720
    libgraphviz 2.41.0.1810022039
    liblzma 5.2.4.1811210837
    libogg 1.3.3.1809090720
    libopencv 3.4.1.1809242322
    libpcl 1.8.1.1809090720
    libpng 1.6.35.1809242322
    libraw 0.19.0.1811210804
    libtbb-dev 2018.6.0.1812011750
    libtheora 1.1.1.1809090720
    libwebp 0.6.1.1811210731
    log4cxx 0.10.0.1809242322
    lz4 1.8.1.1809242322
    ogre 1.10.11.1901312214
    openexr 2.2.1.1811210832
    openjpeg 2.3.0.1811210736
    openssl 1.1.1.1809242322
    pkg-config 0.29.2.1809242322
    poco 1.8.1.1809242322
    protobuf 3.6.1.1811130545
    pyqt5 5.10.1.1809242322
    pyside2 5.10.1.1809242322
    qt5-sdk 5.10.1.1903182340
    qwt 6.1.3.1811212304
    ros-catkin-tools 0.0.1.1903221831
    ros-melodic-desktop_full 20200501.1
    ros-python2 2.7.15.1904300710
    ros-vcpkg 2018.11.23.1906112226
    sbcl 0.0.0.1809242326
    sdl 1.2.15.1809242322
    sdl_image 1.2.12.1809242322
    sip 4.19.8.1809242322
    tiff 4.0.9.1811210742
    tinyxml 2.6.2.1809242322
    tinyxml2 6.1.0.1809242322
    urdfdom 1.0.0.1809242322
    urdfdom_headers 1.0.0.1809242322
    vcpython27 9.0.30729.1809090720
    vcredist140 14.26.28720.3
    vcredist2010 10.0.40219.2
    yaml-cpp 0.6.2.1901170034
    zeromq 4.2.5.1811162203
    zlib 1.2.11.1809242322
    75 packages installed. 
    ```

seanyen, 2020-06-09:
@jcursiolf Thanks for reporting. We will be looking into this soon.

HanDaSeul, 2020-08-06:
@seanyen I just want to ask same exactly happened on mine, too. Was that solved?

seanyen, 2020-09-22:
The IMU loading issue should be addressed in the Noetic release: https://discourse.ros.org/t/ros-on-windows-noetic-release-v20200831-0-0-2009101215/16352

I am closing it first and feel free to reopen if you still see the IMU loading problems.

closed 2020-09-22 by seanyen