ms-iot/ROSOnWindows #231: ROS2 Foxy Binary Installation Incomplete?

stwolfe, 2020-06-09:
I just finished walking through the binary installation of ros2_foxy, and the basics seem to be working just fine.  I can run:

1. ros2 run demo_nodes_cpp talker
2. ros2 run demo_nodes_py listener

But when I run a command like: **ros2 node list**

I get the following error:

Failed to load entry point 'list': [WinError 193] %1 is not a valid Win32 application
Traceback (most recent call last):
  File "C:\dev\ros2_foxy\Scripts\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.9.5', 'console_scripts', 'ros2')()
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2cli\cli.py", line 39, in main
    add_subparsers_on_demand(
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2cli\command\__init__.py", line 249, in add_subparsers_on_demand
    extension.add_arguments(
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2node\command\node.py", line 25, in add_arguments
    add_subparsers_on_demand(
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2cli\command\__init__.py", line 236, in add_subparsers_on_demand
    extension = command_extensions[name]
KeyError: 'list'

**Environment information:**
 - OS version: Windows 10 2004 (OS Build 19041.264)
 - Chocolatey Versions:

Chocolatey v0.10.15
asio 1.12.1
bullet 2.89.0
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
cmake 3.17.3
cmake.install 3.17.3
cunit 2.1.3
eigen 3.3.4
graphviz 2.38.0.20190211
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
log4cxx 0.10.0
python 3.8.3
python3 3.8.3
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
vcredist140 14.26.28720.3
vcredist2013 12.0.40660.20180427
vcredist2015 14.0.24215.20170201

seanyen, 2020-06-09:
@stwolfe Hi, thanks for your reporting. Can you share what the instructions you are following for ROS 2 installation?

stwolfe, 2020-06-09:
@seanyen Thanks for the quick response.  Here is the link I was following...

https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Install-Binary/

stwolfe, 2020-06-10:
And just to add a little color here.  All of the cli commands do the same thing:

ros2 node list
ros2 topic list
ros2 service list
ros2 action list

From what I can tell in trying to debug it is in some LoadLibrary call, but i can't be sure what it is actually trying to load.

seanyen, 2020-06-10:
@stwolfe Based on the error messages, it is most likely you are missing some dependencies or mixing the 32-bit Python with the 64-bit binaries. You can identify what's the `Python.exe` invoked here and use `python -c "import sys; print(sys.version)"` to see if it is a 32-bit one or 64-bit one.

In additions, our team hosts [ROS 2 Eloquent](https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html) build as Chocolatey package. This is an alternative way to install ROS 2 on Windows too.

HorvathDawson, 2020-06-15:
will there be a foxy chocolately package released in the near future?

ooeygui, 2020-06-15:
Hi @HorvathDawson ,
Yes, we are working on it.

stwolfe, 2020-06-17:
@seanyen @ooeygui 

I did finally get the install of ROS2 Foxy working on my system.  It was a combination of things, but basically following the script here: https://index.ros.org/doc/ros2/Installation/Foxy/.  That is after cleaning up almost the entire system and starting over.  

Is there a way to get a notification when your chocolatey package is ready?

Also I tried to get the ros-dotnet package building with my ROS2 Foxy install that was working, but that failed too.  Is there any "official" .NET Bindings for ROS2?

ooeygui, 2020-06-28:
We are contributing to `https://github.com/ros2-dotnet/ros2_dotnet`. ROS2.net currently doesn't support Foxy. 

When the Foxy builds are ready, we'll post a notification to https://discourse.ros.org/.

seanyen, 2020-09-22:
Closing it. The Foxy release was [out](https://discourse.ros.org/t/ros-on-windows-foxy-release-v20200912-0-0-2009161641).

closed 2020-09-22 by seanyen