ms-iot/ROSOnWindows #232: [ROS2 foxy] issues with loading rqt

HorvathDawson, 2020-06-15:
The bug has three parts to it:
1. After installing foxy binaries from https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Install-Binary/ there is an issue where it requires me to install Qt and set the environment variable similar to installing foxy from source. Not sure if this is an oversight in installation instructions or if this is a bug.

2. Once Qt is installed and RQt launches it gives me this stack trace 
`RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_foxy\install\share\qt_gui_cpp/plugin.xml" in package "qt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_foxy\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_foxy\install\share\qt_gui_cpp/plugin.xml" in package "qt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_foxy\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found`

3. RQt after loading has no icons anywhere on the plugin so the formatting looks really bad and hard to follow.

**Environment information:**
 - OS version: 1909 OS build 18363.900
 - Chocolatey Versions: 
asio 1.12.1
bullet 2.89.0
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
cmake 3.17.3
cmake.install 3.17.3
cppcheck 2.0
cunit 2.1.3
curl 7.70.0
eigen 3.3.4
git 2.27.0
git.install 2.27.0
graphviz 2.38.0.20190211
hg 5.3.2
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
log4cxx 0.10.0
peazip 7.3.1
peazip.install 7.3.1
python 3.8.3
python3 3.8.3
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
vcredist140 14.26.28720.3
vcredist2013 12.0.40660.20180427
vcredist2015 14.0.24215.20170201

HorvathDawson, 2020-06-15:
more issues I have noticed: 

- When I try to follow this tutorial https://index.ros.org/doc/ros2/Tutorials/Turtlesim/Introducing-Turtlesim/#turtlesim  I run into an issue right in the beginning where you launch the turtlesim node... I get this error:
`qt.qpa.plugin: Could not find the Qt platform plugin "windows" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.`
- Also when I just try to launch rqt_console by itself I get another error:
`Traceback (most recent call last):
  File "C:\dev\ros2-windows\Scripts\ros2-script.py", line 11, in <module>
    load_entry_point('ros2cli==0.9.5', 'console_scripts', 'ros2')()
  File "C:\dev\ros2-windows\Lib\site-packages\ros2cli\cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2-windows\Lib\site-packages\ros2run\command\run.py", line 70, in main
    return run_executable(path=path, argv=args.argv, prefix=prefix)
  File "C:\dev\ros2-windows\Lib\site-packages\ros2run\api\__init__.py", line 61, in run_executable
    process = subprocess.Popen(cmd)
  File "c:\python38\lib\subprocess.py", line 854, in _init_
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "c:\python38\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 193] %1 is not a valid Win32 application`

ooeygui, 2020-06-28:
We're working on a Foxy chocolatey release which encapsulates all dependencies and environment.

seanyen, 2020-09-22:
@HorvathDawson Recently a Foxy release is [out](https://discourse.ros.org/t/ros-on-windows-foxy-release-v20200912-0-0-2009161641) and it encapsulates all the dependencies. Would you like to give it a try?

I am closing this first since the installation was not from our project. Feel free to reopen an new issue if you hit anything there.

closed 2020-09-22 by seanyen