ms-iot/ROSOnWindows #237: Errors occurs while using rosbag: Could not find library corresponding to plugin

bfeng1, 2020-06-30:
When I run the command

    rosbag play D:\summer_project\20200619_123634.bag

I got the following error:

    [FATAL] [1593471809.093414800]: Could not find library corresponding to plugin rosbag/NoEncryptor. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

However, after I check the folder location, c:\opt\ros\noetic\x64\include\rosbag, I found the file no_encryptor.h , but not sure why the rosbag cannot find the library.

**To Reproduce**
Steps to reproduce the behavior:
followed the tutorial installation for ros-noetic on windows 10:
http://wiki.ros.org/Installation/Windows

After the installation, I run the "roscore" and checked the rosbag functionality by using the command " rosbag play" for one of my example bag file.

    rosbag play D:\summer_project\20200619_123634.bag

**Expected behavior**
I expected that the ros was able to read the bag file sucessfully and play the recored videos stored in the bag file.

**Environment information:**
 - OS version: Windows 10
 - Chocolatey Versions: v0.10.15

seanyen, 2020-07-01:
@bfeng1 Hi! So the `rosbag/NoEncryptor` plugin should be mapped to this library:
`C:\opt\ros\melodic\x64\bin\rosbag_default_encryption_plugins.dll`

If the file does exist, one possible is something going wrong as it gets loaded. Can you share your environment variable `%PATH%` under the command prompt you are running `rosbag play`? (For example, type `set PATH` and share the output.)

ooeygui, 2020-07-02:
Would it be possible for you to post the rosbag?

bfeng1, 2020-07-05:
@seanyen Hi Thank you for responding! I found the file rosbag_default_encryption_plugins.dl in the folder that you mentioned. Here is the output after I ran "set PATH". 

```
C:\Windows\System32>set PATH
Path=C:\opt\rosdeps\x64\bin;C:\opt\ros\noetic\x64\bin;C:\opt\python37amd64\Scripts;C:\opt\python37amd64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\\Extensions\Microsoft\IntelliCode\CLI;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.8 Tools\x64\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\FSharp\;C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\x64;C:\Program Files (x86)\Windows Kits\10\bin\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\\MSBuild\Current\Bin;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\Tools\;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\libnvvp;F:\python_related\python38\Scripts\;F:\python_related\python38\;F:\New folder (2)\Scripts\;F:\New folder (2)\;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;D:\Matlab_student\bin;C:\Program Files\NVIDIA Corporation\Nsight Compute 2019.5.0\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\CMake\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;C:\Program Files\dotnet\;C:\Users\binfe\AppData\Local\Microsoft\WindowsApps;D:\2019 spring\microprocessor\quartus\modelsim_ase\win32aloem;F:\python_related\pycharm\PyCharm Community Edition 2019.3.2\bin;C:\Users\binfe\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\binfe\AppData\Local\atom\bin;C:\Program Files\Mercurial\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\VC\Linux\bin\ConnectionManagerExe
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW
```

bfeng1, 2020-07-05:
@ooeygui Hi I can share the bag file I have been testing, but the githubd did not support the bag file type. Also I tried different bag files and all of them give me the same error. So I do not think the issue is from the bag file I am using. [](url)

ooeygui, 2020-07-06:
@bfeng1 Thank you for trying to post it.

alec3010, 2020-07-14:
Hey guys, I am running into the same issue on my windows installation with my rosbag file that ran just fine on linux. I made shure that my installation is not corrupt by reinstalling but the issue actually persists. If you want I could try and share my bagfile as well. But at this point it rather looks like a bug in the ros for windows installation than an issue with the bagfile.

This is the content of my %PATH%-variable: 

> C:\opt\rosdeps\x64\bin;C:\opt\ros\noetic\x64\bin;C:\opt\python37amd64\Scripts;C:\opt\python37amd64;C:/IFL/compression_compare_win/devel\bin;C:/IFL/compression_compare_win/devel\lib;C:/opt/rosdeps/x64\bin;C:/opt/rosdeps/x64\lib;C:/opt/ros/noetic/x64\bin;C:/opt/ros/noetic/x64\lib;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\\Extensions\Microsoft\IntelliCode\CLI;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\MSBuild\Current\bin\Roslyn;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Team Tools\Performance Tools\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\x64\;C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\x64;C:\Program Files (x86)\Windows Kits\10\bin\x64;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\\MSBuild\Current\Bin;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\Tools\;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\libnvvp;C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\MiKTeX 2.9(1)\miktex\bin\x64\;C:\WINDOWS\system32\config\systemprofile\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\GNU\GnuPG\pub;C:\Program Files (x86)\Skype\Phone\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;C:\Program Files\NVIDIA Corporation\Nsight Compute 2020.1.0\;C:\Program Files\dotnet\;C:\Users\Alexander\AppData\Local\Microsoft\WindowsApps;C:\Users\Alexander\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja;C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\VC\Linux\bin\ConnectionManagerExe

If you need any further information, do not hesitate to contact me

seanyen, 2020-09-22:
@bfeng1 @alec3010 We have a new release https://discourse.ros.org/t/ros-on-windows-noetic-release-v20200831-0-0-2009101215/16352. Can you take a look if this is addressed by this version? Thanks!

IoTDan, 2022-10-11:
closing due to inactivity

closed 2022-10-11 by IoTDan