ms-iot/ROSOnWindows #243: [ROS2][Eloquent] Multiple lifecycle_manager error messages

seanyen, 2020-07-03:
**Describe the bug**

When running `Navigation 2` [exercise](https://ms-iot.github.io/ROSOnWindows/ros2/nav2.html), here are some random error noticed during the execution.

Lifecycle Manager doesn’t start

```
[rviz2.EXE-9] [INFO] [rviz2]: Waiting for the lifecycle_manager/is_active service...
```

When shutting down, the lifecycle manager complains about Bringup node failing
```
[lifecycle_manager.EXE-1] [ERROR] [lifecycle_manager]: Failed to bring up nodes: aborting bringup
```

When Ctrl-C out of a `ros2 launch`, nodes reported that they died, instead of gracefully exit.
```
[ERROR] [lifecycle_manager.EXE-1]: process has died [pid 3528, exit code 3221226505, cmd 'c:\ws\nav2_ws\install\nav2_lifecycle_manager\lib\nav2_lifecycle_manager\lifecycle_manager.EXE --ros-args -r __node:=lifecycle_manager --params-file C:\Users\lamadio\AppData\Local\Temp\launch_params_0gb2w2b3 --params-file C:\Users\lamadio\AppData\Local\Temp\launch_params_voyje9fi --params-file C:\Users\lamadio\AppData\Local\Temp\launch_params_70vn5xg6'].
```

jediofgever, 2020-09-22:
I am seeing same warn   on Linux machine

Inkrr, 2021-12-07:
I'm running into the same "Waiting for the lifecycle_manager/is_active service..." with my current code.

Did you figure out how to fix that?

ooeygui, 2021-12-07:
@Inkrr Are you using Eloquent or Foxy? [Eloquent is end of life](https://dlu.github.io/ros_clock/index.html). 
The Nav2 instructions for the Windows chocolatey Foxy release have been updated and validated here: on [http://aka.ms/ros](https://ms-iot.github.io/ROSOnWindows/ros2/nav2.html).