ms-iot/ROSOnWindows #247: rviz start causes errors due to missing dll and plugin classes

RolfWojtech, 2020-07-03:
**Describe the bug**
After following the installation instructions on http://wiki.ros.org/Installation/Windows (LKG with --pre) launching roscore works but launching rviz causes multiple errors.
First an error box with white X on red circle pops up with the text (translated): the execution of code cannot be continued because image_transport.dll cannot be found. A reinstallation might fix this problem". When clicking ok, the same message pops up for resource_retriever.dll".
I then added C:\opt\ros\noetic\x64\lib to the system path environment variable manually. Now rviz launches but gives the following errors:

```
c:\ws>rviz
[ INFO] [1593788256.216587500]: rviz version 1.14.0
[ INFO] [1593788256.216732000]: compiled against Qt version 5.10.1
[ INFO] [1593788256.216825700]: compiled against OGRE version 1.10.11 (Xalafu)
[ INFO] [1593788256.223010400]: Forcing OpenGl version 0.
[ INFO] [1593788256.886505600]: Stereo is NOT SUPPORTED
[ INFO] [1593788256.886633200]: OpenGl version: 4.6 (GLSL 4.6).
[ERROR] [1593788256.935177700]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Could not find library corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.851570400]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: Could not find library corresponding to plugin rviz/MoveCamera. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.858282700]: PluginlibFactory: The plugin for class 'rviz/Interact' failed to load.  Error: Could not find library corresponding to plugin rviz/Interact. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.862627800]: PluginlibFactory: The plugin for class 'rviz/Select' failed to load.  Error: Could not find library corresponding to plugin rviz/Select. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.866919900]: PluginlibFactory: The plugin for class 'rviz/SetInitialPose' failed to load.  Error: Could not find library corresponding to plugin rviz/SetInitialPose. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.870725300]: PluginlibFactory: The plugin for class 'rviz/SetGoal' failed to load.  Error: Could not find library corresponding to plugin rviz/SetGoal. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.875336100]: PluginlibFactory: The plugin for class 'rviz/Grid' failed to load.  Error: Could not find library corresponding to plugin rviz/Grid. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.879136700]: PluginlibFactory: The plugin for class 'rviz/Interact' failed to load.  Error: Could not find library corresponding to plugin rviz/Interact. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.882986300]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: Could not find library corresponding to plugin rviz/MoveCamera. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.888278800]: PluginlibFactory: The plugin for class 'rviz/Select' failed to load.  Error: Could not find library corresponding to plugin rviz/Select. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.892044000]: PluginlibFactory: The plugin for class 'rviz/FocusCamera' failed to load.  Error: Could not find library corresponding to plugin rviz/FocusCamera. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.896296400]: PluginlibFactory: The plugin for class 'rviz/Measure' failed to load.  Error: Could not find library corresponding to plugin rviz/Measure. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.900515500]: PluginlibFactory: The plugin for class 'rviz/SetInitialPose' failed to load.  Error: Could not find library corresponding to plugin rviz/SetInitialPose. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.904058800]: PluginlibFactory: The plugin for class 'rviz/SetGoal' failed to load.  Error: Could not find library corresponding to plugin rviz/SetGoal. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.907484000]: PluginlibFactory: The plugin for class 'rviz/PublishPoint' failed to load.  Error: Could not find library corresponding to plugin rviz/PublishPoint. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1593788273.911491000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Could not find library corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```
These could be follow up errors of me adding the dll files to the search path manually, not sure what the canonical way would be. I would have assumed setup.bat does this somehow.
The same error seems to affect another user as well who posted it into a closed issue: https://github.com/ros-visualization/rviz/issues/1384#issuecomment-629671290

Any help would be greatly appreciated.

**To Reproduce**
Steps to reproduce the behavior:
* Install according to http://wiki.ros.org/Installation/Windows (LKG with --pre)
* Launch roscore
* Launch rviz 
* Error messages pop up for missing dlls (image_transport.dll, resource_retriever.dll)
* Add C:\opt\ros\noetic\x64\lib to the system path environment variable
* Launch rviz again (Now the dll is found but you get 16 errors on rviz launch)

**Expected behavior**
The dll files should be found directly after following the tutorial.
rviz should launch without error messages for missing classes

**Environment information:**
 - OS version: Microsoft Windows [Version 10.0.18363.900]
- Not sure if relevant but the machine already had a Qt 5.10.0 installation previously
 - Chocolatey Versions: `choco list --local-only`
```
Chocolatey v0.10.15
assimp 4.0.1.1809242322
boost 1.66.0.1904102339
bzip2 1.0.6.1901242108
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
chocolateygui 0.17.1
console_bridge 0.4.0.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
DotNet4.5.2 4.5.2.20140902
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.13.1.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk3 3.22.19.1809242322
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
orocos_kdl 1.3.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
python3-pyqt5 5.10.1.1904112255
python3-sip 4.19.15.1904120807
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-colcon-tools 0.0.1.1905240527
ros-noetic-desktop_full 20200703.1-pre
ros-python37 3.7.2.1903082244
sbcl 0.0.0.1809242326
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcredist140 14.26.28720.3
vcredist2010 10.0.40219.2
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
67 packages installed.

```

ooeygui, 2020-07-03:
Thank you for the bug, we'll look into it asap.

kucar17, 2020-08-30:
Hey there, can you give me any info about the problem? I had this ".dll" error with both Gazebo and Rviz. Adding the lib folder to path cleared the problem with Gazebo but I still have the same ".dll" error with Rviz.

seanyen, 2020-09-22:
This should be addressed in this [Noetic 20200831.0.0.2009101215](https://discourse.ros.org/t/ros-on-windows-noetic-release-v20200831-0-0-2009101215) release. I am closing it now and feel free to reopen if you still have this DLL loading issue. Thanks!

closed 2020-09-22 by seanyen