ms-iot/ROSOnWindows #248: TypeError in rosfindpath.py when using roscd

Brac24, 2020-07-09:
When running roscd roscpp I get a type error in rosfindpath.py on line 74 in which it expects an str object but instead gets a bytes object.
[image]

This is the line the caused the error
[image]

I identified that the culprit was the package_dir variable and investigated further where this variable was set which is line 69 by calling ros_location_find(package_name)
[image]

The return value result_location in the ros_location_find() method is what is setting the package_dir. The result_location variable is being set on either line 40 or 46 which uses the communicate() method which says that by default input and output will be in bytes.
[image]

If I decode package_dir this error is fixed
[image]

The roscd.bat file also tries to change directories doing: cd \d "%target_path%".
This would give me the below error.
[image]
Here I was debugging by printing the %target_path% variable that the batch file is using and then it gives me the error.

Removing the \d fixed this error for me.

The error is essentially fixed for me I just don't know if these changes would be acceptable or if there is a different issue going on.

-----------------------------------------------------------------
Installed ROS noetic

Ran setup.bat as described in tutorials many of the commands are working.

I am using the Windows Terminal app with the below commandline setting in settings.json:
"C:\\Windows\\System32\\cmd.exe /k \"C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\VsDevCmd.bat\" -arch=amd64 -host_arch=amd64 && set ChocolateyInstall=c:\\opt\\chocolatey&& c:\\opt\\ros\\noetic\\x64\\setup.bat"

This error occurred regardless if it was command prompt or Windows Terminal.

Catkin workspace is under C:\Users\brac2\~\catkin_ws 
(was trying to follow the linux tutorial hence the ~)
ran catkin_make and everything built correctly

Using rospack find roscpp returned the correct path

Then tried using roscd roscpp to go to that path and it failed

Expected when using roscd roscpp to go into the path:
C:\opt\ros\noetic\x64\share\roscpp

-----------------------------------------------------------------------
Environment
ROS noetic
Windows 10 Home Version 1909 (OS Build 18363.900)
Chocolatey v0.10.15

ooeygui, 2020-07-09:
Thank you for the detailed report!

lilustga, 2021-08-10:
Closing issue as the fix is now in Noetic

closed 2021-08-10 by lilustga