ms-iot/ROSOnWindows #250: 📢 https://aka.ms/ros/public - The new Chocolatey feed URL for ROS on Windows

seanyen, 2020-07-15:
Hi ROS on Windows Developers!

This is an announcement to inform you that now we adopted an new Chocolatey feed URL: https://aka.ms/ros/public.

If you are an new user, please follow ROS Wiki for the new installation: http://wiki.ros.org/Installation/Windows

If you are an existing user, please reconfigure your Chocolatey sources to get the future package releases.

Here are the simple 3 steps to switch to the new location:

1. Open a command prompt as administrator.

2. Run the commands as follow, which will replace your old source with the new one. 

```Batchfile
set ChocolateyInstall=c:\opt\chocolatey
choco source remove -n=ros-win
choco source add -n=ros-win -s="https://aka.ms/ros/public" --priority=1
```

3. Check your Chocolatey sources. You should see something as below:

```Batchfile
>choco source list
Chocolatey v0.10.15
chocolatey - https://chocolatey.org/api/v2/ | Priority 0|Bypass Proxy - False|Self-Service - False|Admin Only - False.
ros-win - https://aka.ms/ros/public | Priority 1|Bypass Proxy - False|Self-Service - False|Admin Only - False.
```

That's it!

Happy coding ROS Developers! 💪💪💪

ldepn, 2021-03-04:
How to change the install path of the ROS?

ooeygui, 2021-03-04:
@idepn,
We do not currently support changing the default install path. In the inherited ROS build process, paths are baked into the generated files. We have investigated this. 

Can you help me understand your scenario where the default path is undesirable?

ldepn, 2021-03-04:
@ooeygui The space of the system disk is limited，so I want to change in D: disk

ooeygui, 2021-03-04:
@ldepn,
I'm in the same boat. You can use [Windows symbolic links](https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/mklink) to create a `c:\opt` directory which links to a folder larger drive.

ldepn, 2021-03-04:
@ooeygui  thanks！

roycey, 2021-06-22:
This issue resolved, and hence closing

closed 2021-06-22 by roycey