ms-iot/ROSOnWindows #251: using android ros control app with windows 10

HanDaSeul, 2020-07-22:
**Describe the bug**
I'm not using VM or wls, just windows10 with ros melodic. And when i try using my android 10(also tried with android 4.4, 8 too) with control app that is 'ros control', i checked on my windows10 that connected well between phone with windows10, then I put command 'rostopic echo /joy_teleop/cmd_vel'(exact or something similar). When I trig or back the joystick on my phone screen, my windows 10 shows my values and it can be drawn on rqt, too. But the problem is, turtlesim doesn't show any response with my phone, and when I enabled my win_camera, i couldn't see video on my phone.

**Environment information:**
 - OS version: `windows10`, `android10`
 - Chocolatey Versions: `v0.10.15`

[image]
[image]

ooeygui, 2020-07-22:
Thank you for the information and screen shots.
There are two different issues (not sure they are bugs yet):

The android app is likely subscribing and publishing to specific topics, which are not "linked" to the robot. For example, the Windows MF camera - unless remapped - will publish to /win_camera/image_raw. The android app - unless told otherwise - is likely subscribing to something like /camera/image_raw. You can fix this in the launch file of the robot, so that it publishes camera frames to the topic that the android app expects.

The second issue is that the joystick teleop is on a different topic than the turtlesim. Turtlesim is subscribing on /turtlesim1/cmd_vel, which the android app is not publishing to. I don't know what it is publishing to. There either needs to be a remapping or something similar.

ooeygui, 2020-08-17:
@HanDaSeul Were you able to track this down? (I don't have an android device to test on, so I cannot help much further)

ooeygui, 2020-09-01:
Please reopen if this still occurs

closed 2020-09-01 by ooeygui