ms-iot/ROSOnWindows #254: [rosserial_server] Unable to look up message: Unable to import message module

seanyen, 2020-07-31:
When using `rosserial_server` on a ROS node, the following error messages show up:

```
[ WARN] [1596233541.330902100]: Unable to look up message: Unable to import message module xxx_msgs.
[ WARN] [1596233541.331036100]: Advertising on topic [/xxx] with an empty message definition.  Some tools (e.g. rosbag) may not work correctly.
```

One possible reason to this failure is that one of the dependency `PyYaml` cannot incorporate well in the Python\C embedded environment, and for more details. Please see https://github.com/yaml/pyyaml/issues/423.

And as a workaround, you may check the version of `PyYaml` in your system and use `pip install -U --force pyyaml==3.12` to downgrade to the last known good build.

xuliujian, 2020-08-17:
@seanyen , i installed pyyaml=5.3.1, rosserial_server can work. So i don't think the issue is caused by this package.

seanyen, 2020-08-18:
@xuliujian Thanks for sharing your data. Do you mind sharing more about your environment? (Such as, what's the ROS distro, Python version, etc?)