ms-iot/ROSOnWindows #256: opencv-python fails to install (while installing melodic-desktop_full)

lilustga, 2020-08-13:
**Describe the bug**
While installing melodic and it's dependencies on windows, the following error occurs.

Collecting opencv-python
  Using cached https://files.pythonhosted.org/packages/a1/d6/8422797e35f8814b1d9842530566a949d9b5850a466321a6c1d5a99055ee/opencv-python-4.3.0.38.tar.gz
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "c:\users\lior\appdata\local\temp\chocolatey\pip-install-g8hmiv\opencv-python\setup.py", line 448, in <module>
        main()
      File "c:\users\lior\appdata\local\temp\chocolatey\pip-install-g8hmiv\opencv-python\setup.py", line 99, in main
        % {"ext": re.escape(sysconfig.get_config_var("EXT_SUFFIX"))}
      File "c:\opt\python27amd64\lib\re.py", line 210, in escape
        s = list(pattern)
    TypeError: 'NoneType' object is not iterable

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in c:\users\lior\appdata\local\temp\chocolatey\pip-install-g8hmiv\opencv-python\

**To Reproduce**
Steps to reproduce the behavior:
*Follow the instructions to install melodic: http://wiki.ros.org/Installation/Windows

**Environment information:**
 - OS version: `1903`

lilustga, 2020-08-13:
This is the same issue, on linux: https://stackoverflow.com/questions/63346648/python-2-7-installing-opencv-via-pip-virtual-environment.

The workaround is to pin opencv-python to an older version.
'pip2 install opencv-python==4.2.0.32`

rs-s, 2020-08-21:
I have this same issue on Windows and installing the older version of opencv-python didn't resolve it.  
Is there a way to install Melodic without opencv and to add it afterwards?

ooeygui, 2020-08-21:
@rs-s at the moment, melodic ROS install has this baked in as a critical as a dependency. Pinning the version should have resolved this - can you post the output and if you can the chocolatey log?