ms-iot/ROSOnWindows #258: Build Issues using Colcon Build on ROS1 workspace

seanyen, 2020-08-20:
This is a meta ticket to capture the issues that when doing `colcon build` on ROS 1 workspace.

johnwason, 2020-08-21:
After updating using `choco` I am getting this error when building with colcon:

~~~

Starting >>> rp260_parallel_gripper_a
Starting >>> rp260_support
Starting >>> rpi_general_robotics_toolbox_py
--- stderr: rp260_parallel_gripper_a
CMake Error at CMakeLists.txt:2 (project):
  Generator

    Visual Studio 16 2019

  could not find any instance of Visual Studio.

---
Failed   <<< rp260_parallel_gripper_a [0.30s, exited with code 1]
Aborted  <<< rp260_support [0.24s]
Aborted  <<< rpi_general_robotics_toolbox_py [0.20s]

Summary: 0 packages finished [0.59s]
  1 package failed: rp260_parallel_gripper_a
  2 packages aborted: rp260_support rpi_general_robotics_toolbox_py
  3 packages had stderr output: rp260_parallel_gripper_a rp260_support rpi_general_robotics_toolbox_py
  1 package not processed

~~~

v-liujxu, 2020-08-24:
@johnwason , please try to reinstall the visual studio, or configure the VS160COMNTOOLS variable so that CMake correctly detects Visual Studio via following command if you were using Community version.

`export  VS160COMNTOOLS="/c/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/Tools"`

ooeygui, 2020-08-24:
@johnwason I just want to confirm: When you launch the terminal, do you start with the x64 Native Tools Command Prompt for VS2019?

johnwason, 2020-08-24:
@ooeygui yeah I am using the 2019 community developer command prompt.

```
C:\Users\>cl
Microsoft (R) C/C++ Optimizing Compiler Version 19.27.29111 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

usage: cl [ option... ] filename... [ /link linkoption... ]
```

johnwason, 2020-08-24:
Currently I am getting this output:

~~~
Starting >>> rp260_parallel_gripper_a
Starting >>> rp260_support
Starting >>> rp260_testbed
Starting >>> rpi_general_robotics_toolbox_py
--- stderr: rp260_testbed
'do' is not recognized as an internal or external command,
operable program or batch file.
The process cannot access the file because it is being used by another process.
The process cannot access the file because it is being used by another process.
The process cannot access the file because it is being used by another process.
The syntax of the command is incorrect.
Traceback (most recent call last):
  File "C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_testbed/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_testbed/devel/env.bat')
  File "c:\opt\ros\noetic\x64\lib\site-packages\catkin\environment_cache.py", line 62, in generate_environment_script
    output = subprocess.check_output([env_script, sys.executable, '-c', python_code])
  File "C:\opt\python37amd64\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\opt\python37amd64\lib\subprocess.py", line 487, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_testbed/devel/env.bat', 'C:\\opt\\python37amd64\\python.exe', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 255.
CMake Error at C:/opt/ros/noetic/x64/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(C:/opt/python37amd64/python.exe
  "C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_testbed/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  C:/opt/ros/noetic/x64/share/catkin/cmake/all.cmake:208 (safe_execute_process)
  C:/opt/ros/noetic/x64/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

---
Failed   <<< rp260_testbed [8.80s, exited with code 1]
--- stderr: rp260_support
'do' is not recognized as an internal or external command,
operable program or batch file.
The process cannot access the file because it is being used by another process.
The process cannot access the file because it is being used by another process.
The process cannot access the file because it is being used by another process.
The syntax of the command is incorrect.
Traceback (most recent call last):
  File "C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_support/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_support/devel/env.bat')
  File "c:\opt\ros\noetic\x64\lib\site-packages\catkin\environment_cache.py", line 62, in generate_environment_script
    output = subprocess.check_output([env_script, sys.executable, '-c', python_code])
  File "C:\opt\python37amd64\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\opt\python37amd64\lib\subprocess.py", line 487, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_support/devel/env.bat', 'C:\\opt\\python37amd64\\python.exe', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 255.
CMake Error at C:/opt/ros/noetic/x64/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(C:/opt/python37amd64/python.exe
  "C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_support/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  C:/opt/ros/noetic/x64/share/catkin/cmake/all.cmake:208 (safe_execute_process)
  C:/opt/ros/noetic/x64/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

---
Failed   <<< rp260_support [8.89s, exited with code 1]
--- stderr: rp260_parallel_gripper_a
'do' is not recognized as an internal or external command,
operable program or batch file.
The process cannot access the file because it is being used by another process.
The process cannot access the file because it is being used by another process.
The process cannot access the file because it is being used by another process.
The syntax of the command is incorrect.
[... 5 lines cut ...]
  File "C:\opt\python37amd64\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\opt\python37amd64\lib\subprocess.py", line 487, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_parallel_gripper_a/devel/env.bat', 'C:\\opt\\python37amd64\\python.exe', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 255.
CMake Error at C:/opt/ros/noetic/x64/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(C:/opt/python37amd64/python.exe
  "C:/Users/<user>/Documents/rp260/catkin_ws/build/rp260_parallel_gripper_a/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  C:/opt/ros/noetic/x64/share/catkin/cmake/all.cmake:208 (safe_execute_process)
  C:/opt/ros/noetic/x64/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

---
Failed   <<< rp260_parallel_gripper_a [9.01s, exited with code 1]
Aborted  <<< rpi_general_robotics_toolbox_py [10.6s]

Summary: 0 packages finished [11.2s]
  3 packages failed: rp260_parallel_gripper_a rp260_support rp260_testbed
  1 package aborted: rpi_general_robotics_toolbox_py
  3 packages had stderr output: rp260_parallel_gripper_a rp260_support rp260_testbed
  1 package not processed

~~~

seanyen, 2020-08-26:
@johnwason Hi! Thanks for posting the logs. The errors look very similar to what I am trying to address here: https://github.com/ros/catkin/pull/1116

Would you be able to patch `c:\opt\ros\noetic\x64\setup.bat` and `c:\opt\ros\noetic\x64\share\catkin\cmake\templates\setup.bat.in` to confirm if it helps this case?

johnwason, 2020-08-27:
@seanyen I haven't had a chance to look at this today, I will try tomorrow

seanyen, 2020-09-22:
The fix is in the latest [Noetic](https://discourse.ros.org/t/ros-on-windows-noetic-release-v20200831-0-0-2009101215) release. I am closing it first and feel free to reopen if this still requires the team's attention. Thanks!

closed 2020-10-01 by seanyen