ms-iot/ROSOnWindows #260: sdformat.dll is in wrong directory

johnwason, 2020-08-27:
**Describe the bug**
sdformat.dll is in the `lib` directory instead of `bin` in `c:\opt\rosdeps\x64`. It is not found in the path when using gazebo on a clean install.

**To Reproduce**
Attempt to launch gazebo.

    roslaunch gazebo_ros empty_world.launch

The launch will fail due to missing DLL

**Expected behavior**
Gazebo should launch.

**Environment information:**
 Windows 10 with clean noetic install

ooeygui, 2020-08-27:
Thanks for the report and investigation! We'll get right on fixing this. Are you unblocked if you move it manually?

johnwason, 2020-08-27:
Yeah, adding the `lib` directory to `PATH` fixes the problem.

ooeygui, 2020-08-29:
Hi @johnwason,
Sorry for the delay in responding. I've been looking into how your system could get into this state; and need some more information.

What I'm seeing:
When calling `C:\opt\ros\melodic\x64\setup.bat`, this executes a python script called `_setup_util.py` which constructs a new environment for this ROS install. During the construction of the path, it adds bin & lib:

``` python
PATH_TO_ADD_SUFFIX = ['bin']
if IS_WINDOWS:
    # while catkin recommends putting dll's into bin, 3rd party packages often put dll's into lib
    # since Windows finds dll's via the PATH variable, prepend it with path to lib
    PATH_TO_ADD_SUFFIX.extend(['lib'])

# subfolder of workspace prepended to CMAKE_PREFIX_PATH
ENV_VAR_SUBFOLDERS = {
    'CMAKE_PREFIX_PATH': '',
    'LD_LIBRARY_PATH' if not IS_DARWIN else 'DYLD_LIBRARY_PATH': 'lib',
    'PATH': PATH_TO_ADD_SUFFIX,
    'PKG_CONFIG_PATH': os.path.join('lib', 'pkgconfig'),
    'PYTHONPATH': 'lib/site-packages',
}
```

Which results in the following path environment variable:
``` batch
Path=c:\opt\vcpkg;c:\opt\vcpkg\installed\x64-windows\bin;C:/opt/ros/melodic/x64\bin;C:/opt/ros/melodic/x64\lib;C:/opt/rosdeps/x64\bin;C:/opt/rosdeps/x64\lib;C:\opt\python27amd64\;C:\opt\python27amd64\Scripts; <rest omitted> 
```

This environment has both `c:\opt\rosdeps\x64\bin` and `c:\opt\rosdeps\x64\lib`.

Asks:
  * Does your `C:\opt\ros\melodic\x64\_setup_util.py` have this entry near the top? If not, could you attach your file?
  * If so, do you have any scripts which modify the environment after calling `setup.bat`?

johnwason, 2020-08-29:
@hehonglu123 can you check your setup.bat file?

hehonglu123, 2020-08-29:
Here's my `C:\opt\ros\noetic\x64\setup.bat`

~~~
:: generated from colcon_core/shell/template/prefix_chain.bat.em
@echo off

:: This script extends the environment with the environment of other prefix
:: paths which were sourced when this file was generated as well as all
:: packages contained in this prefix path.

:: source this prefix
call:_colcon_prefix_chain_bat_call_script "%%~dp0local_setup.bat"

goto:eof

:: function to source another script with conditional trace output
:: first argument: the path of the script
:_colcon_prefix_chain_bat_call_script
  if exist "%~1" (
    if "%COLCON_TRACE%" NEQ "" echo call "%~1"
    call "%~1%"
  ) else (
    echo not found: "%~1" 1>&2
  )
goto:eof

~~~

seanyen, 2020-09-16:
@johnwason @hehonglu123 We made a new [Noetic Release v20200831.0.0.2009101215](https://discourse.ros.org/t/ros-on-windows-noetic-release-v20200831-0-0-2009101215). This should be addressed and fixed. I am closing it now and feel free to reopen if it is still an issue.

closed 2020-09-16 by seanyen