ms-iot/ROSOnWindows #279: [Noetic] Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link at time xxxx according to authority unknown_publisher

seanyen, 2020-09-25:
When running `robot_localization` in Noetic, you might see lots of this messages flooding on the console:

```
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link at time 2.187000 according to authority unknown_publisher
```

This is an known issue and the related discussions can be found:

* https://answers.ros.org/question/359968/tf_repeated_data-error-with-robot_localization-ukf-package/
* https://github.com/ros/geometry2/issues/467

johannesWen, 2022-03-02:
Hi, 
I had the same problem. In my case it was the "robot_state_publisher" node. I could fix it with adding the "ignore_timestamp" parameter.
```xml
<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" respawn="true" >
  <param name="ignore_timestamp" value="True" />
</node>
```

kvnptl, 2022-09-06:
In order to avoid or hide the warning message on terminal, follow this [#467](https://github.com/ros-planning/navigation/issues/1125#issuecomment-1238647110)