ms-iot/ROSOnWindows #292: [melodic]Unable to use interactive markers in RViz

tota1Noob, 2020-10-13:
I followed the installation tutorial http://wiki.ros.org/Installation/Windows and successfully installed ROS melodic on my windows 10 laptop. Same with MoveIt, following https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html.
But when I tried the tutorial here http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html, I was not able to move the robot with interactive markers. I could see the markers, but they were not highlighted when I hover the mouse over them. And when I tried to drag the robot, only the camera moved.

tota1Noob, 2020-10-14:
Below lists the "Display Devices" part of the dxdiag.exe output. The first and second part of the output refer to the integrated Intel UHD Graphics card, and the third part is the on-board NVIDIA MX250.

           Card name: Intel(R) UHD Graphics
        Manufacturer: Intel Corporation
           Chip type: Intel(R) UHD Graphics Family
            DAC type: Internal
         Device Type: Full Device (POST)
          Device Key: Enum\PCI\VEN_8086&DEV_9BCA&SUBSYS_3FD717AA&REV_04
       Device Status: 0180200A [DN_DRIVER_LOADED|DN_STARTED|DN_DISABLEABLE|DN_NT_ENUMERATOR|DN_NT_DRIVER] 
 Device Problem Code: No Problem
 Driver Problem Code: Unknown
      Display Memory: 8235 MB
    Dedicated Memory: 128 MB
       Shared Memory: 8107 MB
        Current Mode: 2560 x 1600 (32 bit) (60Hz)
         HDR Support: Not Supported
    Display Topology: Extend
 Display Color Space: DXGI_COLOR_SPACE_RGB_FULL_G22_NONE_P709
     Color Primaries: Red(0.639648,0.330078), Green(0.299805,0.599609), Blue(0.150391,0.059570), White Point(0.313477,0.329102)
   Display Luminance: Min Luminance = 0.500000, Max Luminance = 270.000000, MaxFullFrameLuminance = 270.000000
        Monitor Name: Generic PnP Monitor
       Monitor Model: unknown
          Monitor Id: AUO2026
         Native Mode: 2560 x 1600(p) (60.001Hz)
         Output Type: Internal
Monitor Capabilities: HDR Not Supported
Display Pixel Format: DISPLAYCONFIG_PIXELFORMAT_32BPP
      Advanced Color: Not Supported
         Driver Name: C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_93f2d438e2d7d4b7\igdumdim64.dll,C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_93f2d438e2d7d4b7\igd10iumd64.dll,C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_93f2d438e2d7d4b7\igd10iumd64.dll,C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_93f2d438e2d7d4b7\igd12umd64.dll
 Driver File Version: 26.20.0100.7323 (English)
      Driver Version: 26.20.100.7323
         DDI Version: 12
      Feature Levels: 12_1,12_0,11_1,11_0,10_1,10_0,9_3,9_2,9_1
        Driver Model: 2.6 WDDM
 Graphics Preemption: Triangle
  Compute Preemption: Thread
            Miracast: Supported
      Detachable GPU: No
 Hybrid Graphics GPU: Integrated
      Power P-states: Not Supported
      Virtualization: Paravirtualization 
          Block List: No Blocks
  Catalog Attributes: Universal:False Declarative:True 
   Driver Attributes: Final Retail
    Driver Date/Size: 19/10/7 8:00:00, 1643128 bytes
         WHQL Logo'd: Yes
     WHQL Date Stamp: Unknown
   Device Identifier: {D7B78E66-D88A-11CF-8168-B31FBFC2D735}
           Vendor ID: 0x8086
           Device ID: 0x9BCA
           SubSys ID: 0x3FD717AA
         Revision ID: 0x0004
  Driver Strong Name: oem71.inf:5f63e534c1887fe6:iCML_w10_DS:26.20.100.7323:PCI\VEN_8086&DEV_9BCA&SUBSYS_3FD717AA
      Rank Of Driver: 00CF0001
         Video Accel: 
         DXVA2 Modes: DXVA2_ModeMPEG2_VLD  DXVA2_ModeMPEG2_IDCT  DXVA2_ModeVC1_D2010  DXVA2_ModeWMV9_IDCT  DXVA2_ModeVC1_IDCT  DXVA2_ModeH264_VLD_NoFGT  DXVA2_ModeH264_VLD_Stereo_Progressive_NoFGT  DXVA2_ModeH264_VLD_Stereo_NoFGT  DXVA2_ModeH264_VLD_Multiview_NoFGT  DXVA2_ModeVP8_VLD  DXVA2_ModeHEVC_VLD_Main  DXVA2_ModeHEVC_VLD_Main10  DXVA2_ModeVP9_VLD_Profile0  DXVA2_ModeVP9_VLD_10bit_Profile2  
   ......
        D3D9 Overlay: Supported
             DXVA-HD: Supported
        DDraw Status: Enabled
          D3D Status: Enabled
          AGP Status: Enabled
       MPO MaxPlanes: 3
            MPO Caps: RGB,YUV,BILINEAR,HIGH_FILTER,STRETCH_YUV,STRETCH_RGB,IMMEDIATE,HDR (MPO3)
         MPO Stretch: 5.000X - 0.334X
     MPO Media Hints: colorspace Conversion 
         MPO Formats: NV12,YUY2,R16G16B16A16_FLOAT,R10G10B10A2_UNORM,R8G8B8A8_UNORM,B8G8R8A8_UNORM
    PanelFitter Caps: RGB,YUV,BILINEAR,HIGH_FILTER,STRETCH_YUV,STRETCH_RGB,IMMEDIATE,HDR (MPO3)
 PanelFitter Stretch: 5.000X - 0.334X
   Extension Drivers: 
[... 125 lines cut ...]
    Driver Date/Size: 20/9/17 8:00:00, 1048720 bytes
         WHQL Logo'd: Yes
     WHQL Date Stamp: Unknown
   Device Identifier: Unknown
           Vendor ID: 0x10DE
           Device ID: 0x1D13
           SubSys ID: 0x3F1917AA
         Revision ID: 0x00A1
  Driver Strong Name: oem59.inf:0f066de3ef96164d:Section110:27.21.14.6020:pci\ven_10de&dev_1d13&subsys_3f1917aa
      Rank Of Driver: 00CF0001
         Video Accel: Unknown
         DXVA2 Modes: 
      Deinterlace Caps: n/a
        D3D9 Overlay: Unknown
             DXVA-HD: Unknown
        DDraw Status: Enabled
          D3D Status: Enabled
          AGP Status: Enabled
       MPO MaxPlanes: 0
            MPO Caps: Not Supported
         MPO Stretch: Not Supported
     MPO Media Hints: Not Supported
         MPO Formats: Not Supported
    PanelFitter Caps: Not Supported
 PanelFitter Stretch: Not Supported

lilustga, 2020-10-14:
I was able to reproduce this following your steps. This does seem to a be a bug.

Notably, planning still works and the arm can be moved around by adjusting joint angles so I suspect it's an Rviz bug. However, I will need to look into it further.

tota1Noob, 2020-10-14:
@ooeygui @lilustga Huge thanks to you guys! Thought it was me missing steps in the tutorials or something.

The planning and moving robots by adjusting joint angles does work, as I tried them yesterday. Looking forward to a solution for said bug.

matthias88, 2020-10-15:
I have the same issue with ROS noetic and moveit. I have installed ROS on 2 different systems, just for verification.
[image]

ooeygui, 2020-10-16:
We have a fix incoming.

seanyen, 2020-10-16:
@matthias88 @tota1Noob A patch release is out! ([Melodic](https://ros-win.visualstudio.com/ros-win/_packaging?_a=package&feed=public&package=ros-melodic-desktop_full&protocolType=NuGet&version=20201012.0.0.2010150001&view=overview) and [Noetic](https://ros-win.visualstudio.com/ros-win/_packaging?_a=package&feed=public&package=ros-noetic-desktop_full&protocolType=NuGet&version=20201009.0.0.2010160108&view=overview))

Please do `choco upgrade ros-<distro>-desktop_full -y` to acquire them. And let us know how it goes.

matthias88, 2020-10-17:
> @matthias88 @tota1Noob A patch release is out! [...]

@seanyen The patch fixed it under Noetic. Thanks a lot 👍

tota1Noob, 2020-10-17:
@ooeygui @seanyen It worked! Many thanks for the swift fix :D

ooeygui, 2020-11-20:
Corrected in https://github.com/ros-visualization/rviz/commit/d2610074728ffb525c9d5b42fddf8a5d3b8de61c. Thank you!!!

closed 2020-11-20 by ooeygui