ms-iot/ROSOnWindows #294: [Port] Support kuka_experimental on Windows

ooeygui, 2020-10-17:
We have had a commercial request for Kuka ROS node on Windows.
https://answers.ros.org/question/363614/build-kuka_experimental-package-on-windows-10-ros1-noetic/

This package depends on industrial core:
https://github.com/ros-industrial/industrial_core

gavanderhoorn, 2020-10-18:
> This package depends on industrial core:
> https://github.com/ros-industrial/industrial_core

Only for an RViz configuration file.

That dependency is not important.

ooeygui, 2020-10-18:
@gavanderhoorn Thanks for that tip - I started porting it - but will look into the Kuka one. Looking into it further, it was last updated for indigo and still marked experimental. Is there a more current Kuka ROS implementation we should be looking into?

gavanderhoorn, 2020-10-19:
> Looking into it further, it was last updated for indigo and still marked experimental. Is there a more current Kuka ROS implementation we should be looking into?

We follow a branch-when-needed policy. Afaik, everything in there can be built on Indigo, Kinetic and Melodic. Hence no new branch.

And I expect only the RSI and EKI `hardware_interface`s to require any changes. All other packages are robot support packages, which contain no code.

traversaro, 2020-10-19:
For what regards the KUKA RSI server, we ported it in the past for an internal project on Windows. At the time we did not ported the fixes upstream because the Windows interested seemed low, and also because we needed a non-ROS package to consume it as a dependency, so even if the fix was upstreamed we could not use it. However, if you want to take an inspiration the version that we are using is https://gist.github.com/traversaro/93de691ffb1b5344d1b127dca58233d3 (it is a modified version of https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_rsi_hw_interface/include/kuka_rsi_hw_interface/udp_server.h). However, note that, at least for what concern RSI it a protocol that is suppose to exchange data with the robot at 250 Hz, so depending on the specific application it may not be trivial to obtain on Windows the same performance that you obtained (for example) on a Linux with PREEMPT RT patch.

gavanderhoorn, 2020-10-19:
@traversaro wrote:
> note that, at least for what concern RSI it a protocol that is suppose to exchange data with the robot at 250 Hz, so depending on the specific application it may not be trivial to obtain on Windows the same performance that you obtained (for example) on a Linux with PREEMPT RT patch.

yeah, I've mentioned something similar to the user requesting this.

They were more interested in EKI though, which will probably be doable.

But I don't guarantee anything.

ooeygui, 2020-10-19:
@traversaro Thanks for the info. 
250hz is achievable on Windows using system configuration. (We have documentation incoming which will cover sub-millisecond time criticality)

This would be a good test case for it.

gavanderhoorn, 2020-10-29:
Out of curiosity: are there any updates @ooeygui ?

ooeygui, 2020-10-29:
@gavanderhoorn I was able to borrow two kuka arms, and trying to build a workcell in my garage LOL. 
I have a plan for the realtime control loop on Windows. 

If you have interested customers, I would love to chat about them!

gavanderhoorn, 2020-10-31:
> I was able to borrow two kuka arms, and trying to build a workcell in my garage

:) hm, maybe I should start porting stuff too ..

ooeygui, 2020-10-31:
I created a Windows Staging Fork here: https://github.com/ms-iot/kuka_experimental. We will PR up to ROS industrial once we are able to validate on a robot.

ooeygui, 2020-11-02:
builds on windows now...
https://github.com/ms-iot/kuka_experimental/pull/2
Setting up to test it:
[image]

traversaro, 2020-11-02:
Hi @ooeygui, those seems to be KUKA iiwa, and I am afraid that they do not support the RSI protocol, did you checked with KUKA?

ooeygui, 2020-11-02:
@traversaro well, bummer :-(

traversaro, 2020-11-02:
As far as I know those robots support another protocol called Fast Research Protocol, but I may be wrong.

gavanderhoorn, 2020-11-02:
Well .. it depends a bit on which version of the IIWA that is.

But it's likely they don't support it.

> As far as I know those robots support another protocol called Fast Research Interface (FRI), but I may be wrong.

Older IIWAs would use FRI indeed. With newer you'd use SmartServo/DirectServo.

ooeygui, 2020-11-02:
@gavanderhoorn @traversaro thanks for the chat. I haven't reached out to my contacts at Kuka yet. 
These are 2017 models. I am missing a few parts, so haven't been able to power them up yet to get more details.

gavanderhoorn, 2020-11-02:
Is your "other project" going to / interested in using ROS 2 with those robots? If so, send me an email.

gavanderhoorn, 2021-03-05:
@ooeygui: what's the status here? You mentioned (here or on ROS Answers) you had some dependencies of the RSI / EKI packages which also needed some patches. Would that be `ros_control/hardware_interface` by any chance?

It seems we're running into similar issues with `abb_robot_driver`. I've @-mentioned you there.

ooeygui, 2021-03-08:
@gavanderhoorn Thanks for the ping on this. I have a PR into the ms-iot staging branch, but haven't had an opportunity to test it. https://github.com/ms-iot/kuka_experimental/pull/2. I haven't investigated the ros_control issues, I'm going to work with @lilustga who is the manipulation lead in AERo.

gavanderhoorn, 2021-03-08:
I've left a comment on the PR.

I believe we're running into similar problems with base `ros_control` and `ERROR` in https://github.com/ros-industrial/abb_robot_driver/issues/17.