ms-iot/ROSOnWindows #295: Enable locally built Dependency to override system installed version of the dependency

ooeygui, 2020-10-18:
**What:**
Customer has a requirement to support building a dependency using non-default parameters. However, while the dependency is building in the workspace the system version of that dependency overrides the local version.

**Why:**
Customer requires a precision version of bullet3, which is not currently the default build.

ooeygui, 2020-10-22:
In `C:\opt\ros\noetic\x64\etc\catkin\profile.d\vcpkg.bat` the following line overrides local builds:
`set "CMAKE_PREFIX_PATH=C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;%CMAKE_PREFIX_PATH%"`
Recommend changing to:
`set "CMAKE_PREFIX_PATH=%CMAKE_PREFIX_PATH%;C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows"

So that locally built dependencies are found first.

Levi-Armstrong, 2020-10-22:
Another issue I have ran into is that it finds the GTest package but all the cmake variables are empty except `GTEST_INCLUDE_DIRS`. Variables `GTEST_LIBRARIES` and `GTEST_MAIN_LIBRARIES` are empty. This is the case on both linux and windows, but was only an issue when building windows.

``` cmake
find_package(GTest REQUIRED)

if(NOT TARGET GTest::GTest)
  add_library(GTest::GTest INTERFACE IMPORTED)
  set_target_properties(GTest::GTest PROPERTIES INTERFACE_INCLUDE_DIRECTORIES "${GTEST_INCLUDE_DIRS}")
  if (${GTEST_LIBRARIES})
    set_target_properties(GTest::GTest PROPERTIES INTERFACE_LINK_LIBRARIES "${GTEST_LIBRARIES}")
  else()
    if (MSVC)
      set_target_properties(GTest::GTest PROPERTIES INTERFACE_LINK_LIBRARIES "gtest.lib")
    else()
      set_target_properties(GTest::GTest PROPERTIES INTERFACE_LINK_LIBRARIES "libgtest.so")
    endif()
  endif()
endif()

if(NOT TARGET GTest::Main)
  add_library(GTest::Main INTERFACE IMPORTED)
  set_target_properties(GTest::Main PROPERTIES INTERFACE_INCLUDE_DIRECTORIES "${GTEST_INCLUDE_DIRS}")
  if (${GTEST_MAIN_LIBRARIES})
    set_target_properties(GTest::Main PROPERTIES INTERFACE_LINK_LIBRARIES "${GTEST_MAIN_LIBRARIES}")
  else()
    if (MSVC)
      set_target_properties(GTest::Main PROPERTIES INTERFACE_LINK_LIBRARIES "gtest_main.lib")
    else()
      set_target_properties(GTest::Main PROPERTIES INTERFACE_LINK_LIBRARIES "libgtest_main.so")
    endif()
  endif()
endif()
```

Levi-Armstrong, 2020-10-22:
The other issue I have ran into is related having dependencies in my workspace that are also installed on the system. The interesting thing is that find_package finds the correct config but when building it is using the system version for headers and libraries unless I add the BEFORE shown below. This means that I am unable to leverage cmake targets because of this issue for packages that are in workspace and also installed on the system.

Before:
``` cmake
include_directories(${FCL_INCLUDE_DIRS})
link_directories(${FCL_LIBRARY_DIRS})
```

After:
``` cmake
include_directories(BEFORE ${FCL_INCLUDE_DIRS})
link_directories(BEFORE ${FCL_LIBRARY_DIRS})
```

Levi-Armstrong, 2020-10-22:
The other issue I ran into which is not related to this issue but explained [here](https://github.com/ros/console_bridge/issues/91). Oddly this issue was not present until I started building on Windows, and not sure why this was not an issues when I was building the same package on linux.

Levi-Armstrong, 2020-10-25:
@ooeygui I am working on getting the motion planning packages to build and ran into an issue where OMPL has incorrect include path for Eigen. Not sure if this an error in how ompl generates the its Config or related to how they are packaged for windows. When building the ros package for windows do they get moved? If so then this will be a problem for OMPL because it hard code the path for its dependencies in omplConfig.cmake instead of leveraging cmake find_dependency.

```
OMPL INCLUDE DIRS: C:/opt/ros/noetic/x64/include;/include/eigen3
```

ooeygui, 2020-10-29:
OMPL is built using vcpkg. In the ROS deployment, the config file `C:\opt\ros\noetic\x64\share\ompl\omplConfig.cmake`, has this entry: `set(OMPL_INCLUDE_DIRS "${OMPL_INCLUDE_DIR};${_IMPORT_PREFIX}/include/eigen3;;")`

I'm rebuilding ompl in vcpkg to see where that is coming in from.

Levi-Armstrong, 2020-10-30:
@ooeygui I opened up a [PR](https://github.com/ompl/ompl/pull/757) against OMPL to fix the missing libraries boost, flann, etc which they were not including.

Levi-Armstrong, 2020-11-14:
@ooeygui The fix for ompl has been merged. What is the process for getting the fix included into the ros windows release?