ms-iot/ROSOnWindows #298: Error when using rosbag fix in melodic on Windows

void32, 2020-11-03:
**Describe the bug**
When repairing a rosbag file with the .active extension, rosbag fix yields the following error:
`IOError: [Errno 2] No such file or directory: 'C:\\opt\\ros\\melodic\\x64\\share\\joy/migration_rules/Joy.bmr`

**To Reproduce**
1. Reindex broken rosbag:
`D:\rosbags>rosbag reindex 71ca1737-23cf-4909-ae51-fe819c46e351_2019-03-22-15-18-33_14.bag.active`

2. Fix bag:

The latter returns this error:
`D:\rosbags>rosbag fix 71ca1737-23cf-4909-ae51-fe819c46e351_2019-03-22-15-18-33_14.bag.active 71ca1737-23cf-4909-ae51-fe819c46e351_2019-03-22-15-18-33_14.bag
Traceback (most recent call last):
  File "C:\opt\ros\melodic\x64\bin\rosbag", line 35, in <module>
    rosbag.rosbagmain()
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosbag\rosbag_main.py", line 1012, in rosbagmain
    cmds[cmd](argv[2:])
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosbag\rosbag_main.py", line 440, in fix_cmd
    migrator = MessageMigrator(rules, plugins=not options.noplugins)
  File "C:\opt\ros\melodic\x64\lib\site-packages\rosbag\migration.py", line 557, in __init__
    exec(open(pkg_dir + "/" + r).read(), scratch_locals)
IOError: [Errno 2] No such file or directory: 'C:\\opt\\ros\\melodic\\x64\\share\\joy/migration_rules/Joy.bmr'`

**Expected behavior**
rosbag fix should create a repaired rosbag file from the reindexed bag.

**Environment information:**
Microsoft Windows
Version 1903 (OS BUILD 18362.1139)

Chocolatey v0.10.15
assimp 4.0.1.1809242322
boost 1.66.0.1904102339
bullet3 2.87.0.1809242326
bzip2 1.0.6.1901242108
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
console_bridge 0.4.0.1809242322
cppunit 1.12.1.1809242322
cppzmq 4.2.2.1811162207
dlfcn-win32 1.1.1.1811162226
eigen 3.3.4.1809242322
freeimage 3.17.0.1811160037
gazebo9 9.13.1.1
google-mock 1.8.0.1809242326
google-test 1.8.0.1809242322
gtk2 2.22.1.1904131648
ilmbase 2.2.1.1811210800
jasper 2.0.14.1811210919
jxrlib 1.1.0.1811210809
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
lcms 2.8.0.1811210903
libcurl 7.58.0.1809242322
libflann 1.9.1.1809090720
libgraphviz 2.41.0.1810022039
libjpeg-turbo 1.5.3.1809242322
liblzma 5.2.4.1811210837
libogg 1.3.3.1809090720
libopencv 3.4.1.1809242322
libpcl 1.8.1.1809090720
libpng 1.6.35.1809242322
libraw 0.19.0.1811210804
libtbb-dev 2018.6.0.1812011750
libtheora 1.1.1.1809090720
libwebp 0.6.1.1811210731
log4cxx 0.10.0.1809242322
lz4 1.8.1.1809242322
ogre 1.10.11.1901312214
openexr 2.2.1.1811210832
openjpeg 2.3.0.1811210736
openssl 1.1.1.1809242322
pkg-config 0.29.2.1809242322
poco 1.8.1.1809242322
protobuf 3.6.1.1811130545
pyqt5 5.10.1.1809242322
pyside2 5.10.1.1809242322
qt5-sdk 5.10.1.1903182340
qwt 6.1.3.1811212304
ros-catkin-tools 0.0.1.1903221831
ros-melodic-desktop_full 20200821.1
ros-noetic-desktop_full 20200831.0.0.2009101215
ros-python2 2.7.15.1904300710
sbcl 0.0.0.1809242326
sdl 1.2.15.1809242322
sdl_image 1.2.12.1809242322
sip 4.19.8.1809242322
tiff 4.0.9.1811210742
tinyxml 2.6.2.1809242322
tinyxml2 6.1.0.1809242322
urdfdom 1.0.0.1809242322
urdfdom_headers 1.0.0.1809242322
vcpython27 9.0.30729.1809090720
vcredist140 14.26.28720.3
vcredist2010 10.0.40219.2
yaml-cpp 0.6.2.1901170034
zeromq 4.2.5.1811162203
zlib 1.2.11.1809242322
70 packages installed.

Did you know Pro / Business automatically syncs with Programs and
 Features? Learn more about Package Synchronizer at
 https://chocolatey.org/compare

ooeygui, 2020-11-03:
Thank you for the report. (Did this get accidentally reported twice? https://github.com/ms-iot/ROSOnWindows/issues/299).

Could you post a link to the rosbag you are using? I'll take a look.

void32, 2020-11-05:
Yes, GitHub returned an error when I reported the bug the first time(I didn't realize that it was created).

Thanks. You can download the rosbag from here: https://myigis-my.sharepoint.com/personal/mbn_i-gis_dk/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fmbn%5Fi%2Dgis%5Fdk%2FDocuments%2FShare%2Frosbags%2F71ca1737%2D23cf%2D4909%2Dae51%2Dfe819c46e351&originalPath=aHR0cHM6Ly9teWlnaXMtbXkuc2hhcmVwb2ludC5jb20vOmY6L2cvcGVyc29uYWwvbWJuX2ktZ2lzX2RrL0VtN2MwQ2pRdVlSS3FIQmM2N05QaFRNQjdONDlOU0RzRzdjXzNRTkdjYVNpUWc_cnRpbWU9UDZzZlpvNkIyRWc

seanyen, 2020-12-31:
@void32 Hi! I checked your bag files and I think the error was caused by a bug in rosbag on Windows. I will address this issue in the next Melodic release. Stay tuned!

seanyen, 2021-01-04:
@void32 The fix are now included in [this release](https://discourse.ros.org/t/ros-on-windows-melodic-release-v20201221-0-0-2012310001/18269). Please upgrade your ROS installation and retry again. I will be closing this issue for now. Feel free to reopen this issue if you still run into the same problem.

closed 2021-01-04 by seanyen