ms-iot/ROSOnWindows #301: rosdep update error

nawara72, 2020-11-18:
Hi 
I am new to ROS, so excuse my lack of knowledge please
I followed the instruction on https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html
and it all ran well and I have foxy now
I then tried to instal cv_bridge
ran the following (using instruction https://ms-iot.github.io/ROSOnWindows/GettingStarted/UsingROSonWindows.html

c:\opt\ros\melodic\x64\setup.bat
mkdir c:\catkin_ws\src
cd c:\catkin_ws
rosinstall_generator cv_bridge --deps --exclude RPP --tar --flat > pkg.rosinstall
vcs import --force src < pkg.rosinstall
rosdep update

got the following error would appreciate your help

reading in sources list data from /etc/ros\rosdep\sources.list.d
Hit https://raw.githubusercontent.com/ms-iot/rosdistro-db/init_windows/rosdep/win-chocolatey.yaml
Hit https://raw.githubusercontent.com/ms-iot/rosdistro-db/init_windows/rosdep/vcpkg.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml

ERROR: Rosdep experienced an error: DLL load failed: The specified module could not be found.
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.20.0

Traceback (most recent call last):
  File "c:\users\cv\appdata\local\programs\python\python37\lib\site-packages\rosdep2\main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "c:\users\cv\appdata\local\programs\python\python37\lib\site-packages\rosdep2\main.py", line 428, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "c:\users\cv\appdata\local\programs\python\python37\lib\site-packages\rosdep2\main.py", line 437, in _no_args_handler
    return command_handlers[command](options)
  File "c:\users\cv\appdata\local\programs\python\python37\lib\site-packages\rosdep2\main.py", line 649, in command_update
    ros_distro=options.ros_distro)
  File "c:\users\cv\appdata\local\programs\python\python37\lib\site-packages\rosdep2\sources_list.py", line 498, in update_sources_list
    print('Query rosdistro index %s' % get_index_url())
  File "c:\users\cv\appdata\local\programs\python\python37\lib\site-packages\rosdep2\rosdistrohelper.py", line 62, in get_index_url
    _check_cache()
  File "c:\users\cv\appdata\local\programs\python\python37\lib\site-packages\rosdep2\rosdistrohelper.py", line 55, in _check_cache
    if _RDCache.index_url != rosdistro.get_index_url():
  File "c:\opt\ros\foxy\x64\Lib\site-packages\rosdistro\__init__.py", line 86, in get_index_url
    user_cfg_path = os.path.join(user_config_dir('rosdistro'), cfg_file)
  File "c:\opt\ros\foxy\x64\Lib\site-packages\rosdistro\external\appdirs.py", line 192, in user_config_dir
    path = user_data_dir(appname, appauthor, None, roaming)
  File "c:\opt\ros\foxy\x64\Lib\site-packages\rosdistro\external\appdirs.py", line 85, in user_data_dir
    path = os.path.normpath(_get_win_folder(const))
  File "c:\opt\ros\foxy\x64\Lib\site-packages\rosdistro\external\appdirs.py", line 409, in _get_win_folder_with_pywin32
    from win32com.shell import shellcon, shell
ImportError: DLL load failed: The specified module could not be found.

ooeygui, 2020-11-18:
Hello there,
I'm sorry you are having a problem. the issue is that you have a Python in your path which isn't the one included in the ROS distribution. 

Please make sure to remove the python37 out of your system environment and allow the `c:\opt\ros\melodic\x64\setup.bat` use the one included in the distribution.

Additionally it seems like you are trying to use a ROS1 node in a ROS2 composition; they are treated as separate environments which ultimately communicate with each other over a bridge. I just wanted to make sure you saw that.

nawara72, 2020-11-18:
Thanks for your quick reply
I removed the python37 from path and was able to progress abot further then failed
This is what I got

c:\catkin_ws>rosdep update
reading in sources list data from /etc/ros\rosdep\sources.list.d
Hit https://raw.githubusercontent.com/ms-iot/rosdistro-db/init_windows/rosdep/win-chocolatey.yaml
Hit https://raw.githubusercontent.com/ms-iot/rosdistro-db/init_windows/rosdep/vcpkg.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in C:\Users\cv\.ros\rosdep\sources.cache

c:\catkin_ws>rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ament_lint_common: No definition of [ament_cmake_export_dependencies] for OS [windows]
uncrustify_vendor: No definition of [uncrustify] for OS [windows]
ament_xmllint: No definition of [libxml2-utils] for OS [windows]
cv_bridge: No definition of [libboost-python-dev] for OS [windows]
ament_cmake: No definition of [ament_cmake_export_include_directories] for OS [windows]
fastrtps: No definition of [asio] for OS [windows]
Continuing to install resolvable dependencies...
executing command [choco upgrade -y cppcheck]
Chocolatey v0.10.15
Upgrading the following packages:
cppcheck
By upgrading you accept licenses for the packages.
cppcheck is not installed. Installing...
cppcheck not installed. The package was not found with the source(s) listed.
 Source(s): 'https://aka.ms/ros/public'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.

Chocolatey upgraded 0/1 packages. 1 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Failures
 - cppcheck - cppcheck not installed. The package was not found with the source(s) listed.
 Source(s): 'https://aka.ms/ros/public'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.

Did you know the proceeds of Pro (and some proceeds from other
 licensed editions) go into bettering the community infrastructure?
 Your support ensures an active community, keeps Chocolatey tip top,
 plus it nets you some awesome features!
 https://chocolatey.org/compare
ERROR: the following rosdeps failed to install
  chocolatey: command [choco upgrade -y cppcheck] failed
  chocolatey: Failed to detect successful installation of [cppcheck]

ooeygui, 2020-11-18:
That looks better. The Windows distribution does not have separate metapackages for many things, so rosdep doesn't find them in the distribution. I'd love to fix the distribution so that rosdep can find them, but these are just warnings.

At this time though, it is working as best as expected.

Be Well!

closed 2020-11-18 by ooeygui