ms-iot/ROSOnWindows #303: [feature request] TKInter installed with Python on Windows (From ROS Answers)

ooeygui, 2020-11-30:
**Describe the bug**
A clear and concise description of what the bug is.
https://answers.ros.org/question/366222/ros-noetic-for-windows-is-missing-the-tcl-support-and-i-cant-install-it/

Original:
```
After installing ROS noetic for Windows following these instructions: http://wiki.ros.org/Installation/Windows

If I run python > import tkinter - Error!

If uninstall python and install 3.8.3 separetely it imports tkinter, but loses the interface with ROS (useless).

Can you give me a useable fix, please?
```

ooeygui, 2020-11-30:
The version of Python distributed with ROS on Windows does not include the TkInter modules.

beraug, 2020-11-30:
> The version of Python distributed with ROS on Windows does not include the TkInter modules.

Precisely, which is weird considering it does have a considerably new Python version included.

ooeygui, 2020-11-30:
@beraug I'd love to understand what you are using with it - can you point us at a repo we can test with? (I mean once we do the basic module import testing)

beraug, 2020-12-01:
> @beraug I'd love to understand what you are using with it - can you point us at a repo we can test with? (I mean once we do the basic module import testing)

@ooeygui Hi Lou, sure! My implementations are variations and my work cannot be public available yet due to its nature, but one of the base builds that created the error is in this repo: https://github.com/Goldob/iss_docking_automation

It uses Selenium webdriver to spin of game scenarios (which I believe is the reason of the Tkinter crash). My workflow uses Python, Gym, Tensorflow and ROS together for RL agents.

I'm pretty sure you can reproduce the error if you try to build this container, but let me know if you need anything else.

ooeygui, 2020-12-01:
@beraug That's awesome! Thank you for sharing.